ipxe/ipxe #355: Bring in LLDP support

Teknoenie, 2021-05-10:
Hi iPXE devs.  I'm looking to see if support for LLDP is possible or not.  While searching Google I came across this repository[1] which seems to have it, but the code looks quite old.  Would it be possible to get this merged into the primary iPXE code?

This would be incredibly useful for my site where I have ~3k computers that utilize LLDP for identifying their rack positions and naming them based on type, dimensions and the like.

[1] https://github.com/mgenov/ipxe/commit/ea0686cf7cc0dec5e5fde16ee412f7e6392b7a9e

NiKiZe, 2021-05-10:
Looks like this was sent to the iPXE devel mailing list https://lists.ipxe.org/pipermail/ipxe-devel/2014-June/003589.html

Teknoenie, 2021-05-10:
Yes, this is what lead me to check to see if it ever did get merged into iPXE, but a quick search of the iPXE code didn't show anything having LLDP support so I thought I'd double check in case I missed something.
\

NiKiZe, 2021-05-10:
The lldp commit can probably be rebased easily(?) Not sure about the settings commit.

Teknoenie, 2021-05-11:
Is there an estimation as to when this could be done?

NiKiZe, 2021-05-11:
> Is there an estimation as to when this could be done?

Maybe there was a misunderstanding.
If you want to use the LLDP commit on top of current master, it has minimal conflicts and you should be able to rebase it (on your own)

If it is something that is urgent, you could [reach out and offer to pay](https://www.fensystems.co.uk/development/ebootfeatures.php) for the time to implement the feature.

Teknoenie, 2021-06-28:
I think i've been able to rebase the code to master but I'm not really sure what I'm doing as this is my first attempt at submitting a patch to a project like iPXE and I'm not by any stretch a strong git user/coder.

Since I've never done this before I thought it would make more sense to submit a text version of the patch rather than submit any MR/PR.  I tested

```
make
make bin-x86_64-efi/snponly.efi
make bin-x86_64-efi/ipxe.efi
```
all with and without `#define LLDP` in `src/config/general.h` and it seems to not have broken anything.  I have not yet tested that it works and is able to poll LLDP information from a switch yet.  Hopefully I'll get to that later today.

If this is acceptable, how should I go about properly submitting this patch?

[lldp-merge.txt](https://github.com/ipxe/ipxe/files/6724604/lldp-merge.txt)

JuniorJPDJ, 2021-07-08:
Probably making pull request with this change would be preferable (just guessing).
But first test if it works and allows you to poll LLDP info from switch ;D

steverweber, 2023-04-09:
is something pending here? The merge https://github.com/ipxe/ipxe/pull/413 seemed to have resolved the issue.

steverweber, 2023-04-09:
perhaps could use some wiki info on https://ipxe.org/docs

NiKiZe, 2023-04-09:
Should have yes, for now see the git commit for documentation.

closed 2023-04-09 by NiKiZe