ipxe/ipxe #359: building with gcc 11.x fails

eworm-de, 2021-05-18:
Building on current Arch Linux with gcc 11.x (`gcc 11.1.0-1`) fails:

```
  [DEPS] tests/bigint_test.c
  [BUILD] bin/bigint_test.o
In file included from tests/bigint_test.c:38:
tests/bigint_test.c: In function ‘bigint_test_exec’:
tests/bigint_test.c:232:14: error: ‘result_raw’ may be used uninitialized [-Werror=maybe-uninitialized]
  232 |         ok ( memcmp ( result_raw, expected_raw,                         \
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  233 |                       sizeof ( result_raw ) ) == 0 );                   \
      |                       ~~~~~~~~~~~~~~~~~~~~~~~
include/ipxe/test.h:45:19: note: in definition of macro ‘okx’
   45 |         test_ok ( success, file, line, #success )
      |                   ^~~~~~~
tests/bigint_test.c:232:9: note: in expansion of macro ‘ok’
  232 |         ok ( memcmp ( result_raw, expected_raw,                         \
      |         ^~
tests/bigint_test.c:565:9: note: in expansion of macro ‘bigint_add_ok’
  565 |         bigint_add_ok ( BIGINT ( 0x8a ),
      |         ^~~~~~~~~~~~~
In file included from tests/bigint_test.c:36:
include/string.h:32:19: note: by argument 1 of type ‘const void *’ to ‘memcmp’ declared here
   32 | extern int __pure memcmp ( const void *first, const void *second,
      |                   ^~~~~~
tests/bigint_test.c:213:17: note: ‘result_raw’ declared here
  213 |         uint8_t result_raw[ sizeof ( expected_raw ) ];                  \
      |                 ^~~~~~~~~~
tests/bigint_test.c:565:9: note: in expansion of macro ‘bigint_add_ok’
  565 |         bigint_add_ok ( BIGINT ( 0x8a ),
      |         ^~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:952: bin/bigint_test.o] Error 1
```

mcb30, 2021-05-18:
Looks as though Fedora 34 has gcc 11.  I'll update my main dev system later this week and hope to reproduce the problem then.

mcb30, 2021-06-03:
Reproduced as expected with Fedora 34

mcb30, 2021-06-03:
Looks as though it's a result of gcc 11 being more precise at handling asm constraints.  The constraints for `bigint_done()` don't explicitly state that the memory pointed to by `out` will be modified by the `__asm__` statement.

Patch in progress.

PR #373 (2021-06-03): [crypto] Add memory output constraints for big-integer inline assembly
The ARM versions of the big-integer inline assembly functions include
constraints to indicate that the output value is modified by the
assembly code.  These constraints are not present in the equivalent
code for the x86 versions.

As of GCC 11, this results in the compiler reporting that the output
values may be uninitialized.

Fix by including the relevant memory output constraints.

Reported-by: Christian Hesse <mail@eworm.de>

Fixes: #359
closed 2021-06-03 by mcb30