ipxe/ipxe #369: Unable to use VLANs on machine in UEFI-mode

skoef, 2021-06-01:
While trying to PXE boot a machine in UEFI mode with a VLAN tagged on the NIC, I run into an issue when creating a VLAN interface for it. Initially I used the patch from https://github.com/ipxe/ipxe/pull/111 but I run into the same problem with master as well. When I run `vcreate -t 100 net0` (`100` being the desired VLAN and `net0` being `${ifname}` and the NIC used for PXE-ing) I get this error:

```
iPXE> vcreate -t 100 net0
SNPDEV 0x50186408 could not install protocols: Error 0x711fe094 (http://ipxe.org/err/7f11e094)
VLAN net0-100 could not register: Error 0x711fe094 (http://ipxe.org/err/7f11e094)
Could not create VLAN device: Error 0x711fe094 (http://ipxe.org/err/7f11e094)
```

In this case I manually built `ipxe.efi` with `VLAN_CMD` enabled and `DEBUG=vlan,efi_snp` to produce the output above. The machine, a fresh Dell PowerEdge R640 with latest BIOS, iDRAC and Mellanox firmware, has 2 NICs with only VLAN enabled from UEFI network settings on the first:

```
net0: 0c:42:a1:9e:f3:54 using ConnectX-4Lx on 0000:18:00.0 (Ethernet) [closed]
  [Link:down TX:0 TXE:1 RX:0 RXE:0]
  [Link status: Unknown (http://ipxe.org/1a086194)
  [TXE: 1 x "Network unreachable (http://ipxe.org/28086090)"]
net1: 0c:42:a1:9e:f3:55 using ConnectX-4Lx on 0000:18:00.1 (Ethernet) [closed]
  [Link:down TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (http://ipxe.org/1a086194)
```

Creating a VLAN (so running `vcreate -t 100 net1`) works fine, but that NIC is not the I can/want to PXE boot on. The line responsible for the error is https://github.com/ipxe/ipxe/blob/master/src/interface/efi/efi_snp.c#L1722.

This is the only thing standing in the way for me PXE-booting these machines over UEFI, what could be the problem?

NiKiZe, 2021-06-01:
Can you create other vlans on that interface?
Have you tried booting without vlan in iPXE?
How is the vlans configured in firmware? (Tried without vlan in firmware?)
Could you maybe try snponly.efi and without vlan in iPXE? (Trying to have it inherit the interface from firmware, without need for additional setup)

skoef, 2021-06-01:
- I cán create other vlans on that interface. I also checked that perhaps I could create the right VLAN but with a different priority on that interface but that fails as well
- I can't boot PXE without vlan on that machine so I can't check whether iPXE would work or that I could create that particular VLAN on that interface when VLAN is disabled.
- The VLAN is configured like this:
[image]
- Using `snponly.efi` (albeit with VLAN) gives the exact same result.

In an ideal situation, shouldn't iPXE, when it detects a VLAN from UEFI, automatically add a net0-100 (in my case) with specifically registering that back to UEFI (since the device implicitly is already there)? I've tried to hack something like this into it, but my C-skills are not too great and it crashed right away.

NiKiZe, 2021-06-01:
There has probably not been a case where vlan exists prior to iPXE startup. Snponly will provide without vlan might just use it.
When iPXE starts it essentially pulls everything down, and the interface resets. (Exception being SNP/NII) so this might be a driver issue, that the reset is not complete.

skoef, 2021-06-02:
I've tried this on a different Dell server I had lying around, a PowerEdge M630 with Intel X520 NICs and latest available firmware. Gave me the exact same result:

ipxe.efi:
[image]

snponly.efi
[image]

so it is certainly not restricted to the Mellanox ConnectX-Lx NICs I was previously using. I'll try to boot a completely different type of server if I have the chance.

NiKiZe, 2021-06-02:
Did you ever try if snponly gives you the correct vlan, without trying to create it?

skoef, 2021-06-02:
Exact same result with an IBM (Lenovo) system: 

ipxe.efi:
[image]

snponly.efi:
[image]

@NiKiZe trying to get a DHCP lease directly on `net0` fails. That would mean that snponly doesn't transparently respects the given VLAN and goes with it.

skoef, 2021-06-04:
I've split up the UEFI `InstallMultipleProtocolInterfaces` calls to 1 protocol interface at a time and it breaks at installing the `efi_device_path_protocol_guid`. Perhaps that helps in debugging the issue?

mcb30, 2021-06-04:
The issue is almost certainly that the existing VLAN device (created by the platform firmware) is not being deleted when iPXE disconnects existing drivers.  UEFI enforces uniqueness of device paths and mandates the form that device paths take, so if something else has already created a handle with the device path for the VLAN then iPXE will not be able to do so.

Since you are able to reproduce this on a wide variety of platforms, this suggests it may be a fundamental issue with the EDK2 code that many platforms use as a basis for their UEFI implementations.  Are you able to reproduce the issue within a virtual machine running OVMF?

skoef, 2021-06-07:
I'm working on trying to get a working setup to reproduce this, but haven't succeeded so far. Do you have any suggestions how to build such a setup?

linusali, 2021-07-02:
a small pointer if it may help. I had the same issue just like what was described in the first post. But before attempting this failed PXE boot using VLAN, I have tried booting from another firmware that came with RHEL 8.4 built using commit 133f4c47 and it worked well. I know it is an old one, but it worked. Hope it may be of some help.

changdanyang, 2022-01-19:
I think it's the last commit that works fine in such a situation [eecb75ba](https://github.com/ipxe/ipxe/commit/eecb75ba)

skoef, 2022-01-19:
Yeah sorry I never got back to @linusali's comment there, I tried a build of the commit he mentioned but no dice. I'll try yours too @changdanyang soon. Any particular reason why this commit? Like, can you reproduce an issue with UEFI/VLAN booting in builds from later commits while it works for you in 133f4c4?

changdanyang, 2022-01-19:
@skoef I met the same case as yours and meanwhile we need a newer version to support new drivers for some other kind of servers in our landscape.  So I searched the right commit with the bisection method.
Yes, eecb75b is the last commit (based on my tests) that works fine with VLAN tagging.

NiKiZe, 2022-01-19:
Has anyone been able to reproduce this issue in qemu? Or is it all on Dell hardware?

skoef, 2022-01-25:
I can confirm creating VLANs still works in `eecb75b`.

NiKiZe, 2022-01-25:
So first failing commit is 27e886c67b63b9a381f87238ed7a856775199fd9 ?
As asked above is this reproduceable in QEMU?
To fix this a dev need to have a case that can be tested Unless @mcb30 can see something obvious in that commit that can be the issue.

skoef, 2022-01-25:
I just tried 27e886c and it works as well, I'll do a git bisect to determine which commit broke it

skoef, 2022-01-26:
Little nuance: while the vcreate-error started on c70b3e0 (last worked on fbb5989), actually getting a DHCP-lease from a VLAN-tagged interface through `ifconf` doesn't work in fbb5989. When I reverted to eecb75b it both worked. @changdanyang is that your experience as well?

jw14812, 2022-03-24:
I have observed the same behavior when running on a Dell system as well.  As an additional data point, I  disabled all PXE interfaces in the UEFI system settings->system BIOS->Network Settings, to remove any VLAN info.  I then booted to UEFI shell and executed the ipxe.efi from the shell.  After iPXE initialization, I dropped to the iPXE shell and attempted to create a VLAN (specifying the VLAN ID that is applicable to my peer server) using the 'vcreate' command but ran into the same error messages. Using a random VLAN ID it was able to complete successfully, as noted above. When I did a manual revert of the c70b3e0 commit( on top of the latest source) the 'vcreate' error went away.

urielrosen1981, 2022-05-11:
Hi ,

could you please share how I clone the commit c70b3e0 so I can use it ?
I am a newbie at git and would appreciate your help.

urielrosen1981, 2022-05-11:
thanks for your reply,
I think I still missing something here , am I supposed to git clone the entire code like this:
git clone https://github.com/ipxe/ipxe.git
?
and then use the command you wrote ?  
please tell me what I am doing wrong.

urielrosen1981, 2022-05-12:
I managed to pull the code ,
thanks.

jacobweinstock, 2022-11-29:
I'm experiencing this issue. I tried to patch top of tree with [this](https://github.com/tinkerbell/ipxedust/blob/3557d370e4cb714dedd8244a2b230f44ab7e72f8/binary/script/vcreate.patch). It ended up not working across all vendors. Reverting to https://github.com/ipxe/ipxe/commit/eecb75ba4809bd5d3e6d63413a45c5fccbda2bc2, did work for me. Unfortunately, that commit is > 2 years old and it does not contain [Chainloaded DHCP options for UEFI](https://github.com/ipxe/ipxe/pull/265) which I require for my [use case](https://github.com/tinkerbell/proposals/blob/main/proposals/0030/README.md) in the Tinkerbell stack. 

As done in this [comment](https://github.com/ipxe/ipxe/issues/369#issuecomment-855825951), I am also able to boot an ipxe.iso with qemu but not able to simulate the configured VLAN of a nic. I've tried tagged interfaces on a bridged network, tap interfaces, etc. Anyone know how to reproduce this. by chance?

example qemu command with tap interface.
```bash
qemu-system-x86_64 -bios /usr/share/ovmf/OVMF.fd -cdrom ipxe.iso -m 512 -nographic -device e1000,netdev=net1,mac=52:54:00:22:a1:66 -netdev tap,id=net1,ifname=tap0,script=no,downscript=no
```

Would any iPXE devs possibly be available to look into this? I can provide hardware to test this if that would help in any way. 

Thanks for all your work on this great project!
 
CC @mcb30, @NiKiZe

stappersg, 2022-12-09:
FYI: mcb30 linked a pull request 8 minutes ago that will close this issue (upon merge).

PR #838 (2022-12-14): EFI VLAN support
UEFI implements VLAN support within the Managed Network Protocol (MNP)
driver, which may create child VLAN devices automatically based on
stored UEFI variables.  These child devices do not themselves provide
a raw-packet interface via EFI_SIMPLE_NETWORK_PROTOCOL, and may be
consumed only via the EFI_MANAGED_NETWORK_PROTOCOL interface.
    
The device paths constructed for these child devices may conflict with
those for the EFI_SIMPLE_NETWORK_PROTOCOL instances that iPXE attempts
to install for its own VLAN devices.  The upshot is that creating an
iPXE VLAN device (e.g. via the [... cut]
closed 2022-12-14 by mcb30