ipxe/ipxe #377: USB rf-rndis driver fails DHCP on Acer B3-A40

mcb30, 2021-06-05:
Splitting from #354:

iPXE fails to obtain an address via DHCP on an Acer tablet (kindly provided by @wheelnut02) running Android 7.0 (kernel 4.4.22):
```
iPXE 1.21.1+ (g065d) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP AoE ELF MBOOT PXE bzImage Menu PXEXT
net0: 06:16:26:36:46:56 using rf-rndis on 0000:00:04.0-5-1.0 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
iPXE> dhcp
Configuring (net0 06:16:26:36:46:56)................. No configuration methods succeeded (http://ipxe.org/040ee119)
```

mcb30, 2021-06-05:
The same device works without problem using the Linux `rndis_host` driver

mcb30, 2021-06-05:
Packet capture showing the USB control endpoint transactions performed when using the Linux `rndis_host` driver: [rndis.pcapng.gz](https://github.com/ipxe/ipxe/files/6603250/rndis.pcapng.gz)

(Captured using Wireshark and `usbmon`)

mcb30, 2021-06-06:
Didn't find any substantial differences in any of the control endpoint transactions.

As with the traces recorded by @wheelnut02, I occasionally observed packets being received by iPXE, which suggested that it was only the transmit datapath that was not working.

Looking at the bulk OUT packets, I noticed that Linux includes four additional bytes of zero padding after the RNDIS header (before the actual packet content).  Adding an extra four bytes of padding to our RNDIS header (and updating the "data offset" field to match) caused everything to start working!

I'll need to dig through the RNDIS specs to figure out how this discrepancy arose.  There's no harm in always adding the extra four bytes of padding (since the location of the data is described by an "offset" field anyway).  Will need to check to see if the receive datapath (which uses the same data structure) needs to be able to handle packets that don't include the additional padding.

mcb30, 2021-06-06:
Looks as though this is an error in the iPXE RNDIS implementation dating back to when it was first implemented in 2014.  The RNDIS specification includes a 4-byte "Reserved" field in most data structures, but in `REMOTE_NDIS_INITIALIZE_CMPLT` and `REMOTE_NDIS_PACKET_MSG` it's an 8-byte field instead.

wheelnut02, 2021-06-06:
Wow - it did not take you long to fathom that out. I am really impressed. Out of interest, what would happen if the REMOTE_NDIS_INITIALIZE_CMPLT and REMOTE_NDIS_PACKET_MSG were reduced to 4 bytes ? Would it break the existing implementations ?

PR #378 (2021-06-07): [rndis] Fix size of reserved fields
Most RNDIS data structures include a trailing 4-byte reserved field.
For the REMOTE_NDIS_PACKET_MSG and REMOTE_NDIS_INITIALIZE_CMPLT
structures, this is an 8-byte field instead.

iPXE currently uses incorrect structure definitions with a 4-byte
reserved field in all data structures, resulting in data payloads that
overlap the last 4 bytes of the 8-byte reserved field.

RNDIS uses explicit offsets to locate any data payloads beyond the
message header, and so liberal RNDIS parsers (such as those used in
Hyper-V and in the Linux USB Ethernet gadget driver) are  [... cut]
closed 2021-06-07 by mcb30