ipxe/ipxe #381: Freeze: iPXE initialising devices...

nathanshearer, 2021-06-08:
When booting iPXE and it always freezes at "iPXE initialising devices" for several minutes then the system reboots.

This bug is reproducible with the latest master copy of iPXE and with iPXE 1.21.1 on a MacBook Pro 9,1.
This does not happen with iPXE 1.20.1.

I would normally just downgrade iPXE and continue, however 1.20.1 has other bugs that are fixed in the newer versions.

nathanshearer, 2021-06-09:
How can I enable debugging for everything when I build iPXE?

The documentation indicates I need to specify each item that should have debugging enabled, however I do not know which component is freezing. It may be verbose, but at least I'll get some kind of output that may narrow down the problem.

NiKiZe, 2021-06-09:
Yes you need to specify every item, enabling everything does not make sense.

Since you have known working commit and a non working commit the best thing is probably git bisect https://ipxe.org/howto/bisect

nathanshearer, 2021-06-10:
I have started the bisect, however older versions of iPXE cannot be built:

```
make bin-x86_64-efi/ipxe.efi
...
  [BUILD] bin-x86_64-efi/flexboot_nodnic.o
drivers/infiniband/flexboot_nodnic.c: In function ‘flexboot_nodnic_create_qp’:
drivers/infiniband/flexboot_nodnic.c:368:53: error: implicit conversion from ‘enum ib_queue_pair_type’ to ‘nodnic_queue_pair_type’ [-Werror=enum-conversion]
  368 |  status = nodnic_port_create_qp(&port->port_priv, qp->type,
      |                                                   ~~^~~~~~
drivers/infiniband/flexboot_nodnic.c: In function ‘flexboot_nodnic_destroy_qp’:
drivers/infiniband/flexboot_nodnic.c:409:45: error: implicit conversion from ‘enum ib_queue_pair_type’ to ‘nodnic_queue_pair_type’ [-Werror=enum-conversion]
  409 |  nodnic_port_destroy_qp(&port->port_priv, qp->type,
      |                                           ~~^~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:946: bin-x86_64-efi/flexboot_nodnic.o] Error 1
```

mcb30, 2021-06-11:
You can force older versions to build by using `make bin-x86_64-efi/ipxe.efi NO_WERROR=1`

stappersg, 2023-02-27:
Recently on _IRC_
```text
21:18 < redacted> I'm trying to netboot a new modern desktop board, asus z790m plus d4, with an intel 1gb ethernet port.  I 
             am using if option client-arch != 00:00 { filename "ipxe.efi";
21:19 < redacted> in my dhcp config to send ipxe.efi, otherwise undionly.kpxe.  my new motherboard shows its getting 
             ipxe.efi.  but it just stops after iPXE initialising devices...
21:19 < redacted> how can I get some debug output from it to see what's wrong ?
21:21 < stappers> make DEBUG=pci
21:25 < redacted> weird.  no change other than the number of bytes transferred for ipxe.efi
21:25 < redacted> maybe theres a debug=all?
21:28 < redacted> trying DEBUG=pci,buffer:2,heap, as seen at dox.ipxe.org
21:30 < redacted> make[1]: *** No rule to make target `bin/buffer.dbg2.o', needed by `bin/blib.a'.  Stop.
21:31 < redacted> buffer, buffer:2, and heap all cause compile errors
21:33 < redacted> chatgpt said I could use DEBUG=net?
21:37 < redacted> adding -g to CFLAGS int he make file seems to let it compile with DEBUG=all
21:38 < redacted> the nic uses e1000e driver in ubutuntu, intel form 0dc8.  asustek computer device 8672
21:39 < redacted> well darn the compile filed with make[1]: *** No rule to make target `bin/all.dbg1.o', needed by 
             `bin/blib.a'.  Stop.
21:42 < stappers> https://ipxe.org/download#debug_builds
```

Point that I want to make:

Please update wiki with _start debugging with `make DEBUG=pci,efi_pci`_ or whatever correct `.c` to start with.

mcb30, 2023-02-27:
> Recently on _IRC_
> 
> ```
> 21:18 < redacted> I'm trying to netboot a new modern desktop board, asus z790m plus d4, with an intel 1gb ethernet port.  I 
>              am using if option client-arch != 00:00 { filename "ipxe.efi";
> 21:19 < redacted> in my dhcp config to send ipxe.efi, otherwise undionly.kpxe.  my new motherboard shows its getting 
>              ipxe.efi.  but it just stops after iPXE initialising devices...
> 21:19 < redacted> how can I get some debug output from it to see what's wrong ?
> 21:21 < stappers> make DEBUG=pci
> 21:25 < redacted> weird.  no change other than the number of bytes transferred for ipxe.efi
> 21:25 < redacted> maybe theres a debug=all?
> 21:28 < redacted> trying DEBUG=pci,buffer:2,heap, as seen at dox.ipxe.org
> 21:30 < redacted> make[1]: *** No rule to make target `bin/buffer.dbg2.o', needed by `bin/blib.a'.  Stop.
> 21:31 < redacted> buffer, buffer:2, and heap all cause compile errors
> 21:33 < redacted> chatgpt said I could use DEBUG=net?
> 21:37 < redacted> adding -g to CFLAGS int he make file seems to let it compile with DEBUG=all
> 21:38 < redacted> the nic uses e1000e driver in ubutuntu, intel form 0dc8.  asustek computer device 8672
> 21:39 < redacted> well darn the compile filed with make[1]: *** No rule to make target `bin/all.dbg1.o', needed by 
>              `bin/blib.a'.  Stop.
> 21:42 < stappers> https://ipxe.org/download#debug_builds
> ```
> 
> Point that I want to make:
> 
> Please update wiki with _start debugging with `make DEBUG=pci,efi_pci`_ or whatever correct `.c` to start with.

Whoever "redacted" is in that conversation, I don't think they were reading any relevant documentation to start with.

NiKiZe, 2023-02-28:
> Whoever "redacted" is in that conversation, I don't think they were reading any relevant documentation to start with.

In that case most likely they don't even need to debug anything, they just had to look at their tftp server logs to see iPXE looping and break that loop.

lonlundgren, 2026-08-18:
We hit this on a MacBookPro10,1 (early 2013 retina) with current master and root-caused it with on-screen marker instrumentation: timer events create, arm, and fire correctly, and console CheckEvent is never reached — the wedge is `bs->RestoreTPL ( TPL_APPLICATION )` itself, which never returns once startup has completed on this firmware. That explains the whole fingerprint: the single poll loop dies inside the drop, so the prompt freezes and the keyboard goes dead together.

Working fix, field-tested across dozens of boots on the 10,1 and clean under QEMU/OVMF: https://github.com/lonlundgren/ipxe/tree/apple-efi-quirks — `[efi]` floor TPL drops at `TPL_CALLBACK` and move the tick event to `TPL_NOTIFY` (the hang fix), plus two quality-of-life commits the same machines need: `[efifb]` a built-in 8x16 font fallback (Apple EFI has GOP but no HII font protocol, so the framebuffer console otherwise cannot start) and `[fbcon]` glyph scaling for native-only high-DPI GOP modes.

Caveat: with the floor in place the tick-jitter entropy source is unusable, so TLS needs another entropy source — RDRAND covers every 2012+ Mac. PR opened with the three commits.