ipxe/ipxe #385: ipxe and Cisco UCS C220 M5SX

glenc2004, 2021-06-10:
Hello.

   I'm wondering if anyone has had issues using ipxe boot on a Cisco UCS C220 M5SX with Cisco UCS VIC 1457 nics. During the initial bootup I have the users provide IP address, hostname, netmask and gateway. Once those are entered I then enable the network interface and run the chain command to get to the 2nd stage of the configuration process. But when the chain command is executed, the network just stops. Here is how I activate the network inside the built-in menu in the iso.

set net0/$ipaddress
set net0/$netmask
set net0/$gateway
ifopen net0
chain --timeout 60000 http://$ks/ks/html/menu.ipxe

When the ISO is booted it does see the Cisco network interface and I ran a ifstat before and after and the network does definitely get disabled when the chain command is run.

How can I diagnose if this is a driver issue or something else. I was able to fully load an OS (Redhat Linux 7) from DVD and I had complete network access so I know at least from that standpoint it's probably not an infrastructure issue within UCS.

Thanks!

Glen Collins

glenc2004, 2021-06-10:
By the way I forgot to mention these are new servers and my old servers with MLOM 1225 have no issues.

Also I downloaded the latest version as of yesterday 6/9/2021 and have been using that version for this issue.

NiKiZe, 2021-06-12:
first, what does ifstat show, and what is the link connected to? (can it be that the link switches state after a few packets?)

stappersg, 2022-03-07:
iPXE  IRC channel, 2022-03-03,  timezone  UTC+1
```text
14:59 < redacted> Hey @stappers - The machine has 4 NICs, but
         only one is connected. iPXE was initiated as part of the PXE
         process. To give some context -  I am trying to use ipxe on a
         Cisco UCS C220-M5 machine. It uses NIC model- UCS VIC 1457. What
         ever I do I cannot get networking up using IPv6. IPv4 works like
         charm. I tried to use both SLAAC and Stateful DHCPv6
```
It says that IPv4 is working, but  IPv6 not.

I hope this helps @glenc2004

NiKiZe, 2022-07-04:
Not much data to go on and no feedback, closing. (We can reopen if more data is provided)

closed 2022-07-04 by NiKiZe