ipxe/ipxe #386: Redfish Host Interface disabled

sayhikevin, 2021-06-11:
We learned that if we disabled Redfish Host Interface within the iPXE, we're getting the following error when loading the IPXE.
We're not ready to use the Redfish Host Interface yet and still depend on the in-band KCS interface for initial BMC provisioning. 
So we globally disable the Redfish Host Interface on all of our platforms for security purposes.  Any idea how to fix this while keep the Redfish host interface disabled?  

CPU Apic ID 0 report error.
iPXE 1.21.1+ (gd7bc9) -- Open Source Network Boot Firmware -- http://ipxe.org [ipxe.org]
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu
iPXE&gt; !!!! X64 Exception Type - 0D(#GP - General Protection) CPU Apic ID - 00000000 !!!!
ExceptionData - 0000000000000000
RIP - 0000000530646870, CS - 0000000000000038, RFLAGS - 0000000000010202
RAX - 8000000000000006, RCX - 000000005B9D3560, RDX - 0000000000000004
RBX - 0000000000000004, RSP - 000000006C864448, RBP - 000000005D435F98
RSI - 0000000000000058, RDI - 000000005B9D3560
R8 - 000000005D435F98, R9 - 0000000530646870, R10 - 000000005B9D3580
R11 - 00000000643E8E40, R12 - 000000005A5B29C0, R13 - 0000000000000004
R14 - 0000000000000004, R15 - 0000000000000000
DS - 0000000000000030, ES - 0000000000000030, FS - 0000000000000030
GS - 0000000000000030, SS - 0000000000000030
CR0 - 0000000080010013, CR2 - 0000000000000000, CR3 - 000000006C401000
CR4 - 0000000000000668, CR8 - 0000000000000000
DR0 - 0000000000000000, DR1 - 0000000000000000, DR2 - 0000000000000000
DR3 - 0000000000000000, DR6 - 00000000FFFF0FF0, DR7 - 0000000000000400
GDTR - 000000006BC41698 0000000000000047, LDTR - 0000000000000000
IDTR - 000000005FE53018 0000000000000FFF, TR - 0000000000000000
FXSAVE_STATE - 000000006C8640A0
!!!! Can't find image information. !!!!

woa7, 2021-06-11:
can you provide some info about the hardware and your selected ipxe compile options?

mcb30, 2021-06-14:
> We learned that if we disabled Redfish Host Interface within the iPXE, we're getting the following error when loading the IPXE.
> We're not ready to use the Redfish Host Interface yet and still depend [...]

I'm sorry, I don't understand what you're saying.  What do you mean by "we disabled Redfish Host Interface within the iPXE"?

mcb30, 2021-06-22:
Got some further information on this via IRC.  The "Redfish Host Interface" is not a network interface of any kind and is not in any way visible to iPXE: it's just a communication channel that the Supermicro BMC may make use of for communication with the host.  It can be disabled without disabling the BMC entirely, since the KCS is also present as an alternative (older) BMC-to-host interface.

I've left it with Supermicro to arrange for access to reproduce the issue (either via remote KVM or by shipping a physical system to me in the UK).