ipxe/ipxe #389: PC ITAUTEC - very slow download

killo2005, 2021-06-16:
An ITAUTEC model ST4265 PC with Realtek NIC (model below) has problems reported below.
Starting in BIOS mode (ipxe.pxe and undionly.kpxe) the download is extremely slow.
Starting in UEFI mode (ipxe.efi), the PC goes into an infinite loop.
After several tests suggested by the developers in web.libera.chat, patch was done in ipxe/src/interface/efi/efi_veto.c and the loop was fixed, persisting the slowness to make the downloads

Testing was done with offboard NIC (Intel 100) and it was found that the slowness problem continues. It was observed that when the PC is restarted with Ctrl + Alt + Del the download does not pass 1 Mbits; when it is turned off and on by the power button, the download speed reaches the maximum of the NIC (note: this test is with the NIC offboard). With the onboard NIC (realtek) the download speed is around 1 Mbits on average

Below hardware data

03:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 04)

------------------------------------------------------------------------------------------------
dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 2.7 present.

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: Copyright Itautec S.A. ST 4265
	Version: F7s BI-SL2
	Release Date: 05/15/2013
	Address: 0xF0000
	Runtime Size: 64 kB
	ROM Size: 8 MB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		BIOS ROM is socketed
		EDD is supported
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		Print screen service is supported (int 5h)
		8042 keyboard services are supported (int 9h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		ACPI is supported
		USB legacy is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 4.6
	
------------------------------------------------------------------------------------------------	
dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 2.7 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Itautec S.A.
	Product Name: ST 4265
	Version: To be filled by O.E.M.
	Serial Number: To be filled by O.E.M.
	UUID: 03400200-04a7-052d-f706-fe0700080009
	Wake-up Type: Power Switch
	SKU Number: To be filled by O.E.M.
	Family: To be filled by O.E.M.

NiKiZe, 2021-07-13:
You have verified 2 different NICs and have same behavior correct?

Slow download speed after reboot
But normal speed after power of and power on

and this is with all of ipxe.pxe, undionly.kpxe, and ipxe.efi (after patch)
did you try snponly.efi as well, or does that fail?

Have you tried firmware upgrade if there is any?
One thing to try and verify is if your NICs can properly use interrupts, or if they have to fall back to polling.

Also try to rule out the link itself, maybe the switch port goes into some weird state, highly unlikely, but better to do the simple tests first.

NiKiZe, 2022-02-07:
Closing due to no feedback on asked questions.

closed 2022-02-07 by NiKiZe