ipxe/ipxe #390: Failure to bring the Realtek RTL-8110SC/8169 interface up

vl-80, 2021-06-17:
I encountered an issue where iPXE fails to bring the Realtek `RTL-8110SC/8169` NIC controller up.

The motherboard is `ZOTAC IONITX-S-E`.

The problem is that when the OS (Linux, systemd) is gracefully shutting down or rebooting it brings the network interface down. At the very end of the shutdown process the kernel reports:

    r8169: 000:03:0b.0 enp3s11: Link is Down

and the LEDs on the card turn off.

If at this point we boot iPXE (from a USB drive) it will be unable to bring the interface up.

With the debug turned on the output on the screen is:

    REALTEK 0xe04bc appears to be an RTL8169
    MII 0xe54c restarted autonegotiation
    MII 0xe54c reset after 0 ms
    MII 0xe54c restarted autonegotiation
    MII 0xe54c registers 00-07: 1000 7949 001c c912 01e1 0000 0004 2001
    MII 0xe54c registers 08-0f: 0000 0300 0000 0000 1007 f880 0200 3000
    REALTEK 0x04bc PHY status is 04 (GMII, 10Mbps, LinkDown, HalfDuplex)

    (embedded script is calling ifopen)

    REALTEK 0x04bc ring 20 is at [bfdfa800,bfdfa840)
    REALTEK 0x04bc ring e4 is at [bfdfa900,bfdfa940)
    MII 0xe54c registers 00-07: 1000 7949 001c c912 0141 0000 0004 2001
    MII 0xe54c registers 08-0f: 0000 0300 0000 0000 1007 f880 0200 3000
    REALTEK 0x04bc PHY status is 04 (GMII, 10Mbps, LinkDown, HalfDuplex)

The only way to recover from this is to power cycle the system - physically unplug the power cord.

`CTRL + ALT + DELETE`, reset button, power button do not help. Only physically unplugging the power works.

At the same time the built-in realtek PXE boot manager suffers from the same issue. It also fails with "Media test failure". But if Linux OS is booted from a USB drive it brings the interface up.

So, there is something that Linux kernel does that brings the interface back up, but I have failed to pin point it.

What I did discover is that kernel ends up calling `genphy_suspend()` and it sets the interface down. When I commented out this function the interface was remaining working between reboots and iPXE worked just fine.

https://elixir.bootlin.com/linux/v5.4.109/source/drivers/net/phy/phy_device.c#L2000

```c
int genphy_suspend(struct phy_device *phydev)
{
	return phy_set_bits(phydev, MII_BMCR, BMCR_PDOWN);
}
EXPORT_SYMBOL(genphy_suspend);
```

The "opposite" function genphy_resume() in the kernel simply clears the `MBCR_DOWN` bit:

```c
int genphy_resume(struct phy_device *phydev)
{
	return phy_clear_bits(phydev, MII_BMCR, BMCR_PDOWN);
}
EXPORT_SYMBOL(genphy_resume);
```

I tried simulating these actions in iPXE, but I was unable to solve the problem.

So, apparently Linux does something else to bring the interface back to life and it would be great if we can make iPXE do the same.

mcb30, 2021-06-17:
Fantastic issue report: thank you very much for the level of detail and testing that you have done!

iPXE does already clear the power-down bit, as part of a combined reset operation: https://github.com/ipxe/ipxe/blob/51c88a4a62116d77a854b55100c6955d6f1eb6dc/src/drivers/net/mii.c#L79-L81

It's possible that some PHYs require these register writes to be separated, e.g. by clearing the power-down bit before initiating a reset.  For example (untested):
```diff
diff --git a/src/drivers/net/mii.c b/src/drivers/net/mii.c
index 87605f0cb..41c4bef21 100644
--- a/src/drivers/net/mii.c
+++ b/src/drivers/net/mii.c
@@ -76,7 +76,14 @@ int mii_reset ( struct mii_device *mii ) {
        int bmcr;
        int rc;
 
-       /* Power-up, enable autonegotiation and initiate reset */
+       /* Power up PHY and enable autonegotiation */
+       if ( ( rc = mii_write ( mii, MII_BMCR, BMCR_ANENABLE ) ) != 0 ) {
+               DBGC ( mii, "MII %p could not write BMCR: %s\n",
+                      mii, strerror ( rc ) );
+               return rc;
+       }
+
+       /* Initiate reset */
        if ( ( rc = mii_write ( mii, MII_BMCR,
                                ( BMCR_RESET | BMCR_ANENABLE ) ) ) != 0 ) {
                DBGC ( mii, "MII %p could not write BMCR: %s\n",
```

Could you try the above patch (and possibly some variations on it) to see if anything seems to solve the issue on your system?

Thanks!

vl-80, 2021-06-17:
This patch did not solve it.

I also tried

```c
mii_write ( mii, MII_BMCR, 0 )
```
with no success.

Then, I modified the `mii_reset` to check value of BMCR register at various points during the reset:

```c
/**
 * Reset MII device
 *
 * @v mii                MII device
 * @ret rc                Return status code
 */
int mii_reset ( struct mii_device *mii ) {
        unsigned int i;
        int bmcr;
        int rc;

        bmcr = mii_read ( mii, MII_BMCR );

        /* Read PHY status */
        if ( bmcr < 0 ) {
                rc = bmcr;
                DBGC ( mii, "MII %p could not read BMCR: %s\n",
                       mii, strerror ( bmcr ) );
                return rc;
        }
        else
        {
                DBGC ( mii, "MII %p BMCR: 0x%08x\n", mii, bmcr );
        }

        mdelay( 25 );

        /* Power-down PHY */
        if ( ( rc = mii_write ( mii, MII_BMCR, BMCR_PDOWN ) ) != 0 ) {
                DBGC ( mii, "MII %p could not write BMCR: %s\n",
                       mii, strerror ( rc ) );
                return rc;
        }
        else
        {
                DBGC ( mii, "MII %p Power Down PHY\n", mii );

                mdelay( 100 );

                bmcr = mii_read ( mii, MII_BMCR );

                if ( bmcr < 0 ) {
                        rc = bmcr;
                        DBGC ( mii, "MII %p could not read BMCR: %s\n",
                               mii, strerror ( rc ) );
                        return rc;
                }
                else
                {
                        DBGC ( mii, "MII %p BMCR: 0x%08x\n", mii, bmcr );
                }
        }

        mdelay( 25 );

        /* Power-up PHY */
        if ( ( rc = mii_write ( mii, MII_BMCR, 0 ) ) != 0 ) {
                DBGC ( mii, "MII %p could not write BMCR: %s\n",
                       mii, strerror ( rc ) );
                return rc;
        }
        else
        {
                DBGC ( mii, "MII %p Power-up PHY\n", mii );

                mdelay( 100 );

                bmcr = mii_read ( mii, MII_BMCR );

                if ( bmcr < 0 ) {
                        rc = bmcr;
                        DBGC ( mii, "MII %p could not read BMCR: %s\n",
                               mii, strerror ( rc ) );
                        return rc;
                }
                else
                {
                        DBGC ( mii, "MII %p BMCR: 0x%08x\n", mii, bmcr );
                }
        }

        mdelay( 25 );

        /* Initiate reset */
        if ( ( rc = mii_write ( mii, MII_BMCR,
                                ( BMCR_RESET | BMCR_ANENABLE ) ) ) != 0 ) {
                DBGC ( mii, "MII %p could not write BMCR: %s\n",
                       mii, strerror ( rc ) );
                return rc;
        }
        else
        {
                DBGC ( mii, "MII %p Reset\n", mii );

                mdelay( 100 );

                bmcr = mii_read ( mii, MII_BMCR );

                if ( bmcr < 0 ) {
                        rc = bmcr;
                        DBGC ( mii, "MII %p could not read BMCR: %s\n",
                               mii, strerror ( rc ) );
                        return rc;
                }
                else
                {
                        DBGC ( mii, "MII %p BMCR: 0x%08x\n", mii, bmcr );
                }
        }

        /* Wait for reset to complete */
        for ( i = 0 ; i < MII_RESET_MAX_WAIT_MS ; i++ ) {

                /* Check if reset has completed */
                bmcr = mii_read ( mii, MII_BMCR );
                if ( bmcr < 0 ) {
                        rc = bmcr;
                        DBGC ( mii, "MII %p could not read BMCR: %s\n",
                               mii, strerror ( rc ) );
                        return rc;
                }

                /* If reset is not complete, delay 1ms and retry */
                if ( bmcr & BMCR_RESET ) {
                        mdelay ( 1 );
                        continue;
                }

                /* Force autonegotation on again, in case it was
                 * cleared by the reset.
                 */
                if ( ( rc = mii_restart ( mii ) ) != 0 )
                        return rc;

                DBGC ( mii, "MII %p reset after %dms\n", mii, i );
                return 0;
        }

        DBGC ( mii, "MII %p timed out waiting for reset\n", mii );
        return -ETIMEDOUT;
}
```

The output I got was:
```

MII 0xe004c BMCR: 0x00001000
MII 0xe004c Power Down PHY
MII 0xe004c BMCR: 0x00000800
MII 0xe004c Power-up PHY
MII 0xe004c BMCR: 0x00000000
MII 0xe004c Reset
MII 0xe004c BMCR: 0x00001000
MII 0xe004c Restarted autonegatioation
MII 0xe004c reset after 0 ms
MII 0xe004c Restarted autonegatioation
```

So, immediately on boot, the BMCR is `0x00001000`. The power down bit is not set and autonegatioation is enabled.

I tried explicitly setting and clearing the BMCR_PDOWN bit and while the value of the bit changes, the NIC does not come alive.

vl-80, 2021-06-17:
More debug data.

State of some registers during problem and normal state:

```
Register    Problem    Normal
0x3e        0x00       0x20        <---- Difference
0x50        0x00       0x00
0x51        0x05       0x05
0x52        0x1f       0x1f
0x53        0x10       0x10
0x54        0xa1       0xa1
0x55        0x00       0x00
0x56        0x03       0x03
0x6c        0x04       0x0b        <---- Difference
0xe0        0x0b       0x0b
```

Phy registers:
```
Register    Problem    Normal
0x00        0x1000     0x1000
0x01        0x7949     0x796d      <---- Difference
0x04        0x01e1     0x01e1
0x05        0x0000     0x01e1      <---- Difference
0x09        0x0300     0x0300
0x0a        0x0000     0x0000
0x0d        0xf880     0xf880
0x0f        0x3000     0x3000
0x12        0x0000     0x0000
0x13        0x0000     0x3c40      <---- Difference
0x1c        0x846d     0x846c      <---- Difference
```

I do not know what the register `0x1c` defined in kernel as `MII_NCONFIG` is responsible for. The values of this register for problem and normal states differ in bit `0`.

PR #397 (2021-06-24): [realtek] Work around hardware bug on RTL8211B
The RTL8211B seems to have a bug that prevents the link from coming up
unless the MII_MMD_DATA register is cleared.

The Linux kernel driver applies this workaround (in rtl8211b_resume())
only to the specific RTL8211B PHY model, along with a matching
workaround to set bit 9 of MII_MMD_DATA when suspending the PHY.
Since we have no need to ever suspend the PHY, and since writing a
zero ought to be harmless, we just clear the register unconditionally.

Debugged-by: VL-80 <vl-80@users.noreply.github.com>

Fixes: #390 
Fixes: #394
closed 2021-06-24 by mcb30