ipxe/ipxe #395: Just a question: why ipxe sends another DHCP request to get various DHCP options instead of getting them directly when invoked by PXE firmware which passing the DHCP data in a specific memory area?

jjqq2013, 2021-06-20:
I found that in the pxelinux (at 6.0), the whole boot process just need a single DHCP request conceptually, 
how does the pxelinux get the DHCP options:
> pxelinux.configfile code 209, 
pxelinux.pathprefix code 210
I've checked the source code of pxelinux, turns out, the PXE firmware passes the DHCP response in a specific memory area then invoke the pxelinux binary, therefore, pxelinux does not send another DHCP to get DHCP options.

When I check iPXE's DHCP configuration, I find the first time it loads undionly.kpxe,  then undionly.kpxe sends another DHCP request to get the DHCP option "filename" again, of course result in another value,
```
  option client-architecture code 93 = unsigned integer 16;
  if exists user-class and option user-class = "iPXE" {
      filename "http://my.web.server/real_boot_script.php";
  } elsif option client-architecture = 00:00 {
      filename "undionly.kpxe";
  } else {
      filename "ipxe.efi";
  }
```

So my question is, the undionly.kpxe has the chance to get the original DHCP response from specific memory, with another DHCP option say "ipxe.filename" defined at the beginning, but it did not, is there any reason not do this?

NiKiZe, 2021-06-20:
You have to do another query to get all options, and also some might be different.

If you don't want this you can have embedded script with ifopen net0 which will use tha cached offer.

jjqq2013, 2021-06-21:
@NiKiZe  thank you. but I still don't get the reason why undionly.kpxe send another DHCP request instead of following pxelinux-like flow:

1.  define a separate dhcp option such as 209 as ipxe.configfile, 
2. PXE firmware sends DHCP request, gets DHCP response, with 3 important dhcp options, not only just two.
```
next-server      tftp-server
filename       undionly.kpxe
ipxe.configfile       http://a.b.c/ipxe_script   ####### note this  ########
```
3. PXE firmware download undionly.kpxe from tftp server
4. PXE firmware load undionly.kpxe, passing the DHCP response to it.
5. undionly.kpxe gets dhcp option ipxe.configfile without sending another DHCP request
6. undionly.kpxe download ipxe.configfile(http://a.b.c/ipxe_script)

 let me wait for several days to see if anyone know this.

NiKiZe, 2021-06-21:
So why does any is do a separate DHCP request when it can just load it from memory?

One reason is to get all options, don't remember the details, but a bootp packet are not required to have gateway or DNS, since it was only designed to work on LAN and not wan.

jjqq2013, 2021-06-21:
>So why does any is do a separate DHCP request when it can just load it from memory?
yes.

OK, seems there were some considerations when implementing iPXE, maybe it is because some PXE firmware does not pass the DHCP response to undionly.kpxe at all. Just curious. Not a big deal.

NiKiZe, 2021-07-09:
Again, not all needed DHCP options are always available in BOOTP packets, many DHCP servers sends them anyway, but are not required to send more then requested.
Also you might want to send different options top iPXE than what you did to pxe client.

As I wrote, if you want the cached packet, do use ifopen net0 which will load and use the cached packet. (pcbios and partial efi support atm.)
this might be a legacy thing from gPXE, but it works nicely, also see https://ipxe.org/howto/chainloading#breaking_the_infinite_loop

closed 2021-07-09 by NiKiZe