ipxe/ipxe #398: kernel for i386 and arm/arm64 is missing at http://boot.ipxe.org/demo

woa7, 2021-06-23:
```
qemu-system-i386 -cpu base -accel kvm \
-netdev user,id=net0,tftp=$HOME/configs/tftp/,bootfile=http://boot.ipxe.org/demo/boot.php \
-device virtio-net-pci,netdev=net0 -object rng-random,id=virtio-rng0,filename=/dev/urandom \
-device virtio-rng-pci,rng=virtio-rng0,id=rng0,bus=pci.0,addr=0x9 -boot n -nographic \
-monitor telnet::45454,server,nowait -serial mon:stdio | tee /tmp/out2.txt

```

```
SeaBIOS (version 1.14.0-2)

iPXE (http://ipxe.org) 00:03.0 CA00 PCI2.10 PnP PMM+07F8F540+07ECF540 CA00
                                                                               

Booting from ROM...
iPXE (PCI 00:03.0) starting execution...ok
iPXE initialising devices...ok

iPXE 1.0.0+git-20190125.36a4c85-5.1 -- Open Source Network Boot Firmware -- http
://ipxe.org
Features: DNS HTTP iSCSI NFS TFTP AoE ELF MBOOT PXE bzImage Menu PXEXT

net0: 52:54:00:12:34:56 using virtio-net on 0000:00:03.0 (open)
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
Configuring (net0 52:54:00:12:34:56)...... ok
net0: 10.0.2.15/255.255.255.0 gw 10.0.2.2
net0: fec0::5054:ff:fe12:3456/64 gw fe80::2
net0: fe80::5054:ff:fe12:3456/64
Next server: 10.0.2.2
Filename: http://boot.ipxe.org/demo/boot.php
http://boot.ipxe.org/demo/boot.php... ok
boot.php : 127 bytes [script]
vmlinuz-3.16.0-rc4... ok
initrd.img... ok
This kernel requires an x86-64 CPU, but only detected an i486 CPU.
Unable to boot - please use a kernel appropriate for your CPU.
QEMU: Terminated
```

NiKiZe, 2021-07-09:
I think that i386 is a special case, maybe some kind of detection could be added to give you a message in the script.
At the same time, as noted in #399 switching to something else might be preferable.

NiKiZe, 2021-07-09:
What is your actual use case? Is it worth the effort to do this?

closed 2025-05-14 by NiKiZe