ipxe/ipxe #405: Building bad legacy BIOS iso image

JuniorJPDJ, 2021-07-08:
I'm trying to create ISO image working on both, EFI and legacy bios using your script `./util/genfsimg`.
I ended up having working disk image on all supported platforms and ISO image working on all EFI platforms and breaking on legacy BIOS.
Am I missing something or is this a bug?
ISOLINUX based image complains about ldlinux.c32 missing when booting, but SYSLINUX image is working correctly.

Legacy BIOS, SYSLINUX based disk image:
[image]

Legacy BIOS, ISOLINUX based CD/disk image (fails when booting from CD-ROM and when booting as disk):
[image]

Script I used to make images (and build iPXE):
https://git.juniorjpdj.pl/rootownia/docker/netboot/-/blob/master/ipxe-builder/build.sh
Dependencies I installed before building:
https://git.juniorjpdj.pl/rootownia/docker/netboot/-/blob/master/ipxe-builder/Dockerfile

NiKiZe, 2021-07-09:
Your script is to complex to try and figure out what is going on.
Could your post a minimal script that shows the issue?

If possible also include qemu commandline that can be used to test the 2 modes.

JuniorJPDJ, 2021-07-10:
Run clean debian and:
```
$ apt update -y
$ apt install -y --no-install-recommends build-essential gcc-aarch64-linux-gnu gcc-arm-linux-gnueabi wget ca-certificates liblzma-dev xorriso syslinux isolinux syslinux-utils mtools
```

build some iPXE images for targets:
bin-i386-pcbios/ipxe.lkrn bin-i386-efi/ipxe.efi bin-x86_64-efi/ipxe.efi bin-arm32-efi/snp.efi bin-arm64-efi/snp.efi

then try to create iso:

```
./util/genfsimg -o netboot.iso bin-i386-pcbios/ipxe.lkrn bin-i386-efi/ipxe.efi bin-x86_64-efi/ipxe.efi bin-arm32-efi/snp.efi bin-arm64-efi/snp.efi
```

I've no QEMU cmdline to run it, just try booting it on ANY legacy bios system or VM, it'll fail.

mcb30, 2021-07-26:
> ```
> ./util/genfsimg -o netboot.iso bin-i386-pcbios/ipxe.lkrn bin-i386-efi/ipxe.efi bin-x86_64-efi/ipxe.efi bin-arm32-efi/snp.efi bin-arm64-efi/snp.efi
> ```
> 
> I've no QEMU cmdline to run it, just try booting it on ANY legacy bios system or VM, it'll fail.

The above works fine for me on Fedora.

The most likely explanation is that you either do not have `ldlinux.c32` installed on your build host, or that Debian places it in a location that `genfsimg` does not search for: https://github.com/ipxe/ipxe/blob/4d180be5175e176ec27680557c1d67536a847610/src/util/genfsimg#L95-L106

If it's the latter, then please let us know where this file gets placed on Debian (or submit a PR that adds the relevant path).

JuniorJPDJ, 2021-07-26:
Well. It was my bad.
I was sure `ldlinux.c32` is also needed for USB images, but it was not.
It was entirely missing from my system as it's in `syslinux-common` package, which.. well.. is not being installed in dep to `syslinux` or `isolinux` which is weird.
Now it works.
Thanks for attention and help :D

closed 2021-07-26 by JuniorJPDJ