ipxe/ipxe #407: Invalid argument (1c0de802) error with large certificate chains

bjh21, 2021-07-09:
If I have an HTTPS server whose certificate chain is over 4K long, I get an `Invalid argument (http://ipxe.org/1c0de802)` error when trying to load an image from it.  I've tested this with chain lengths of 4272 bytes (fails) and 4016 bytes (succeeds)

The long certificate chain was generated by using [dehydrated](dehydrated-io/dehydrated) to get a certificate from Let's Encrypt with the default settings apart from setting `KEY_ALGO=rsa` because iPXE doesn't support ECDSA. The current default Let's Encrypt certificate chain is rather long to [support older Android clients](https://letsencrypt.org/2020/12/21/extending-android-compatibility.html).

Setting `PREFERRED_CHAIN="ISRG Root X1"` or `KEYSIZE=2048` each produced chains short enough for iPXE.

Ideally, I think iPXE should support longer certificate chains, since it can now take actual effort to avoid having one.  Failing that, the chain-length limit could usefully be documented at https://ipxe.org/err/1c0de8 and https://ipxe.org/crypto.

NiKiZe, 2021-07-09:
Think this is the same issue as #116 intends to fix

stappersg, 2023-04-02:
For your information:
* The above mentioned https://github.com/ipxe/ipxe/pull/116 got implemented as https://github.com/ipxe/ipxe/pull/930.
* This issue is now aware of that update.
* It might make sense to update this issue.

mcb30, 2023-04-02:
Fixed in #930

closed 2023-04-02 by mcb30

stappersg, 2023-04-03:
On Sun, Apr 02, 2023 at 03:40:00PM -0700, Michael Brown wrote:
> Fixed in #930 

Thanks for closing this issue.