ipxe/ipxe #408: Getting error: No such file or directory on loading image

dkhanure, 2021-07-12:
Hi Team,
 
I m trying to load a new image on a Bigbend (telnet 10.105.237.53 2032) router using the following commands.
 
Get to iPXE mode from boot manager.
Give IP addresses to boot image from tftp
set net0/ip 5.35.22.149
set net0/gateway 5.35.0.1
set net0/netmask 255.255.0.0
ifopen net0
after which ping to tftp is working
ping 202.153.144.25
64 bytes from 202.153.144.25: seq=1
64 bytes from 202.153.144.25: seq=2
64 bytes from 202.153.144.25: seq=3
64 bytes from 202.153.144.25: seq=4
then after giving command to boot the image, throws error
boot tftp://202.153.144.25//auto/tftp-blr-users2/dkhanure/ncs540l-x64.iso
iPXE> boot tftp://202.153.144.25//auto/tftp-blr-users2/dkhanure/ncs540l-x64.iso
tftp://202.153.144.25//auto/tftp-blr-users2/dkhanure/ncs540l-x64.iso... No such file or directory (http://ipxe.org/2d12608e)
IMAGE ncs540l-x64.iso freed
iPXE>
I do have the mentioned file in my tftp path.
-rw-r--r-- 1 dkhanure  eng        964268032 Jul 12 10:30 ncs540l-x64.iso
-ads-1501:/auto/tftp-blr-users2/dkhanure>
 
I have checked in the github discussion but could not get any relevant data from https://ipxe.org/err/2d1260. Any known issues?? Can you please help me with issue??
 
Thank you,
Divya K.

Teknoenie, 2021-07-12:
You are asking to download this ISO image over a TFTP connection.  Is 202.153.144.25 running a TFTP daemon on it and if so is `/auto/tftp-blr-users2` the root of the TFTP server itself?  Usually it is something like `/var/tftpboot`.

NiKiZe, 2021-07-12:
As already stated, check your path, if that fails double check your tftp servers log for errors to identify path it is trying to read from.

But loading a iso over tftp is not something that is even close to reasonable.

Until this is proven to be a possible bug with iPXE I will give this from issues to discussion

closed 2021-07-12 by NiKiZe