ipxe/ipxe #416: Problemn with currticks() on the EFI platform

pe-bo, 2021-07-19:
We found an interesting problem in the `currticks` function on the EFI platform.

If the `efi_raise_tpl` function is called twice, then the `efi_external_tpl` variable is set to `TPL_CALLBACK` and the timer is effectively not running, so the `efi_jiffies` variable does not increase.

This happens, for example, in the driver remove callback during shutdown (the TPL is first increased in `efi_snp_claim` and again when the devices are removed).

mcb30, 2021-07-19:
This behaviour is by design: see the commit message in https://github.com/ipxe/ipxe/commit/e10a40d41 for details.

The old TPL is recorded within the `struct efi_saved_tpl`, which makes it possible to nest pairs of `efi_raise_tpl()/efi_restore_tpl()` calls.

Marking as closed since I don't see any description of an actual problem: please follow up with details if a real problem exists.

closed 2021-07-19 by mcb30

pe-bo, 2021-07-19:
Yes, I know. Real problem is gracefully removing device with timeout. I have such device. How to solve it?

mcb30, 2021-07-19:
Any device removal code path should not be using `currticks()` in the first place.  Which upstream device driver is doing this?

pe-bo, 2021-07-19:
It is my own device driver.

Is it possible to use TPL_NOTIFY for EFI timer? Are there some drawbacks (yes, efi_jiffies now increments asynchronously)?

mcb30, 2021-07-19:
Use `udelay()` or `mdelay()` if needed for delays and waits in driver code.  See e.g. `intel_disable_ring()`: https://github.com/ipxe/ipxe/blob/c64dfff0a93b5f967ae21339313e61baf0b7a3a0/src/drivers/net/intel.c#L444-L466