ipxe/ipxe #419: Document "cpuvendor", "cpumodel", and "CPUID_SETTINGS"

PicoMitchell, 2021-07-29:
I only just noticed these variables exist when looking at recent commits and saw them here: https://github.com/ipxe/ipxe/commit/4d180be5175e176ec27680557c1d67536a847610

I double checked and these are not mentioned on https://ipxe.org/cfg

The documentation for "cpuvendor" and "cpumodel" could be very similar to https://ipxe.org/cfg/memsize and state that they require "CPUID_SETTINGS" which would also require adding documentation for that similar to https://ipxe.org/buildcfg/memmap_settings.

As always, thank you so much for iPXE!

NiKiZe, 2021-07-31:
Best source for documentation is probably https://ipxe.org/cmd/config

PicoMitchell, 2021-07-31:
That won't show settings that  weren't enabled during build though, right?

The issue here was that I didn't even know to enable "CPUID_SETTINGS" since I didn't know the "cpuvendor" and "cpumodel" existed.

Even when I did learn they existed, I had to read the source code to figure out how to enable them. Other users may not have been able to understand that code and wouldn't have know how to enable them without some documentation like there is for other settings.

mcb30, 2021-08-02:
Good point, and thank you for suggesting which existing documentation could most sensibly be used as a template.  I have added documentation for these at https://ipxe.org/cfg/cpumodel, https://ipxe.org/cfg/cpuvendor, and https://ipxe.org/buildcfg/cpuid_settings - please review and reopen if there are any issues.

closed 2021-08-02 by mcb30