ipxe/ipxe #429: Request: Add "memsize" in UEFI

PicoMitchell, 2021-08-08:
Hi @mcb30, I'm aware that [memmsize](https://ipxe.org/cfg/memsize) (enabled via [MEMMAP_SETTINGS](https://ipxe.org/buildcfg/memmap_settings)) does not currently work on UEFI. If a UEFI build of iPXE is attempted with MEMMAP_SETTINGS enabled, the build will fail.

I found a [reply from you](https://lists.ipxe.org/pipermail/ipxe-devel/2016-January/004560.html) about this issue back in 2016:
> MEMMAP_SETTINGS relies on the BIOS e820 memory map and so is not 
available under UEFI. The equivalent functionality to retrieve the UEFI 
memory map is not implemented.

It seems that is should be technically possible to retrieve RAM info in UEFI mode somehow since the UEFI version of [MemTest86](https://www.memtest86.com) is able to retrieve the RAM size as well as more detailed RAM info.

But, of course, I don't know how challenging adding this capability to iPXE would really be.

Is this something that could be added to UEFI iPXE?

Thank you so much for all your work on iPXE over so many years!

NiKiZe, 2021-08-08:
This was introduced for pcbios in c692a690dafe4c6bdb406c4c5e0ae66bfcb2f6cb ?
Information is probably available, but maybe in different format.

For the sake of documentation, What is the detailed information that you want to get?

PicoMitchell, 2021-08-08:
I would like to at least be able to get the same RAM info in UEFI mode that is available in BIOS mode, which is just the amount of installed RAM in MB.

If further RAM details are easily attainable (such as DDR# and speed), that would also be great to have exposed as settings in iPXE! But, at least just the amount of installed RAM would be nice in UEFI mode.

PicoMitchell, 2022-11-08:
I recently noticed that the open source Memtest86+ version 6 now supports UEFI (https://github.com/memtest86plus/memtest86plus).

Maybe some of that code could be a helpful reference in bringing `memsize` detection to iPXE in UEFI mode.