ipxe/ipxe #430: Feature request: Allow specifying range requests on imgfetch/kernel

NiKiZe, 2021-08-08:
imgfetch/initrd can currently load files from a number of protocols.
`-n` can be used to specify "efi filename" and an additional argument can be used to specify "cpio filename"

I would like to have the option to specify a `--range=start+length` (also `--range=byte` which would only give start position and read to end of file)
This would allow to download a part of a file.

Use case would be to collect positions of specific files in official files and refer to that.
Take [a Gentoo livecd](https://distfiles.gentoo.org/releases/amd64/autobuilds/20210801T170533Z/install-amd64-minimal-20210801T170533Z.iso) and grab the positions of different files in the ISO: (leaving out non interesting files from the listing)
```bash
# isoinfo -l -i install-amd64-minimal-20210801T170533Z.iso
Directory listing of /
      5458 -r-xr-xr-x   1    0    0  399257600 Aug  1 2021 [   5458 00] image.squashfs

Directory listing of /boot/
    200412 -r-xr-xr-x   1    0    0    4080544 Aug  1 2021 [ 200412 00] gentoo
    202405 -r-xr-xr-x   1    0    0    8943464 Aug  1 2021 [ 202405 00] gentoo.igz
```
Using `$(200412*2048)` syntax below, In real script real numbers would be used instead.

From this we could construct the offsets:
```ipxe
#!ipxe
set iso http://distfiles.gentoo.org/releases/amd64/autobuilds/20210801T170533Z/install-amd64-minimal-20210801T170533Z.iso

kernel --range=$(200412*2048)+4080544 ${iso} root=/dev/ram0 init=/linuxrc  dokeymap looptype=squashfs loop=/image.squashfs  cdroot initrd=initrd.magic

initrd --range=$(202405*2048)+8943464 ${iso} gentoo.igz
initrd --range=$(5458*2048)+399257600 ${iso} image.squashfs

imgstat
boot
```

This could work as an alternative for me in NiKiZe/Gentoo-iPXE#2, But goes for any ISO. (many ask how do I boot without extracting the ISO).
Hosting .ipxe scripts is easy, but extracted ISOs is more difficult.

Test to verify getting the correct data:
```bash
$ dd if=install-amd64-minimal-20210801T170533Z.iso bs=2048 skip=200412 | file -s -
/dev/stdin: Linux kernel x86 boot executable bzImage, version 5.10.52-gentoo-x86_64 (root@catalyst) #1 SMP Sun Aug 1 17:46:01 UTC 2021, RO-rootFS, swap_dev 0x3, Normal VGA
$ dd if=install-amd64-minimal-20210801T170533Z.iso bs=2048 skip=202405 | file -s -
/dev/stdin: XZ compressed data, checksum NONE
$ dd if=install-amd64-minimal-20210801T170533Z.iso bs=2048 skip=5458 | file -s -
/dev/stdin: Squashfs filesystem, little endian, version 4.0, zlib compressed, 399253568 bytes, 36488 inodes, blocksize: 131072 bytes, created: Sun Aug  1 17:58:20 2021
```

NiKiZe, 2021-08-21:
Looking into this the way that these `cmd`s work is to do `imgacquire` / `imgdownload_string` and finally do `http_open_uri` but it depends on the protocol.
I do not see a clean way to handle range for all protocols - so this leads me to think if there is some way to specify range in URL instead, but that also gives of the feeling of a hack.

That would be `parse_uri` but there is already `##params`
maybe a `##range` is not that hackish anyway?

One thing to consider is how any of the protocols already using range (peercache and sanboot) would deal with range URLs? Would it maybe be ok to leave that as "unsupported", or maybe it's not to much work calculating offset?

stappersg, 2022-05-27:
As I understand the current situation, this issue waits on some one adding support for images via the `image_extract()` mechanism (including adding a `name` parameter to the `.extract()` method). This would allow [imgextract](https://ipxe.org/cmd/imgextract) to be used to extract files from a downloaded ISO image.

Option _2._ of @mcb30 's answer is outside of the _iPXE source code_, but still an _iPXE documentation_ thing.

NiKiZe, 2023-03-08:
A possible solution (ignoring negotiation with server if it supports range requests or not) might be to use the new --header from 96bb6ba

For this to be realistically usable, params should be enabled by default in efi builds.

mcb30, 2023-03-08:
> A possible solution (ignoring negotiation with server if it supports range requests or not) might be to use the new --header from [96bb6ba](https://github.com/ipxe/ipxe/commit/96bb6ba441653a30729ade38dc6c23bc9e2d2339)

It's worth noting that the iPXE HTTP client would then be unaware that it was making a range request.  This becomes significant in some situations.  For example, if using a SAN HTTP URI, then iPXE would end up creating an invalid request with two conflicting `Range` headers present.