ipxe/ipxe #438: System freeze on ipxe exit / boot continue

Stephenm64, 2021-08-10:
Seeing various systems freeze and lock up on a black screen on timeout or exiting the bootloader and continuing the boot process.

This issue was not seen with v1.20.1, issue persists in v1.21.1 and current master.

performed a git bisect:
```
1295b4acff1f2014261c40d9f9d2107ffd668d92 is the first bad commit
commit 1295b4acff1f2014261c40d9f9d2107ffd668d92
Author: Michael Brown <mcb30@ipxe.org>
Date:   Mon Nov 23 15:34:13 2020 +0000

    [efi] Allow initialisation via SNP interface even while claimed
```

from looking at other issues this may be related to #349 and possibly #213, another user seeing boot exit issues: #360

edgood1, 2021-08-17:
I am also similar behavior on newer systems. Upon running the exit command, seeing this:

```!!!! X64 Exception Type - 0D(#GP - General Protection)  CPU Apic ID - 00000000 !!!!
ExceptionData - 0000000000000000
RIP  - 0000000066878D44, CS  - 0000000000000038, RFLAGS - 0000000000010246
RAX  - AFAFAFAFAFAFAFAF, RCX - 00000000608E4498, RDX - 00000000608E4498
RBX  - 00000000608E4498, RSP - 000000006CD4D390, RBP - 0000000000000008
RSI  - 000000005E751D90, RDI - 0000000000000004
R8   - 00000000668791E8, R9  - 0000000080000200, R10 - 0000000000000004
R11  - 0000000000000004, R12 - 0000000000000000, R13 - 000000006687DC90
R14  - 000000006CD75420, R15 - 0000000000000002
DS   - 0000000000000030, ES  - 0000000000000030, FS  - 0000000000000030
GS   - 0000000000000030, SS  - 0000000000000030
CR0  - 0000000080000013, CR2 - 0000000000000000, CR3 - 000000006CBCD000
CR4  - 0000000000000668, CR8 - 0000000000000000
DR0  - 0000000000000000, DR1 - 0000000000000000, DR2 - 0000000000000000
DR3  - 0000000000000000, DR6 - 00000000FFFF0FF0, DR7 - 0000000000000400
GDTR - 0000000066C122E0 0000000000000047, LDTR - 0000000000000000
IDTR - 00000000649FB018 0000000000000FFF,   TR - 0000000000000000
FXSAVE_STATE - 000000006CD4CFF0
!!!! Find image d:\a3\_work\12\s\development\biossource\Build\NeonCity\RELEASE_VS2015\X64\AmiNetworkPkg\UefiNetworkStack\Common\MnpDxe\MnpDxe\DEBUG\MnpDxe.pdb (ImageBase=0000000066875000, EntryPoint=00000000668754D8) !!!!
```

NiKiZe, 2021-08-17:
> I am also seeing similar behavior on newer systems. Upon running the ipxe exit command, seeing this:
> 
> ```
> !!!! X64 Exception Type - 0D(#GP - General Protection)  CPU Apic ID - 00000000 !!!!
> ExceptionData - 0000000000000000
> RIP  - 0000000066878D44, CS  - 0000000000000038, RFLAGS - 0000000000010246
> RAX  - AFAFAFAFAFAFAFAF, RCX - 00000000608E4498, RDX - 00000000608E4498
> RBX  - 00000000608E4498, RSP - 000000006CD4D390, RBP - 0000000000000008
> RSI  - 000000005E751D90, RDI - 0000000000000004
> R8   - 00000000668791E8, R9  - 0000000080000200, R10 - 0000000000000004
> R11  - 0000000000000004, R12 - 0000000000000000, R13 - 000000006687DC90
> R14  - 000000006CD75420, R15 - 0000000000000002
> DS   - 0000000000000030, ES  - 0000000000000030, FS  - 0000000000000030
> GS   - 0000000000000030, SS  - 0000000000000030
> CR0  - 0000000080000013, CR2 - 0000000000000000, CR3 - 000000006CBCD000
> CR4  - 0000000000000668, CR8 - 0000000000000000
> DR0  - 0000000000000000, DR1 - 0000000000000000, DR2 - 0000000000000000
> DR3  - 0000000000000000, DR6 - 00000000FFFF0FF0, DR7 - 0000000000000400
> GDTR - 0000000066C122E0 0000000000000047, LDTR - 0000000000000000
> IDTR - 00000000649FB018 0000000000000FFF,   TR - 0000000000000000
> FXSAVE_STATE - 000000006CD4CFF0
> !!!! Find image d:\a3\_work\12\s\development\biossource\Build\NeonCity\RELEASE_VS2015\X64\AmiNetworkPkg\UefiNetworkStack\Common\MnpDxe\MnpDxe\DEBUG\MnpDxe.pdb (ImageBase=0000000066875000, EntryPoint=00000000668754D8) !!!!
> ```

I find this dump interesting.
Could you give more info about the system? As well as information about the source of iPXE binary, such as version of GCC used if compiled by you?
Could you also test if prebuilt ipxe.efi and snponly.efi from boot.ipxe.org works or not?

edgood1, 2021-08-17:
I just tried it with the prebuild ipxe.efi from boot.ipxe.org and got the same error message. 

If I pull a fresh copy of master, run
`git revert --strategy resolve 1295b4acff1f2014261c40d9f9d2107ffd668d92`
and build, the problem goes away.

edgood1, 2021-08-17:
gcc is from ubuntu 20.04 build-essential. version 9.3.0

NiKiZe, 2021-08-17:
> I just tried it with the prebuild ipxe.efi from boot.ipxe.org and got the same error.
> 
> If I pull a fresh copy of master, run
> `git revert --strategy resolve 1295b4acff1f2014261c40d9f9d2107ffd668d92`
> and build, the problem goes away.

Thanks for checking, since there have been reports of issues which seems to go away with the prebuilt ones I wanted to verify if they are related or not, which they don't seem to be.

Stephenm64, 2021-08-18:
I actually also reverted the commit 1295b4acff1f2014261c40d9f9d2107ffd668d92 on the master a day after I posted this issue and it has be working as expected.

aggieNick02, 2021-11-08:
I'm seeing this as well with the prebuilt ipxe.efi. Not on all machines, but some. I'm happy to provide any details that would be useful

aggieNick02, 2021-11-11:
Like others, running
```git revert --strategy resolve 1295b4acff1f2014261c40d9f9d2107ffd668d92```
and then
```
cd src
make bin-x86_64-efi/ipxe.efi
```
yielded an ipxe.efi that does not exhibit the problem for the machines that hang with the file taken from the prebuilt ISO.

NiKiZe, 2022-07-04:
Any chance that efi firmware update resolves this in most cases?

Stephenm64, 2022-07-04:
Personally of the systems I have seen with the issue I have not seen any bios firmware updates resolve the issue. I have found though with these systems a build of snponly.efi may sometimes work.

stappersg, 2022-11-22:
[Link to another _"That `git revert` works also for me"_.](https://github.com/ipxe/ipxe/discussions/789#discussioncomment-4198802)

aggieNick02, 2025-04-23:
A somewhat related update. I had been using a manually built ipxe.efi as described above (from Nov 2021 with the one commit reverted) without issue until recently. It worked fine in conjunction with Windows 11 22H2.

However, it will not work with Windows 11 24H2. I have a Hyper-V VM which first hits ipxe.efi and then falls through to the locally installed OS. When Windows 11 24H2 is the locally installed OS, it will immediately cause the PC to reboot, leading to a boot->ipxe.efi->start win1124h2->reboot cycle. The exact same setup works fine with Win1122H2.

Anyways, building the latest ipxe.efi made the problem go away. Unfortunately I do not remember what hardware was having the issues from this issue 438 almost 4 years ago, so I'll have to wait and see if I run into it again.

closed 2026-04-23 by NiKiZe