ipxe/ipxe #440: Intel I217-LM not reset properly once used in another OS?

c---, 2021-08-11:
This is an ASUS Q87M-E motherboard with the built-in Intel I217-LM network controller running diskless (iSCSI) Linux and Windows.

Upon first power-up/boot the network adapter runs at full gigabit speed in iPXE. However, once an OS has initialized the card (eg. Linux or Windows) then on subsequent reboots the card runs VERY slow (10Mbps?) during any iPXE operation. Once an OS is loaded again then the card works at full speed.

So I think something in the iPXE reset is not working correctly and the link speed drops way down. What are Linux or Windows doing that resets the card properly and makes it run at full speed? iPXE needs to do whatever that is.

Only a hard power-off and restart will make iPXE run at full speed again.

This has been a problem for me for like 7 years now since I've had this board. It's really driving me crazy because every time I reboot I have to wait like 5 minutes while iPXE slowly loads network media.

I have tried every version of iPXE over the years. Currently running off _master_ git code. Various BIOS settings seem have no effect. The problem persists.

NiKiZe, 2021-08-12:
So you are now using git master that you built yourself? Can we have a  version number from the iPXE banner please?
Also could you grab the exakt device ID, if you are running I would ask for `lspci -nn` but output from ifstat in iPXE could also do.

c---, 2021-08-12:
Yes, I'm running git master that I built. iPXE banner says:
```
iPXE 1.21.1+ (ge09e1)
Features: DNS HTTP HTTPS iSCSI AoE ELF MBOOT PXE bzImage Menu PXEXT
```

lspci -nn produces:
```
00:19.0 Ethernet controller [0200]: Intel Corporation Ethernet Connection I217-LM [8086:153a] (rev 05)
```

NiKiZe, 2021-08-12:
Thanks! ( Do you have local commits? https://github.com/ipxe/ipxe/commit/ge09e1 does not exist 😉 )
We can see here that `8086:153a` has a special flag to do the right thing to reset, but maybe there is more gotchas here.
https://github.com/ipxe/ipxe/blob/adb2ed907ea0b67a7e39a65ca1bbcf7224415e1b/src/drivers/net/intel.c#L1144

Could you try and change `INTEL_NO_PHY_RST` to `0` and see if you see the issue more or less often?
Have you tried using undionly.kpxe instead which does not have the iPXE native Intel driver, and does that behave differently?

c---, 2021-08-12:
The only change I have made is to enable HTTPS. I'm on commit e0d24696b9d9d78d621e3dedfcb3d2c094bdd8a3

I have tried removing the flag INTEL_NO_PHY_RST and setting it to 0. The problem still occurs. I have actually tried that a few times over the years because I did notice that change in the code.

I'm currently using unionly.kpxe and have tried ipxe.pxe as well and they're both slow. I just now tried UEFI mode with ipxe.efi and that DOES boot at full speed but UEFI on this motherboard is incredibly unstable so it's not really usable. Switching back to BIOS from UEFI the network is still slow.

NiKiZe, 2021-08-12:
The banner of iPXE does include commit that it was built from.

I have seen similar in EFI, but remember that your issue is dependent on the state of the NIC.
It would be great to try and resolve this issue on a driver level, but for a one of, maybe there is BIOS/Firmware updates that could work around the issue.

One other thought tho, I'm assuming you are chaining into ipxe.pxe by using dhcp+tftp?
Have you tried to boot from USB (using ipxe.usb) my reasoning is that maybe the BIOS PXE stack is part of the issue of the state that iPXE is not able to reset.

Another, more dangerous thing to try is to use the `INTEL_I219` flag. (again, the correct thing here is do proper debugs, but is not very likely)

c---, 2021-08-12:
The motherboard BIOS is the latest I know of: 3801 03/22/2019.

Yes, I'm chaining with DHCP+ TFTP. I just tried booting from a USB image and it still has the slow network.

I tried setting flag INTEL_I219 and it made no difference.

NiKiZe, 2021-08-12:
Thanks for checking and reporting on all those things.
The i219 workaround was introduced in 546dd51de8459d4d09958891f426fa2c73ff090d
I will try to have a go at trying to find anything similar for i217, but no promises for when.

c---, 2021-08-12:
Is there an easy way to capture debugging output?

The motherboard does have serial redirection for the console but it has no serial port so I'd have to rig something up to the serial header on the board.

Just looking at the debug output it doesn't look like there is much information. It resets the device, gets/sets the MAC address, then a bunch of rx/tx data starts flowing. Hmmm, I'll look at this tomorrow.

NiKiZe, 2021-08-12:
There is serial if you can hook that up, there is also ip/syslog logging, since you have net that might be an option.
also the .usb image has a logging partition. Sorry this is all from memory, and not something I have actually used myself, so details is vauge.

c---, 2021-08-12:
Using the USB CONSOLE_INT13 log was easy enough:
[fastnet.txt](https://github.com/ipxe/ipxe/files/6974620/fast.txt)
[slownet.txt](https://github.com/ipxe/ipxe/files/6974621/slow.txt)

One thing that stands out is the "link status" if different between the two. 400806**8**3 in fast mode and 400806**0**3 in slow mode.

NiKiZe, 2026-10-19:
Reading this now, it should be up to the OS driver to reset the state correctly.
I do wonder if this is still an issue on this NIC/board
And if the behavior changes with 1573f8c?

c---, 2026-10-21:
>          And if the behavior changes with [1573f8c](https://github.com/ipxe/ipxe/commit/1573f8c084ffd9e763992a6a739f308a9edd95a1)?

I'll try it the next time I get a chance. I'm currently using an external NIC and the machine is a critical server so I can't test until the next reboot cycle.