ipxe/ipxe #447: Feature Request: boot from local disk in uefi

edgood1, 2021-08-16:
I need in the ability to run the equivalent of `sanboot --no-describe --drive 0x80` for uefi enabled hosts. That command only will boot a local disk in legacy bios mode.

The method of allowing ipxe to exit and fall back to disk via the bootlist is not an option in our scenario.

I found this thread https://lists.ipxe.org/pipermail/ipxe-devel/2019-February/006504.html but it doesn't seem that that this feature ever was merged.

NiKiZe, 2021-08-16:
Related or close to duplicate of
#428 #213

johnnykeats, 2021-11-09:
Not sure it helps. On BlueBanquise project (https://github.com/bluebanquise/bluebanquise), we used a grub2 image with an embed script that look for OS and boot it.
It however does not work on all scenario. We could use it on almost anything, but @oxedions reported this was not working on Nvidia DGX systems.

To build the grub2 image:
```
        if [ $distribution == "Ubuntu" ]; then
            grub-mkstandalone -O $ipxe_arch-efi -o grub2_efi_autofind.img "boot/grub/grub.cfg=grub2-efi-autofind.cfg"
            grub-mkstandalone -O $ipxe_arch-efi -o grub2_shell.img "boot/grub/grub.cfg=grub2-shell.cfg"
        else
            grub2-mkstandalone -O $ipxe_arch-efi -o grub2_efi_autofind.img "boot/grub/grub.cfg=grub2-efi-autofind.cfg"
            grub2-mkstandalone -O $ipxe_arch-efi -o grub2_shell.img "boot/grub/grub.cfg=grub2-shell.cfg"
        fi
```

(Code taken from https://github.com/bluebanquise/infrastructure/blob/master/auto_builder.sh)

**grub2-efi-autofind.cfg** content is:

```
# EFI disk boot - auto find script
# Benoit Leveugle <benoit.leveugle@sphenisc.com>

echo " _________________"
echo "|# :           : #|"
echo "|  : Blue      :  |"
echo "|  :  Banquise :  |"
echo "|  :EFI - Grub2:  |"
echo "|  :___________:  |"
echo "|     _________   |"
echo "|    | __      |  |"
echo "|    ||  |     |  |"
echo "\____||__|_____|__|lc"
echo
echo " Loading modules..."
insmod part_gpt
insmod fat
insmod chain
insmod part_msdos
insmod ext2
echo
echo "Scanning, first pass..."
for efi in (*,gpt*)/efi/*/grub.cfg (*,gpt*)/efi/*/*/grub.cfg (*,gpt*)/grub.cfg (*,gpt*)/*/grub.cfg ; do
                regexp --set=1:efi_device '^\((.*)\)/' "${efi}"
done
echo "Scanning, second pass..."
for efi in (*,gpt*)/efi/*/grub.cfg (*,gpt*)/efi/*/*/grub.cfg (*,gpt*)/grub.cfg (*,gpt*)/*/grub.cfg ; do
                regexp --set=1:efi_device '^\((.*)\)/' "${efi}"
                if [ -e "${efi}" ]; then
                    efi_found=true
                    echo " >> Found operating system! <<"
                    echo " Path: ${efi}"
                    echo " Booting in 5s..."
                    sleep --interruptible --verbose 5
                    configfile "${efi}"
                    boot
                fi
done

echo "No EFI known OS found. Fall back on shell."
sleep 2s
```

(Code taken from https://github.com/bluebanquise/infrastructure/blob/master/packages/ipxe-bluebanquise/grub2-efi-autofind.cfg)

And then to load it, just use:

`chain http://${next-server}/preboot_execution_environment/bin/${arch}/grub2_efi_autofind.img
`
(Code taken from https://github.com/bluebanquise/bluebanquise/blob/master/roles/core/pxe_stack/templates/menu.ipxe.j2)

We also use `exit` trick, but this way was cleaner for us.

srett, 2021-11-10:
@johnnykeats  I'm not fluent in grub, but isn't this only looking for a `grub.cfg` on the local disk? I.e. it wouldn't boot a standard Windows EFI install....

johnnykeats, 2021-11-15:
You are right, apologies. This is a workaround for Linux Grub2 based systems only.
Agree a feature to boot on EFI disks natively would be very nice.

FrancoCapraro, 2022-02-25:
@edgood1 you could resolve the problem ?

edgood1, 2022-03-03:
> @edgood1 you made something to solve the problem ?

@FrancoCapraro : The three part patch that Aaron Young posted here: https://lists.ipxe.org/pipermail/ipxe-devel/2019-February/006506.html 
is the best solution I've found.

jensgw, 2022-03-04:
The feature in Aaron Young's patch would also interesst me.

edgood1, 2022-07-13:
#612 seems to be the solution to this issue.

edgood1, 2023-01-05:
This may be obvious, but a workaround to this issue, if you know the name and location of the boot image files on your EFI system partition is to do something like this in your ipxe script
```
dhcp
initrd file://EFI/initrd.img
kernel file://EFI/kernel.efi initrd=initrd.img root=UUID=xxxxxxxx some_kernel_params 
boot
```

TomKranenburg, 2023-07-12:
> #612 seems to be the solution to this issue.

Worked for me and eliminated and bit of a headache. Thanks for this. I actually assumed this would be in the main branch already.

eoli3n, 2023-11-14:
> This may be obvious, but a workaround to this issue, if you know the name and location of the boot image files on your EFI system partition is to do something like this in your ipxe script
>
> ```
> dhcp [...]

Should this work without the patch ?

edgood1, 2023-11-14:
> Should this work without the patch ?

Yes, it should work. Confirm the layout of the EFI system partition and the location of the file. I think ``//EFI`` is the root of the EFI system partition, so you might have to use `file://EFI/EFI/debian/grubx64.efi` if you have an `EFI` directory. I've ran into this before.

eoli3n, 2023-11-14:
Hm, no more luck

[image]

```
$ tree /boot/efi
/boot/efi
└── EFI
    └── debian
        ├── BOOTX64.CSV
        ├── fbx64.efi
        ├── grub.cfg
        ├── grubx64.efi
        ├── mmx64.efi
        └── shimx64.efi
```

srett, 2023-11-14:
The first part after `file://` has to be the name (label) of that partition, use gdisk or similar to verify.

eoli3n, 2023-11-14:
Problem is that I will use this on 900 hosts, so obviously with differents label. And I will first boot to deploy OS, so no way to preset a label on disk.

TomKranenburg, 2023-11-14:
I really feel this is a core feature. Would love to see it in the main branch. Until then in order to keep things running I have to manually patch it each time.

This doesn't bother me of course but it does mean novices will struggle when they're first employed. We could make things much easier for newbies I think.

oxedions, 2023-11-14:
Here is the workaround we used here to boot our clusters. We adapted what @johnnykeats did for Linux to boot any OS thanks to @JBE on https://stackoverflow.com/questions/67506022/create-a-grub-script-that-auto-detect-efi-systems-and-boot-first-one 

Idea is to use a grub loader that looks for BOOTX64.EFI, which is default on Windows or Linux.

Script to build grub image is here: https://github.com/bluebanquise/infrastructure/blob/52570aa0353212d71b6e210c7239e0bdbeadb41a/packages/bluebanquise-ipxe/build.sh#L80

And grub script that is injected into the grub image and that looks for BOOTX64.EFI is here (you can remove the second part, which is not needed and will be removed): https://github.com/bluebanquise/infrastructure/blob/master/packages/bluebanquise-ipxe/grub2-efi-autofind.cfg

Booting on this image from iPXE is simple: https://github.com/bluebanquise/bluebanquise/blob/2a8bdac383dc437d271849ba7dc16aedaada7283/collections/infrastructure/roles/pxe_stack/templates/menu.ipxe.j2#L113

Hope it helps.

eoli3n, 2023-11-22:
Thanks, any chance to PR this to get it merged officially ?

mcb30, 2024-03-05:
The ability to boot from a local disk under UEFI is present upstream as of commit https://github.com/ipxe/ipxe/commit/b1c13cc43

closed 2024-03-05 by mcb30

TomKranenburg, 2024-03-05:
> The ability to boot from a local disk under UEFI is present upstream as of commit [b1c13cc43](https://github.com/ipxe/ipxe/commit/b1c13cc43)

You're a hero @mcb30

eoli3n, 2024-03-05:
Thanks a lot for this !

dash1540, 2024-09-11:
this work only with fat ?
I have usb drive with 2 partitions, EFI fat part and ntfs part, i can chain from EFI part but cannot chain from ntfs part, and efimap command not found

NiKiZe, 2024-09-11:
> this work only with efi fat partition? I have usb drive with 2 partitions, EFI fat part and ntfs part, i can chain file from EFI part but cannot chain from ntfs part, and efimap command not found

Does your efi firmware support reading NTFS filesystems? The only expected support in uefi is fat32 which is in the uefi spec. iPXE does not do any filesystem handling at all.

there is no efimap command, the support that was merged is in b1c13cc43

dash1540, 2024-09-11:
im using qemu-system-x86_64 with OVMF_CODE.fd, its a uefix64 env, also tested with vmware 17.5.2.

first i load grubx64.efi and chain ipxe.efi with command 'chainloader /ntfspart/ipxe.efi' from ipxe shell i try to chain file from ntfs, its work when load from efi part

is there any command or hack to list all block devices from ipxe shell ?

mcb30, 2024-09-11:
> first i load grubx64.efi and chain ipxe.efi with grub command 'chainloader /ntfspart/ipxe.efi'

GRUB uses its own filesystem drivers, and ignores the UEFI firmware's filesystem support.

iPXE uses the filesystem support provided by the UEFI firmware.  This is generally limited to using FAT-formatted disks, since that is the filesystem that UEFI specifies.

We are not going to add filesystem drivers in iPXE, so you will not be able to use iPXE to boot from an NTFS partition (or EXT4, or XFS, or BTRFS, etc).

dash1540, 2024-09-11:
thanks for the answers.