ipxe/ipxe #449: [net/dhcpv6] DHCPv6 addressing in non-/64 networks - ipv6_create_routes()

Manawyrm, 2021-08-28:
Hi! 

I'm currently trying to use iPXE in an IPv6-only setup, where each system has their own little IPv6 subnet and is directly talking to it's router for all communication.
All networks are /126 for this reason, there's no IPv4 or anything.

Router advertisements are sent and look like this: 
```php
interface br0
{
	AdvSendAdvert on;
	AdvManagedFlag on;
	AdvOtherConfigFlag on;
	AdvReachableTime 0;
	AdvRetransTimer 0;
	AdvCurHopLimit 64;
	AdvDefaultLifetime 30;
	AdvHomeAgentFlag off;
	AdvDefaultPreference medium;
	AdvSourceLLAddress on;

	prefix 2001:db8:0:3::4/126
	{
		AdvValidLifetime 2592000;
		AdvPreferredLifetime 604800;
		AdvOnLink on;
		AdvAutonomous on;
		AdvRouterAddr off;
	};
};
```

As you can see, Managed and OtherConfig flags are enabled and the network is smaller than /64, so iPXE cannot assign an IPv6 address via SLAAC.
After receiving this ICMPv6 RA packet, iPXE will start to query the DHCPv6 server and receive an IPv6 address. 

iPXE now has an IPv6-address (via DHCPv6) and knows the subnet length and default gateway (via ICMPv6 RA).
In the settings, net0 is filled correctly: 
[image]

net0.ndp.0 only has the gateway and subnet length, though: 
[image]

and net0.dhcpv6 has the IPv6 address, DNS server, etc.: 
[image]

This is the debug output and route table:
```
iPXE> ifconf -c ipv6
Configuring [ipv6] (net0 52:54:00:f2:4f:11)...NDP net0 found default router fe80::1e34:daff:fec8:1369 for on-link autonomous prefix 2001:db8:0:3::4/126
DHCPv6 net0 has XID ce980f
DHCPv6 net0 entering SOLICIT state
NDP net0 found default router fe80::1e34:daff:fec8:1369 for on-link autonomous prefix 2001:db8:0:3::4/126
DHCPv6 net0 received ADVERTISE from fe80::1e34:daff:fec8:1369
DHCPv6 net0 received ADVERTISE is for 2001:db8:0:3::6
DHCPv6 net0 entering REQUEST state
DHCPv6 net0 received REPLY from fe80::1e34:daff:fec8:1369
DHCPv6 net0 received REPLY is for 2001:db8:0:3::6
DHCPv6 net0 complete
 ok
iPXE> route
net0: 2001:db8:0:3::6/64
net0: fe80::5054:ff:fef2:4f11/64
```

The route on net0 does not have a gateway address attached to it.
Communication is therefore broken to all v6-targets outside of the Layer 2.

I've investigated this further and found the function ipv6_create_routes() in ipv6.c.
ipv6_create_routes() accepts a settings object (will get "net0" from the caller) and will run through the settings-tree recursively.
It needs 3 prerequisits to work properly:
- setting `ip6`
- setting `len6`
- setting `gateway6`

Here's the catch: 
All of these settings must be in the same layer of the settings tree. Also: The first layer (net0, which would have all 3 options) is ignored.
The way ipv6_create_routes() is built, requires all settings to be in the same layer and the `( origin != settings )`-check requires them to not be in `net0` but a further-down layer.

Just comment the `( origin != settings )` check in https://github.com/ipxe/ipxe/blob/e09e1142a3bd8bdb702efc92994c419a53e9933b/src/net/ipv6.c#L1292 out and it starts to work perfectly.

I'm not sure, which side effects this has, but I've tested a regular SLAAC /64 network with the changes and it worked fine as well.

How could this be fixed properly upstream? 

Thanks,
Tobias

MalteJ, 2024-02-23:
I think this has been fixed with https://github.com/ipxe/ipxe/commit/8f5fc1614
See also #732 

CC: @mcb30

Manawyrm, 2024-02-23:
> I think this has been fixed with [8f5fc1614](https://github.com/ipxe/ipxe/commit/8f5fc1614) See also #732

Hm, I don't work at the company with that setup anymore, so I can't easily test/confirm this right now.

closed 2024-06-27 by mcb30

mcb30, 2024-06-27:
Should be fixed via #1245