ipxe/ipxe #456: Sanboot and https redirects

antonym, 2021-09-04:
I was attempting to sanboot an image from Github Releases, and found it was hanging on booting the image.  Releases do a redirect from https://github.com to https://github-releases.githubusercontent.com.  If I point sanboot to use the redirected URL, it boots appropriately.

Would it be possible for sanboot to be able to handle redirects in any way?

mcb30, 2021-09-06:
Redirects are handled by the HTTP client code and should function independently of the block device (SAN) layer.

Could you try building with `DEBUG=httpcore,tls` and post the debug output?

ibuildthecloud, 2021-11-08:
What I've found related to this issue is that github will redirect to a S3 URL and the redirection logic in ipxe will effectively mangle the URL.  What is happening is that the Location header is URI parsed and then encoded.  The issue is that the parsing and encoding does not come back to the same string that was in the Location header because there is a disagreement as to which characters should be URL encoded (specifically / and =).  For example.

Location is
```
HTTP 0xec204 RX Location: https://objects.githubusercontent.com/github-productio
n-release-asset-2e65be/419402169/fad6c39c-c555-4fb1-a1c3-8e3847c98cdd?X-Amz-Algo
rithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20211108%2Fus-eas
t-1%2Fs3%2Faws4_request&X-Amz-Date=20211108T225507Z&X-Amz-Expires=300&X-Amz-Sign
ature=e6c31fdb5ef59296825a8ad0b1730e7628f6907f1783569ee40e72a86c8c9b8e&X-Amz-Sig
nedHeaders=host&actor_id=0&key_id=0&repo_id=419402169&response-content-dispositi
on=attachment%3B%20filename%3Drancheros-v0.1.0-alpha03-amd64.ipxe&response-conte
nt-type=application%2Foctet-stream
```
and the resulting get is
```
HTTP 0xee424 TX GET /github-production-release-asset-2e65be/419402169/fad6c39c-c
555-4fb1-a1c3-8e3847c98cdd?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKI
AIWNJYAX4CSVEH53A/20211108/us-east-1/s3/aws4_request&X-Amz-Date=20211108T225507Z
&X-Amz-Expires=300&X-Amz-Signature=e6c31fdb5ef59296825a8ad0b1730e7628f6907f17835
69ee40e72a86c8c9b8e&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=4194021
69&response-content-disposition=attachment;%20filename=rancheros-v0.1.0-alpha03-
amd64.ipxe&response-content-type=application/octet-stream
```

If I manually take the string from ipxe and then encode the characters the same way as the location header it will work.

I double checked as to what cURL does it an it does not seem to change the encoding of the Location header and instead just takes the raw path fragment as is.  I think the correct behavior is to not url decode/encode the string and instead just deal with the raw string.

mcb30, 2021-11-12:
Should be fixed as of https://github.com/ipxe/ipxe/commit/1844aacc8

closed 2021-11-12 by mcb30

antonym, 2021-11-13:
I found this fixed an issue I was having trying to chain an iPXE file from a Github releases folder, but I'm still seeing issues with Sanboot where it continuously loops from URL to URL.  I was trying to load this URL via UEFI:

sanboot https://github.com/netbootxyz/asset-mirror/releases/download/9.3-4e73affb/memtest86-usb.img

It works fine locally from a webserver, but from Github it just appears to loop through URL redirection.

ibuildthecloud, 2021-11-23:
I can confirm that this fixed my issue with GitHub redirects.  Thanks!

PR #516 (2021-11-12): [uri] Retain original encodings for path, query, and fragment fields
iPXE decodes any percent-encoded characters during the URI parsing
stage, thereby allowing protocol implementations to consume the raw
field values directly without further decoding.

When reconstructing a URI string for use in an HTTP request line, the
percent-encoding is currently reapplied in a reversible way: we
guarantee that our reconstructed URI string could be decoded to give
the same raw field values.

This technically violates RFC3986, which states that "URIs that differ
in the replacement of a reserved character wi [... cut]