ipxe/ipxe #471: Make targets for arm64 returns " error: unrecognized command line option ‘-mlittle-endian’ "

rustyscottweber, 2021-09-21:
Unable to make Arm tftp images due to the following errors.

`[root@ipxe src]# make ARCH=arm64 bin-arm64-efi/ipxe.efi
  [DEPS] crypto/certstore.c
gcc: error: unrecognized command line option ‘-mlittle-endian’
  [DEPS] crypto/privkey.c
gcc: error: unrecognized command line option ‘-mlittle-endian’
  [BUILD] bin-arm64-efi/__umoddi3.o
gcc: error: unrecognized command line option ‘-mlittle-endian’
make: *** [bin-arm64-efi/__umoddi3.o] Error 1
`

NiKiZe, 2021-09-21:
Which version of gcc do you have? And can you give more information about the system where this runs?

Remember that you need to provide enough information so that others can reproduce, or at least understand the issue.

mcb30, 2021-09-21:
For cross-compiling, you will need to specify an approriate `CROSS=...` parameter to specify your cross-toolchain prefix.  For example:
```
make CROSS=aarch64-linux-gnu- bin-arm64-efi/ipxe.efi
```
Note that you do not need to (and should not) specify `ARCH` explicitly: this will be calculated based on the `bin-arm64-efi` portion of the build target.

jerrywoo96, 2022-01-31:
Someone add the cross compile command option in the docs thanks. had to came here to find out why i cannot compile arm targets in my x64 pc.

https://ipxe.org/appnote/buildtargets

NiKiZe, 2022-01-31:
> Someone add the cross compile command option in the docs thanks. had to came here to find out why i cannot compile arm targets in my x64 pc.
> 
> https://ipxe.org/appnote/buildtargets

Added, feel free to come with feedback if you do not have access to edit the page yourself.

RikiReal, 2023-06-17:
In case anybody has the same issue the "-" in "CROSS=aarch64-linux-gnu-" is important, as it's inserted literally in the Makefile.
Without the "-", the command would be aarch64-linux-gnugcc, which doesn't exist.

https://github.com/ipxe/ipxe/blob/4fa4052c7ebb59e4d4aa396f1563c89118623ec7/src/Makefile#L29