ipxe/ipxe #472: Attempting to make 32 bit arm uefi image results in never ending search for dependencies.

rustyscottweber, 2021-09-21:
```
nohup make ARCH=arm bin-arm32-efi/ipxe.efi &
```

rustyscottweber, 2021-09-21:
Looks like this is only while attempting to cross compile.  Attempting to build this target on a 32bit arm device results in a successful build.

NiKiZe, 2021-09-21:
Have you made any modifications to configurations or similar? and did you try with `make V=1` ?

rustyscottweber, 2021-09-21:
No modifications. I have not yet tried using V=1. However, adding an additional parameter for centos (May differ for ubuntu/debian):
```
CROSS_COMPILE=arm-linux-gnu-
```
seems to allow the build to get much further. It still doesn't build, but at least I get the same error that I do on a native build.

rustyscottweber, 2021-09-21:
Looks like this was something screwy with my build environment. After attempting with the cross compile parameter, I can no longer reproduce the issue. Closing issue, but maybe this will help posterity.

closed 2021-09-21 by rustyscottweber