ipxe/ipxe #473: undefined refference to 'readq' and 'writeq' while building arm32 base efi targets

rustyscottweber, 2021-09-21:
Issue is independent of whether I compile natively or cross compile.
From a centos 7.9 x86_64 based system:
```
make V=1 CROSS_COMPILE=arm-linux-gnu- ARCH=arm32 bin-arm32-efi/ipxe.efi
gcc --version
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```
From an Ubuntu 20.04 arm32 based system:
```
make bin-arm32-efi/ipxe.efi
gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE
```
During linking, the following error is presented:
```
vxge_config.c:(.text.vxge_hw_vpath_rx_doorbell_init+0x12): undefined reference to `readq'
vxge_config.c:(.text.vxge_hw_vpath_rx_doorbell_init+0x30): undefined reference to `writeq'
arm-linux-gnu-ld: bin-arm32-efi/ipxe.efi.tmp: hidden symbol `writeq' isn't defined
arm-linux-gnu-ld: final link failed: Bad value
make: *** [bin-arm32-efi/ipxe.efi.tmp] Error 1
```
Full build output: [build_output.txt](https://github.com/ipxe/ipxe/files/7202683/build_output.txt)

NiKiZe, 2021-09-21:
I think that `readq` is not available on all platforms, try building specific drivers and not all, for example build `snp.efi` or `snponly.efi`

I do know that this (or similar) has been seen somewhere before, unfortunately not able to find it.

mcb30, 2021-09-21:
> I think that `readq` is not available on all platforms, try building specific drivers and not all, for example build `snp.efi` or `snponly.efi`

This is correct.  Some hardware requires atomic 64-bit accesses, and those drivers cannot compile on platforms that don't provide 64-bit I/O operations.

ARM32 is only semi-supported in iPXE: the platform exists, but there are several quirks such as this.  ARM64 is fully supported.

closed 2021-09-21 by NiKiZe

rustyscottweber, 2021-09-21:
Both of those targets work. I'll go give those images a try.  At least this issue exists to help those lost on the google wasteland to better waters.

NiKiZe, 2021-09-21:
Also updated https://ipxe.org/appnote/buildtargets with a note about this.