ipxe/ipxe #477: Feature Request: Allow GPT image partitions to be used with sanboot/sanhook

Bobberty, 2021-09-23:
Pretty simple....
I might be missing something, but it doesn't appear that GPT image partitions are supported by sanboot/sanhook.

I've only had success with MBR partitions.

Failed Image (GPT):
sanboot http://ftp.freebsd.org/pub/FreeBSD/releases/arm64/aarch64/ISO-IMAGES/13.0/FreeBSD-13.0-RELEASE-arm64-aarch64-mini-memstick.img

Successful Image (MBR): 
sanboot https://ftp.openbsd.org/pub/OpenBSD/snapshots/arm64/miniroot70.img

NiKiZe, 2021-09-23:
if you are running in efi mode, then sanboot registers a block device, and tells the firmware to start from that block device, here it is all up to the firmware to read the GPT table and make partitions available and bootable.

If you are running in pcbios mode then sanboot again registers a devices by intercepting INT13 and then starts executing the disk, note that pcbios has no concepts of partitions at all, it just executes instructions.

So are you running in pcbios mode, or efi mode? Regardless I'd say that there is nothing that actually can be fixed here on the iPXE side of things.

Bobberty, 2021-09-23:
It is EFI Mode.
I'm running EDK II on an RPi4.  All netboot.  Probably an edge scenario.  I'm compiling for a bit of EFI Block debugging within IPXE now.

NiKiZe, 2021-09-23:
Ok, so then it is totally up to the firmware, and not really an iPXE issue
does the EDK II build have GPT support?

Bobberty, 2021-09-23:
Yep.
Got to love debugging.  It looks like iPXE does things properly.  The EFI binary may be stuck

closed 2021-09-23 by Bobberty