ipxe/ipxe #517: unlike in ipxe.efi ${next-server} unpopulated in undionly.kpxe

bcb01, 2021-11-13:
Hi guys, 

I need some help to get my BIOS/UEFI PXE server running. I set it up with dnsmasq (proxy mode), ipxe.efi and undionly.kpxe (with embedded script), both using the same boot script later on. Serving works fine in terms of delivering the right file to the specific (BIOS or UEFI) client. From there on it gets a little weird. While ipxe.efi finds its boot script automatically from the DHCP options and after that starts the desired OS (using the ${next-server} variable in the boot script) like intended, undionly.kpxe doesn't find or boot anything at all.
To get this done, I need to specify the PXE server IP and the boot script location in static form in an embedded script, and after that I need to do the same in the boot script to be able to load the OSes. The config tool in the ipxe shell shows the "next-server" and "filename" options populated with an IP and a path in the UEFI version (ipxe.efi), but unpopulated in the BIOS version (undionly.kpxe). So dnsmasq seems to deliver it but undionly.kpxe doesn't receive or interprets it and is therefore unable to use this info later on, which leads to the described behaviour.

This is a bit annoying, because I want to use the PXE server in different networks and don't want to alter the related scripts every time.

Did I miss something or is this the desired behaviour? If so, how can I get around this issue without changing scripts for every new network (DHCP) setting? Would be nice if someone out there could point me in the right direction to get this issue sorted. Thx in advance. 

Chris

bcb01, 2021-11-13:
[image]
dnsmasq.conf

[image]
ipxe shell config (ipxe.efi)

[image]
ipxe shell config (undionly.kpxe)

n-ice-ch, 2021-12-29:
I have a similar issue like bcb01 (but maybe they correspond?).
My environment:
- ASUS Router as dumb DHCP-Server
- Debian 11 (as virtualbox guest on a Windows client) with installed and configured dnsmasq server
- BIOS-Client (an additional virtualbox guest) with PXE boot: ${next-server} shows the ip address from ASUS Router instead of the dnsmasq server VM.
--> And I'm confused as it gets load the undionly.kpxe from the dnsmasq server but later on, it does not remember the correct ip for TFTP server?

NiKiZe, 2022-02-25:
When booting you will have 2 DHCP discover phases.
First BIOS/Firmware that runs PXE boot and sends out the request and where you get ipxe.efi or undionly.kpxe back
If there is no embedded script iPXE will then do a second DHCP discover with option 175 set.

Configuration at https://ipxe.org/appnote/proxydhcp should be working for this.

Have you tried to build iPXE with DEBUG=dhcp and grab the output? Also have you done tcpdump? maybe there is bugs in dnsmasq that makes it not send the correct data, even if it looks like it does in it's output?

bcb01, 2022-04-25:
Nope, didn't try these so far. But will do asap. Thx for now for your hint. :)

NiKiZe, 2022-05-05:
Yes that is what I said, if you take a look at the example I posted on ipxe.org you should be able to see this.

I could take another look if you post your full dnsmasq config. But this is not a bug or issue with iPXE (so closing this issue.)

closed 2022-05-05 by NiKiZe