ipxe/ipxe #523: ISCSI protocol usage issues in UEFI Shell environment

kkimdong, 2021-11-24:
Hi guys,

I need help running my UEFI PXE server.

Booting to UEFI Shell was successful using PXE Bootloader (ipxe.efi ver 1.21.1).

UEFI Shell is a diskless environment, which is connected by the ISCSI protocol.

the problem is

 When copying or saving 1.txt file in UEFI Shell of the client PC, it takes a lot of time, about 2-3 minutes.

Any way to solve this?

 
PS: There is no problem in saving or copying 1.txt file after iSCSI connection in Windows7 environment.

NiKiZe, 2021-11-24:
Use a packet capture to try and figure out what is going on.

kkimdong, 2021-11-24:
> Use a packet capture to try and figure out what is going on.

Hi

In the picture file, I can see that the TCP Protocol is repeatedly displayed slowly on the screen.

What should i do? Please let me know

[image]

mcb30, 2021-11-24:
The UEFI shell is not generally usable for creating files.  I have frequently observed similar problems on other filesystems (e.g. a physically attached USB key).

Why are you trying to create a file on the iSCSI disk from within the UEFI shell?

kkimdong, 2021-11-25:
Hi~
We test Memory like Memtest86 program in environment UEFI Shell. 
I have confirmed that there is no problem when saving or copying files in UEFI Shell environment using iSCSI function in Server's BIOS menu.
Is there a way to copy or save in UEFI Shell after iSCSI protocol connection using PXE?

mcb30, 2021-11-25:
It should in theory work, but the whole block device layer in iPXE is optimised for reading rather than writing.  In a typical OS boot, at most one 512-byte write occurs to the SAN device.  To reduce code size, iPXE doesn't bother trying to make this unusual case fast.

kkimdong, 2021-11-30:
A static IP address of 192.168.0.255 was set on the client PC.
After PXE booting, it was confirmed that when ISCSI is connected, it runs normally.
Only static IP address 192.168.0.255 is running normally
Other IP addresses do not work

Can you find a workaround with the above method?

mcb30, 2021-11-30:
> A static IP address of 192.168.0.255 was set on the client PC. After PXE booting, it was confirmed that when ISCSI is connected, it normal operation. Only static IP address 192.168.0.255 is running normally Other IP addresses do not operation

That IP address does not appear to be valid.  Addresses in 192.168.0.x would most usually be used with a subnet mask of 255.255.255.0.  With that subnet mask, the address 192.168.0.255 is the broadcast address, and cannot be assigned as an individual IP address.

Marking this issue as closed since there does not appear to be any clear topic.

closed 2021-11-30 by mcb30