ipxe/ipxe #529: HPE 1Gb 4-port 331i Adapter / No configuration methods succeeded

fre4ki, 2021-12-01:
Hey guys,

when i try to boot my hpe-server (HPE Ethernet 1Gb 4-port 331i Adapter) with my modified ipxe.efi file, there is an error "No configuration methods succeeded". On VMware it works fine.

[image]

When i try to boot my hpe-server with an prebuild ipxe.efi file, there is no error.

**>> ifstat with original ipxe.efi file (working)**

[image]

**>> ifstat with modified ipxe.efi file (not working)**

[image]

[image]

Hope someone can help to fix this.

Kind regards

Andy

mcb30, 2021-12-01:
To get any meaningful response, you'll need to say exactly what you modified in your "modified ipxe.efi".

fre4ki, 2021-12-01:
> To get any meaningful response, you'll need to say exactly what you modified in your "modified ipxe.efi".

Ohh sure, i just edit the console settings to allow background images and edit the protocol settings.

NiKiZe, 2021-12-01:
I think that is the the driver, which is buggy, and cold boot vs only reboot can give differences. But your tests are not consistent. Please update to show clearly what works and not, use the same commands etc. If possible, also check DHCP server for if requests get there or not.

fre4ki, 2021-12-02:
Coldboot makes no difference. But i see another message now:

[image]

I think it is a driver topic too, but how can i adjust this?

stappersg, 2022-05-02:
In #635 is also about a _HPE 1Gb 4-port 331i Adapter_.