ipxe/ipxe #540: Realtek RTL8125B 2.5GB LAN NIC support

TheLinuxGuy, 2021-12-26:
Could you add support for Realtek RTL8125B 2.5GB LAN NIC?

I've tested with netboot.xyz and it fails to be detected. I believe the driver may need to be added to this project for netboot.xyz team to merge and fix? https://github.com/netbootxyz/netboot.xyz/issues/1062

NiKiZe, 2021-12-26:
Not all NICs have native support in iPXE.
You can usually use undionly.kpxe or snp.efi to still use it.

If native support for for rtl8125b is needed in iPXE, then someone needs to fund its development. Which is done from technicall documentation. And has little to nothing to do with the Linux kernel.

TheLinuxGuy, 2021-12-26:
That's good to know. I asked netboot.xyz folks if there is an .iso that defaults to undionly.kpxe boot - it looks like the only way they documented to use undionly is via DHCP (?)

NiKiZe, 2021-12-26:
Undionly is only usable over dhcp

Wyk72, 2022-02-15:
I have a couple of those NICs and they have a (BIOS/legacy) PXE boot rom that works , you can chainload iPXE or whatever  you like.  I can dump them with a EEPROM reader, but it's .. old school stuff.

I've used them on Alpine Linux w kernel 5.10.x with no issues, chainloading iPXE. 

Newer motherboards from 2021/2022 have a UEFI PXE efi binary inside for those chips: I guess you can extract it from there, with some hacking skills, if really needed. Most boards drom 2020 have direct http efi boot, btw, that makes things a lot easier.

a31amit, 2022-04-28:
It is mentioned above that snponly.efi should work, but in my case with this new motherboard having onboard RTL8125B NIC is stuck on IPXE initializing. snponly.efi is coming from DHCP server . 

Are there suggestions that can be tried? or Is it not even supported with snponly method ?

NiKiZe, 2022-04-28:
> It is mentioned above that snponly.efi should work, but in my case with this new motherboard having onboard RTL8125B NIC is stuck on IPXE initializing. snponly.efi is coming from DHCP server .
> 
> Are there suggestions that can be tried? or Is it not even supported with snponly method ?

snp is the standard EFI driver interface, so if there is a different driver loaded in efi for your nic then iPXE can use that.
Why it hangs can be a number of reasons, must of them are due to buggy efi implementations.

You can try and use DEBUG= builds and try to narrow down where it hangs. but start with making sure you are using the latest code from git master.

stappersg, 2022-05-02:
See also #641

a31amit, 2022-05-09:
Update - I initially thought it may be due to secure boot since motherboard wasn’t allowing me to disable CSM and use legacy boot. Only available option it use with UEFI firmware. I try with grub and shim signed which stuck as well . 

At last option I tried firmware update which solve this issue for me. 

However it’s still a riddle for me even though secure boot enabled and CSM disabled . Booting through snponly.efi is working fine . 

Seems like a buggy efi implementation indeed. 

Thanks for suggestions

notr1ch, 2022-07-01:
I can't seem to get this NIC on my motherboard to boot even with snponly.efi. I was able to disable secure boot and TFTP chain to ipxe successfully but then it fails at various different points - sometimes it will send a DHCP DISCOVER, receive an OFFER but never send a REQUEST. Sometimes it will not resolve my DNS name in the filename. Sometimes it will not connect to the HTTP server address (even if specified by IP). It's almost like it  drops all traffic after a certain point.

Very similar behavior was reported at https://www.illumos.org/issues/13729#note-2 which makes me think this card definitely has some buggy firmware.

stappersg, 2022-08-26:
For those who come along
* Get true technical information about the NIC
* Understand the iPXE code base
* Enable others to help you
* Keep reminding yourself on the goal