ipxe/ipxe #541: iPXE initialising devices hangs for Realtek 8168 in UEFI mode

mforkel, 2021-12-27:
When I try to boot a ThinkPad L530 in UEFI mode with ipxe.efi, it hangs with `iPXE initialising devices...`.

In Linux, the r8169 driver used and the NIC is identified by lspci as
```
0c:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 07)
```
Windows reports a Realtek PCI GbE Family Controller with VEN 10EC, DEV 8168. This signature is identified as ipxe_name rtl8168 in [Hardware expected to work (unconfirmed)](https://ipxe.org/appnote/hardware_drivers).

I compiled a version of ipxe.efi for the current revision 9062544 (25 Nov 2021) with
```
DEBUG="device,runtime,pci,init,nii,snp,snpnet,snponly,efi_watchdog,efi_image,image_cmd,image,efi_autoboot,efi_timer,efi_driver:3,efi_debug,efi_init,efi_pci,efi_wrap,intel,intelxl"
```
Its output is (transcribed from the screenshot, so typos are likely):
```
...
iPXE initialising devices...INIT startup efi_tick...
EFI timer started at 32 ticks per seconds
INIT startup devices
Adding 3c509 root bus
Adding EFI root bus
EFIDRV connecting our drivers
EFIPCI 0000:00:1d.0 (8086:1e26 class 0c0320) has no driver
...
EFIPCI 0000:00:16.1 cannot read PCI configuration: No such deuice (http://ipxe.org/2c044087)
EFIPCl 0000:00:1b.0 (8086:1e20 class 040300) has no driver
EFIPCI 0000:0c:00.0 (10ec:8168 class 020000) has driver "rt18168"
EFIDRU PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) has driver "PCI"
EFIDRU PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) before disconnecting:
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) PciIo supported
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) PciIo opened 3x (H) [Empty debug directory entry] by DxeCore(?) for <NULL>
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) PciIo opened lx (D) by Realtek Ethernet Controller UNDI Driver
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) PciIo opened lx (C) by Realtek Ethernet Controller UNDI Driver for PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0)/MAC(3C970ED18F6A,0x0)
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) PciIo opened lx (G) by Simple Network Protocol Driver for PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0)/MAC(3C970ED18F6A,0x0)
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) PciIo opened 12x (G) [Empty debug directory entry] by MemoryNapped(0xB,0xC618F010,0xC6B9F00F)/FvFile(73DAB30F-3F9C-4160-B064-439CE16C3EDE) for <NULL>
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) DevicePath supported
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) DevicePath opened 47x (H) [Empty debug directory entry] by DxeCore(?) for <NULL>
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) DevicePath opened 28x (G) [Empty debug directory entry] by MemoryNapped(0xB,0xC618F810,0xC6B3F00FFvFile(73DAB30F-3FSC-4160-B064-439CE16C3EDE for <NULL>
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) DevicePath opened 2x (T) by Realtek Ethernet Controller UNDI Driver
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) DevicePath opened lx (D) by Realtek Ethernet Controller UNDI Driver
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) DevicePath opened 2x (T) [Empty debug directory entry] by MemoryNapped(0xB,0xC618F810,0xC6B3F00F)/FvFile(51CCF399-4FDF~4E55-A45B-E123F84D456ft)
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) DevicePath opened 2x (T) by UNKNOWN<0xca3feb18,DriverBinding> 
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) DevicePath opened 2x (T) by Simple Network Protocol Driver
HANDLE PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) DevicePath opened 2x (T) [Empty debug directory entry] by MemoryNapped(0xB,0xC618F010,0xC6B9F00F)/FvFile(864B10FE-3D5E-4908-88CC-F08D4EDC0050)
EFIDRV PciRoot(0x0)/Pci(0xlC,0x3)/Pci(0x0,0x0) disconnecting existing drivers
```

Booting in Legay mode works fine (again, typo alert):
```
Intel UNDI, PXE-2.1 (build 083)
Copyright (C) 1997-2000 Intel Corporation
This Product is covered by one or nore of the following patents:
US5,387,459, US5,434,872, US5,732,094, US6,578,884, USB,115,776 and
US6,327,625

Realtek PCIe GBE Fanily Controller Series v2.41 (06/08/11)
...
PXE->EB: !PXE at 993E:0070, entry point at 993E:0100
         UNDI code segment 993E:328C, data segment 8CF4:C4A0 (563-626kB)
         UNDI device is PCI 0C:00.0, type DIX+802.3
         563kB free base memory after PXE unload
iPXE initialising devices...ok

iPXE 1.0.0+git-20190125.36a4c85-5.1 -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI NFS TFTP Aoe MBOOT PXE bzImage Menu PXEXT

iPXE>
```

This seems to be an old problem, see [Hangs on "initialising devices"](https://forum.ipxe.org/showthread.php?tid=1072).

NiKiZe, 2021-12-27:
Tried snponly.efi? Have you made sure you are using latest firmware? There is many issues with rtl, many revisions and all using same id.

mforkel, 2021-12-27:
Unfortunately, it is the same problem with snponly.efi. I tried the one included in Debian Bullseye (revision 36a4c85, 25 Jan 2019) and a self-built version (revision 9062544, 25 Nov 2021).

Yes, it's the latest firmware from Lenovo, UEFI BIOS 2.76 (G3ETB6WW), 2019-10-04.

What can I do to further identify the NIC?

NiKiZe, 2021-12-27:
By checking snponly.efi, we now know that it is not the rtl driver in iPXE that is the issue, but rather something else.
snponly.efi will only try and load the the nic that it was loaded via. using the NII driver path.

The most likely conclusion to me is that you have either a firmware bug when the path up to this unloaded, or some other firmware bug.

If you where to try a fresh build of intel.efi (which only has the intel driver and not SNI/NII driver) then there would be no driver for your NIC, and there would be no unload of the existing hook for the rtl NIC.
That way you should be able to narrow down what actually fails.

In terms of fix, I have no idea yet.

mforkel, 2021-12-27:
Thanks for your help! I freely admit that this is beyond my level of expertise. So I hope I did the right thing and the results are helpful.

I compiled a version of intel.efi from revision 9062544 with `DEBUG="device,runtime,pci,init,nii,snp,snpnet,snponly,efi_watchdog,efi_image,image_cmd,image,efi_autoboot,efi_timer,efi_driver:3,efi_debug,efi_init,efi_pci,efi_wrap,intel,intelxl"`. This is the output I get with Ctrl-B (all typos on me):
```
...
EFI PciRoot(0x0)/Pci(Ox1C.0x3)/Pci(0x0,0x0)/MAC(3C970ED18F6A,0x0)/IPv4(0.0.0.0) found autoboot link-layer address:
00000000 : 3c 97 Oe d1 8f 6a                               : <....j
iPXE initialising devices...INIT startup efi_tick...
EFI tiner started at 32 ticks per second
INIT startup devices...
Adding EFI root bus
EFIDRV connecting our drivers
EFIPCI 0000:00:1d.0 (8886:1e26 class 0C0326) has no driver
EFIPCI 0000:00:1a.0 (8086:1e2d class 0c0320) has no driver
EFIPCI 0000:00:14.0 (8086:1e31 class 0C0330) has no driver
EFIPCI 0000:00:1c.0 (8086:1e10 class 060400) has no driver
EFIPCI 0000:01:00.0 (10ec:522S class ff0000) has no driver
EFIPCI 0000:00:02.0 (8086:0166 class 030000) has no driver
EFIPCI 0000:00:1f.0 (8086:1e59 class 060100) has no driver
EFIPCI 0000:00:1c.1 (8086:1e12 class 060400) has no driver
EFIPCI 0000:00:1c.2 (8006:1e14 class 060400) has no driver
EFIPCI 0006:00:1c.3 (8086:1e16 class 060400) has no driver
EFIPCI 0000:00:1f.2 (8806:1603 class 016601) has no driver
EFIPCI 0000:06:00.0 (10ec:8176 class 028006) has no driver
EFIPCI 0000:00:00.0 (8086:0154 class 060000) has no driver
EFIPCI 0000:00:16.0 (8086:1e3a class 078000) has no driver
EFIPCI 0000:00:16.1 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:00:1b.0 (8886:1e20 class 040300) has no driver
EFIPCI 0000:0c:00.0 (10ec:8168 class 020000) has no driver
EFIPCI 0000:00:1f.3 (8006:1e22 class 0C0500) has no driver
EFIPCI 0000:00:1f.6 cannot read PCI configuration: No such device (http://ipxe.org/2c044e07)
Adding PCI root bus
INIT startup efi_autoexec...
INIT startup cachedhcp...
INIT startup conplete
ok

iPXE 1.21.1+ (g90625) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP AoE EFI Menu
iPXE>
```

NiKiZe, 2021-12-27:
And if you compare that to a run with snponly.efi built from same source and with same options?

mforkel, 2021-12-27:
I have a problem capturing the output. There is way more text than fits on one screen And the resolution of my digital camera isn't good enough, so I can't read the text in the video i made.

Anything I should try to look for in the output? Or a suggestion how to capture it? An option to save the output to a file and maybe even send it somewhere would be nice :-)