ipxe/ipxe #542: Wrong UUID again (raw instead of pretty print)

n-ice-ch, 2021-12-30:
The UUID on Virtualbox get obtained in the wrong format.
And I guess there are other situations where the UUID from iPXE is the wrong formatted.

As nowadays all relevant system do use the pretty format for UUID, we should support that in iPXE too.

Microsoft can explain it here:

- https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-2000-server/cc978312(v=technet.10)?redirectedfrom=MSDN

There was already some code commit but unfortunatley this seems to be removed in the latest sources?

- https://github.com/ipxe/ipxe/commit/9e896d0

May can I ask to commit a final solution or at least put additional values into the settings?
${uuid}
${uuid_raw}
${uuid_conv}

NiKiZe, 2021-12-30:
Do you have a reproducible example that uses the latest version of iPXE?

n-ice-ch, 2021-12-30:
> Do you have a reproducible example that uses the latest version of iPXE?

With pleasure.

How can I provide you the information?
What you need exactly?
How can I generate DEBUG information?

I can only see which UUID get displayed with command CONFIG or by ECHO ${uuid}
My environment is a virtualbox guest V6.1.30.

NiKiZe, 2021-12-30:
> With pleasure.
>
> How can I provide you the information? What you need exactly? How can I generate DEBUG information?
>
> I can only see which UUID get displayed with command CONFIG or by ECHO ${uuid} My [...]

I would expect that you give a command line for vbox with guid, and iPXE showing the uuid output using echo.
Preferably over serial or similar so that no UI is needed.

Just a sample that shows the issue and is easy for anyone to reproduce.

n-ice-ch, 2021-12-31:
[pxe-client.vdi.zip](https://github.com/ipxe/ipxe/files/7796308/pxe-client.vdi.zip)
[undionly.zip](https://github.com/ipxe/ipxe/files/7796309/undionly.zip)
[image]
[image]

mcb30, 2021-12-31:
> [image]

Your first screenshot shows iPXE retrieving the same UUID as the built-in PXE client, and therefore demonstrates that iPXE is behaving correctly.  The second screenshot shows that `VBoxManage.exe` is displaying the incorrect (i.e. byte-swapped) UUID.

Closing since this is an error in VirtualBox, not in iPXE.

NiKiZe, 2021-12-31:
So how do I start vbox from command line and define uuid on that command line? To reproduce what you are seeing?

You need to explain how to create this scenario so that I or someone can get the same output.
That is what reproducible mean.

n-ice-ch, 2021-12-31:
@NiKiZe 
Take the attached VM, run it on your virtualbox and process a PXE boot with the attached undionly.kpxe.
You are free to install a Microsoft Windows OS on the VM and check the obtained UUID.
--> The command VBoxManage.exe can be used to get the UUID (in coverted and "correct" form.

@mcb30 
It is really a pity that you do not want to see the issue and the and what this means if you want to use iPXE in general.
The UUIDread by iPXE is in raw format. This is basically not wrong but unusual as de-facto industry standard is the converted format:
- Microsoft SCCM
- Matrix42 Empirum
- Altiris

You can turn it any way you want, but with this iPXE cannot be used to identify devices already at PXE startup. 

Fortunately it is open source. So I can get the UUID output correctly (converted format) by changing the source code.
But others will struggle with that fact of obtained raw UUID.

NiKiZe, 2021-12-31:
I expect that I can do this by running from the command line without downloading some 3rd party binary!

Did you ignore that iPXE does the exact same thing as the Intel PXE ROM? By doing differently we would introduce a different inconsistency.

n-ice-ch, 2021-12-31:
@NiKiZe 
To run a VM you need a 3rd party product.
I do not ignore that Intel PXE ROM does show up the **`RAW`** uuid value.
But it is a fact that the converted one gets used to identify the device by de-facto industry standard solution.

berlin4apk, 2021-12-31:
@n-ice-ch have you a link to the RFC about the uuid formatting / displaying?

mcb30, 2021-12-31:
> **Request for change** Add the value ${uuid_conv} to iPXE.

@n-ice-ch Could you be quite a lot more specific about what you are trying to achieve?

From the way that you are discussing the issue, it seems unlikely that you are concerned with the byte ordering of the 16-byte UUID within the DHCP request packet.  Adding your `${uuid_conv}` value (or, more accurately, adding a `:leuuid` setting type to allow existing UUID settings to be formatted within scripts as little-endian values using e.g. `${uuid:leuuid}`) would not change anything about the DHCP request packet contents; it would only provide a way to choose to (mis)format the UUID within an iPXE script.

However, the "de-facto industry standard solutions" that you quote will generally be using the DHCP request packet, and will not know anything about iPXE scripts.  In this context, your request for a `${uuid_conv}` value makes no sense.

Please clarify **precisely** what it is that you consider to be incorrect behaviour, using terminology that is much  precise than "pretty print".  From what you have written so far, it is very unclear to me what you think that your words mean.

n-ice-ch, 2022-01-02:
@mcb30 
I do use iPXE to deploy an operating system on computers.

Steps:
- Load iPXE from dnsmasq server
- Chainload a webservice (php): 
-- Check for known UUID. If not, register a new device.
-- Check for a pending os-deployment action
-- Load WinPE or Linux to do the os-deployment action

To make sure that I can correctly identify a device, I do use the UUID. As I'm coming from the Microsoft world, the UUID has to be in the reference like Microsoft is using it.
--> And that seems to be **`MY MISTAKE!`**

I know the UUID discussion about RAW and littleendian format already. 
In the past I had a job to build up inventory database. And thre we had always the same discussion about the UUID.

I see that iPXE tries to support the official RFC and it makes sense for me about your last comments.

Therefore I suggest to add ${uuid:leuuid} to make sure, the UUID can get obtained in this format and the original iPXE users get not disturbed.

NiKiZe, 2022-01-02:
Start Linux on the same machine, run dmidecode, what do you get? And how do that compare to the vale that you set when you created the machine?

n-ice-ch, 2022-01-04:
@NiKiZe 
I get the same result as with iPXE. 
As I write in my last comment: 
- I am trapped in the world of Microsoft and thought that this must be true for everyone else as well.

I have changed the source code for iPXE to deliver me the UUID in the format which Microsoft is using.
An additional variable **`${uuid:leuuid}`** would help me for the future of using iPXE in a Microsoft environment.

mcb30, 2022-01-04:
@n-ice-ch As of commit https://github.com/ipxe/ipxe/commit/f43c2fd69 you can use `${uuid:guid}`.

closed 2022-01-04 by mcb30

n-ice-ch, 2022-01-18:
Thank you very much for the commit!
It works like expected now and I can support the Microsoft world as well.

Important: I learned the difference between the UUID and GUID and will handle that different than in the past years.