ipxe/ipxe #554: iPXE fails to properly parse "relative reference" URIs with a colon

ignatk, 2022-01-13:
We have some relative reference URIs in the form:

`/foo/bar/<some mac address>/boot.ipxe`

Because mac addresses have colons, it seems iPXE has trouble parsing such an URI in a relative form.

When we used absolute URIs, like `https://<address>/foo/bar/<some mac address>/boot.ipxe`, it worked fine.

Here is an example of a failed boot (iPXE compiled with `DEBUG=open`):

```
INTF 0x7dfab3a8+30 attempted to open unsupported URI scheme "/overrides/macs/98"
Could not start download: Operation not supported (http://ipxe.org/3c092083)
Could not boot: Operation not supported (http://ipxe.org/3c092083)
```

so it seems iPXE treats anything before the first colon as an URI scheme.

Based on my understanding of https://datatracker.ietf.org/doc/html/rfc3986#section-4.2 generally allows colons in relative reference URIs, so I believe it is a bug.

NiKiZe, 2022-01-13:
I would agree reading section 3.3 https://www.rfc-editor.org/rfc/rfc3986#section-3.3

However recommended use is `${net0/mac:hexhyp}` as mentioned in https://ipxe.org/cfg/mac#notes
You might see more places where : is not handled correctly in paths.

ignatk, 2022-01-13:
Thanks for the pointer. We actually switched to `hexraw` without any colons/hypens (most likely because I forgot `hexhyp` format existed)

mcb30, 2022-01-13:
@ignatk Thanks for the report - should now be fixed.

PR #555 (2022-01-13): [uri] Allow for relative URIs that include colons within the path
RFC3986 allows for colons to appear within the path component of a
relative URI, but iPXE will currently parse such URIs incorrectly by
interpreting the text before the colon as the URI scheme.

Fix by checking for valid characters when identifying the URI scheme.
Deliberately deviate from the RFC3986 definition of valid characters
by accepting "_" (which was incorrectly used in the iPXE-specific
"ib_srp" URI scheme and so must be accepted for compatibility with
existing deployments), and by omitting the code to check for
charac [... cut]
closed 2022-01-13 by mcb30