ipxe/ipxe #557: missing settings.h

lgwapnitsky, 2022-01-16:
Just did a fresh pull from git to build, and I"m receiving the following:

```
In file included from config/config.c:27:
./config/settings.h:21:10: fatal error: config/local/settings.h: No such file or directory
   21 | #include <config/local/settings.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

```

Any ideas as to what's going on? The repo doesn't have the file, either.

NiKiZe, 2022-01-16:
Actually same as #202

`config/local/*.h` is automatically generated if missing, but it does require that make can identify them as missing.
Might be a timing issue, that local time is in the past compared to git clone date or similar.

One thing that almost always works is to create new clone.

I will be closing since this is not an issue with iPXE, but rather with the environment.
Will be happy to understand why this happens to get better documentation tho.

closed 2022-01-16 by NiKiZe

lgwapnitsky, 2022-01-16:
tried again, failed. Looks as though it's an issue compiling on Alpine. Got it working on my Ubuntu instance.

Thanks!

NiKiZe, 2022-01-16:
Local fs and not NFS?

Could you verify timestamps in config?
Did you modify anything before first build? Or maybe the first build was canceled?

These files should be created automatically. If they don't you can use touch to create the missing empty files.