ipxe/ipxe #584: 8168 rev 0c with iPXE cannot get DHCP after reboot

zboszor, 2022-02-07:
I have two different hardware that use 8168 (10ec:8168 rev **0c**) that they can remote boot properly with iPXE only if they were powered down previously but not after reboot. This happens after a reboot on them:
```
Configuring (net0: <MAC address> )...................... No configuration methods succeeded (http://ipxe.org/040ee119)
No more network devices

Press Ctrl-B for the iPXE command line...
```
Another hardware that also uses 8168 (10ec:8168 rev **07**) can do remote boot and reboots properly any number of times.

NiKiZe, 2022-02-07:
Other 8168 issues not same, (but for documentation and reference)
#145 Rev 07 slow, Rev 06 ok
#389 Rev 04 slow
#541 Rev 07 EFI hang

I have seen/heard similar issues several times before, but have not taken notice of rev.

How do you start iPXE on these machines? does undionly.kpxe work, but ipxe.pxe fail? or which exact build target and version are you using?
You might want to try and follow up with a debug build of the realtek driver. (you can find more debug steps in the issues above, even if they do not provide you with an answer to the issue itself)

zboszor, 2022-02-07:
I only used `ipxe.lkrn`. I tried several commits including the current latest.
But I found a workaround: I need to run `ethtool -s <ifname> wol g` and this allows iPXE to reboot and get IP via DHCP.

steverweber, 2022-03-31:
not sure if it helps but in the past I think i hit a bug like this this might have helped.
```
ifclose ||
sleep 1
# not sure this ifopen would do anything here because dhcp likely does ifopen.
ifopen ||
sleep 1
dhcp
```

paulhutchings, 2022-04-18:
Not sure if I should make a new issue or add to this one, but I have been encountering the same problem. Mine is `10ec:8168` rev **16**.

I found [this forum post](https://forum.ipxe.org/showthread.php?tid=7216) from several years ago that is the _exact_ issue, right down to the difference between (I believe it's one of the PHY registers?) `01e1` (success) and `0de1` (fail) when I record with a debug build of ipxe. That thread resulted in [this fix](https://github.com/ipxe/ipxe/commit/ccb6e5c) being pushed.

I've also observed the same behavior when manually setting the network config via:
```
set net0/ip <ip_address>
set net0/netmask 255.255.255.0
set net0/gateway <router_ip>
set net0/dns <router_ip>
```
so I don't believe this to be a problem specific to DHCP (but would be happy to be proven wrong).

A couple of things I don't get:
 - That fix for 8168 was almost 10 years ago, are the newer revs not behaving the same (that fix doesn't work for them), or was there a regression somehow since then?
 - The OP of that thread indicated they were able to fix the issue at first by changing driver settings in Windows. I'm only using Linux on this machine, so I'm not sure if it's possible to do the same workaround?

The next thing I wanted to do was look at those lines from that linked commit and see if anything has changed since it was added and possibly build from an older snapshot of the code to see if that changes anything. Does that seem reasonable? Low-level software is far from my area of expertise or knowledge, but I'm happy to try and debug and provide information as best I can.

cvtsi2sd, 2023-05-18:
RTL8168 rev 0c, ipxe.lkrn through syslinux, exact same problem. I can confirm that it's not DHCP-related, the problem is there even in my static-IP configuration case.

Checking with Wireshark, I can see that during the TFTP get attempt the only packets that arrive to the server are ARP WHO-HAS, to which it dutifully replies, but seem to get lost to the client NIC, which continues to retry until timeout.

I tried the `ethtool` WoL trick, it seems to work, but just for the next reboot (i.e. it doesn't seem to be a permanent setting); IDK if that was implied by the previous comment.

zboszor, 2023-05-18:
> I tried the `ethtool` WoL trick, it seems to work, but just for the next reboot (i.e. it doesn't seem to be a permanent setting); IDK if that was implied by the previous comment.

It is temporary. The r8169 kernel driver disables WoL, even if it was enabled by the firmware.

You need to put the quoted command in a system script like rc.local or an equivalent that runs during boot.