ipxe/ipxe #593: RTL8111/8168/8411 in ZOTAC CI331 nano and CI329 nano

MAButz, 2022-02-16:
Hello,
I have tested the ZOTAC CI331 nano in pure UEFI Mode (the is no legacy Mode more available) to boot via TFTP a Linux Image, with the new iPXE Version compiled from git, i have the following problem:

After the iPXE Menu at the Point to load the vmlinuz a have the TFTP is requesting the vmlinux file, but at the TFTP-Server has no request for this file and a moment later i have the following error:  
-> Connection timed out 

The same Setup with the ZOTAC CI329 nano in pure UEFI (with the iPXE Version 1.21.2) works perfect.
When i use the new iPXE Version compiled from git with the ZOTAC CI329 nano, i have also -> Connection timed out 

Any idea?
Thanks

NiKiZe, 2022-02-16:
Please post the full error message, including any URL to ipxe.org
Test both ipxe.efi and snponly.efi or snp.efi

Please be clearer on what works and not you say that it works perfectly, and then that it don't. When does it do what?

MAButz, 2022-02-16:
Ok with the ipxe.efi and snponly.efi (build form git with DEBUG=open,tftp,nii,snponly,init,efi_image,image_cmd,efi_driver,efi_debug,realtek)
this works NOT with CI331 and CI329

The DEBUG output:
INTF 0x6fe4bba8+30 opening tftp URI
tftp://10.0.20.5/ltsp/amd64/vmlinuz...TFTP 0x6fe4bc28 requesting "ltsp/amd64/vmlinuz"
INTF 0x6fe4bc28+38 opening (SOCK_DGRAN.AF_INET) socket
TFTP 0x6fe4bc28 requesting "ltsp/amd64/vmlinuz"
TFTP 0x6fe4bc28 requesting "ltsp/amd64/vmlinuz"
TFTP 0x6fe4bc28 requesting "ltsp/amd64/vmlinuz"
.TFTP 0x6fe4bc28 requesting "ltsp/amd64/vmlinuz"
..TFTP 0x6fe4bc28 requesting "ltsp/amd64/vmlinuz"
....TFTP 0x6fe4bc28 requesting "ltsp/amd64/vmlinuz"
........TFTP 0x6fe4bc28 finished with status -1276272786 (Connection timed out (https://ipxe.org/4c26092))
 Connection timed out (https://ipxe.org/4c26092))

NiKiZe, 2022-02-16:
But before this DHCP works as expected?
I read it as this not working at all, regardless of hardware.

Please verify the message, the URL is not valid which I assume is typo mistake.

MAButz, 2022-02-16:
Oh sorry, yes Typo
The correct URL https://ipxe.org/4c126092

Screenshot Form the Output.
[image]

The Procedere.
After Turn on PC, DHCP give an IP and the tftp send the snponly.efi then Shows die ipxe menu and then should load the Linux vmlinuz, this runs in timeout.

Thanks

NiKiZe, 2022-02-16:
Have you tried tcpdump or similar to see what comes out the wire?
Also after the timeout what does `ifstat` show?

MAButz, 2022-02-17:
Hi, no i have not tried tcpdump on the (TFTP-Server) to see the requests from the client.
Ok, i would build a setup to catch the ETH-Frames (tomorrow) with wireshark.

At first i would try (tomorrow) ifstat after the timeout.

A comment, with the snponly.ipxe from the iPXE Version 1.21.2 it works with the ZBOX CI329 nano box.
But with the same snponly.ipxe from the iPXE Version 1.21.2 with the ZBOX CI331 nano box, it runs in the "Connection timeout".

At this Time i have thinking about a firmware bug from the ZBOX, but the newest snponly.ipxe (build from git) it works NOT with the ZBOX CI329 nano and with the ZBOX CI331 nano.
  
Thanks

mcb30, 2022-02-17:
> A comment, with the snponly.ipxe from the iPXE Version 1.21.2 it works with the ZBOX CI329 nano box. But with the same snponly.ipxe from the iPXE Version 1.21.2 with the ZBOX CI331 nano box, it runs [...]

As a general suggestion: whenever you have something that works with an older build but not with a newer build, then try using https://ipxe.org/howto/bisect to identify the commit that introduced the problem.

MAButz, 2022-02-17:
Ok, that is i good idea, i will test it.

MAButz, 2022-02-18:
> Have you tried tcpdump or similar to see what comes out the wire? Also after the timeout what does `ifstat` show?

```
iPXE> ifstat
neto: 00:01:2e:96:cb:cc using NII on NII-0000:02:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:1 RX:0 RXE:0] 
  [Link status: Unknown (https://ipxe.org/1a086194)] 
  [TXE: 1 x "Network unreachable (https://ipxe.org/28086090)"]
iPXE> 

```
I have build with DEBUG=netdevice, and now the output:

```
Downloading NBP file...

 NBP file downloaded successfully.

iPXE initialising devices...NETDEV neto registered (phys NII-0000:01:00.0 hwaddr 00:01:2e:a2:cc:85)

NETDEV net0 transmission 0x6da51290 failed: Network unreachable (https://ipxe.org/28086090)

ok

iPXE 1.21.1+ (gbc5c) --Open Source Network Boot Firmware - https://ipxe.org/ Features: DNS HTTP iSCSI TFTP AOE EFI 
tftp://10.0.20.5/1tsp/amd64/vmlinuz........................... Connection timed out (https://ipxe.org/4c126092)

Booting failed, dropping to shell 
iPXE> 
```

At this point i will try git bisect, i think network dump is unnecessary.

MAButz, 2022-02-18:
> As a general suggestion: whenever you have something that works with an older build but not with a newer build, then try using https://ipxe.org/howto/bisect to identify the commit that introduced [...]

The output of the end of the bisection process:

```
6ba671acd922ee046b257c5119b8a0f64d275473 is the first bad commit
commit 6ba671acd922ee046b257c5119b8a0f64d275473
Author: Michael Brown <mcb30@ipxe.org>
Date:   Mon Jan 17 16:17:17 2022 +0000

    [efi] Attempt to fetch autoexec script via TFTP
    
    Attempt to fetch the autoexec.ipxe script via TFTP using the PXE base
    code protocol installed on the loaded image's device handle, if
    present.
    
    This provides a generic alternative to the use of an embedded script
    for chainloaded binaries, which is particularly useful in a UEFI
    Secure Boot environment since it allows the script to be modified
    without the need to sign a new binary.
    
    As a side effect, this also provides a third method for breaking the
    PXE chainloading loop (as an alternative to requiring an embedded
    script or custom DHCP server configuration).
    
    Signed-off-by: Michael Brown <mcb30@ipxe.org>

 src/interface/efi/efi_autoexec.c | 176 +++++++++++++++++++++++++++++++++++++++
 1 file changed, 176 insertions(+)
```

NiKiZe, 2022-02-18:
Ifstat showed no received packets. Does DHCP work?
Do you have any autoexec.ipxe file on your tftp server? Have you checked the logs of the tftp server?

MAButz, 2022-02-18:
yes DHCP works. yes i have an autoexec.ipxe file on the tftp server.
In the autoexec.ipxe file are the path to vmlinux file to load, where the connection lost.
On the tftp server, I see the request and send of the autoexec.ipxe, but later there is no request ob the vmlinuz file

NiKiZe, 2022-02-18:
From your ifstat output there is no received packets, so there is no DHCP packets comming in.

If you have autoexec.ipxe, then you need dhcp or at least ifopen to open the interface and get ip.

In your failure case I think you will find that iPXE does not have any IP at all.

MAButz, 2022-02-21:
Now i have a working Version with the ZOTAC ZBOX CI331 nano and CI329 nano.
With this Version https://github.com/ipxe/ipxe/tree/ec746c00012317ba228d7043fa141d5e6a48c17c, is the Version selected from the bisection process.
And now i use the ipxe.efi file and not the snponly.efi!

Thank for your Help

NiKiZe, 2022-02-21:
As I understand it you have a file named autoexec.ipxe on your tftp server.
What changed is that autoexec.ipxe is downloaded during the firmware process, this generates an embedded script, and you do not run dhcp inside that script.

ways to make this work is
* add dhcp to your autoexec.ipxe file
* change the filename to something other than autoexec.ipxe

The behavior is intended. If none of the proposed changes does work, please let us know and we will re-open.

closed 2022-02-21 by NiKiZe

steverweber, 2022-03-05:
i'll seeing some strangeness with `autoexec.ipxe`

I'm building latest as of `Mar 05 ` with 
```
make bin-x86_64-efi/ipxe.efi EMBED=init.ipxe
```

tftp logs
```
Mar 05 17:39:03 pxe-p02 in.tftpd[38515]: RRQ from 129.97.179.250 filename fast
Mar 05 17:39:08 pxe-p02 in.tftpd[38517]: RRQ from 129.97.179.250 filename autoexec.ipxe
```

Thought to myself cool I can use autoexec.ipxe to do some testing/debug my `EMBED=init.ipxe`

Tried something simple...
```
root@pxe-p02:/tmp/ipxe/src# cat /srv/tftp/autoexec.ipxe 
#!ipxe
:woot
echo ooooooooooooooo
goto woot
```

However it seems that the `autoexec.ipxe` script is ignored because my `EMBED=init.ipxe` runs before it.
Perhaps that's by design?.

Also it would be nice if there was a define to disable autoexec.ipxe because im not using it and it makes my tftp logs messy :)

NiKiZe, 2022-03-05:
> i'll seeing some strangeness with `autoexec.ipxe`
> 
> I'm building latest as of `Mar 05 ` with
> 
> ```
> make bin-x86_64-efi/ipxe.efi EMBED=init.ipxe
> ```
> 
> tftp logs
> 
> ```
> Mar 05 18:02:40 pxe-p02 in.tftpd[38573]: RRQ from xxx.97.179.250 filename fast
> Mar 05 18:02:40 pxe-p02 in.tftpd[38573]: tftp: client does not accept options
> Mar 05 18:02:40 pxe-p02 in.tftpd[38574]: RRQ from xxx.97.179.250 filename fast
> Mar 05 18:02:40 pxe-p02 in.tftpd[38575]: RRQ from xxx.97.179.250 filename autoexec.ipxe
> Mar 05 18:02:40 pxe-p02 in.tftpd[38575]: tftp: client does not accept options
> Mar 05 18:02:40 pxe-p02 in.tftpd[38576]: RRQ from xxx.97.179.250 filename autoexec.ipxe
> ```
> 
> Thought to myself cool I can use autoexec.ipxe to do some testing/debug my `EMBED=init.ipxe`
> 
> Tried something simple...
> 
> ```
> root@pxe-p02:/tmp/ipxe/src# cat /srv/tftp/autoexec.ipxe 
> #!ipxe
> :woot
> echo ooooooooooooooo
> goto woot
> ```
> 
> However it seems that the `autoexec.ipxe` script is ignored because my `EMBED=init.ipxe` runs before it. Perhaps that's by design?.
> 
> Also it would be nice if there was a define to disable autoexec.ipxe because im not using it and it makes my tftp logs messy :)

You should probably create a new issue about way to disable autoexec.ipxe