ipxe/ipxe #606: Certificate validation fails when the last certificate of the chain is signed by an expired CA (DST Root CA X3)

sbraz, 2022-03-02:
Hi,
It looks like iPXE behaves [like older OpenSSL versions](https://www.openssl.org/blog/blog/2021/09/13/LetsEncryptRootCertExpire/) when encountering an expired CA at the end of a chain of trust.

When it attempts to download an image from a server with this chain (which I believe is still Let's Encrypt's default [because of Android devices](https://letsencrypt.org/2020/12/21/extending-android-compatibility.html)), it will fail:
```
Certificate chain          
 0 s:CN = example.com
   i:C = US, O = Let's Encrypt, CN = R3
 1 s:C = US, O = Let's Encrypt, CN = R3
   i:C = US, O = Internet Security Research Group, CN = ISRG Root X1
 2 s:C = US, O = Internet Security Research Group, CN = ISRG Root X1
   i:O = Digital Signature Trust Co., CN = DST Root CA X3
```

I would expect iPXE to check that the `R3` certificate was signed by a trusted authority (the `X1` CA is present in the official list of cross-signed certificate as `c92eb62f.der`) and therefore consider that the chain is valid.

Instead, it downloads the cross-signed `X3` CA (`0e4aff85.der`, which was last cross-signed on 2021-12-01 and expired on 2022-03-01) and fails validation.

Logs:

<details>

```
CERTSTORE added certificate example.com
X509 chain 0x3e76c748 added X509 0x3e770008 "example.com"
CERTSTORE added certificate R3
X509 chain 0x3e76c748 added X509 0x3e7758c8 "R3"
CERTSTORE added certificate ISRG Root X1
X509 chain 0x3e76c748 added X509 0x3e776808 "ISRG Root X1"
VALIDATOR 0x3e771de8 "example.com" validating X509 chain 0x3e76c748
X509 chain 0x3e76c748 found no usable certificates
VALIDATOR 0x3e771de8 "example.com" downloading "ISRG Root X1" cross-signature from http://ca.ipxe.org/auto/0e4aff85.der?subject=MD8xJDAiBgNVBAoTG0RpZ2l0YWwgU2lnbmF0dXJlIFRydXN0IENvLjEXMBUGA1UEAxMORFNUIFJvb3QgQ0EgWDM=
VALIDATOR 0x3e771de8 "example.com" transfer complete
CERTSTORE added certificate DST Root CA X3
X509 chain 0x3e76c4a8 added X509 0x3e76efa8 "DST Root CA X3"
VALIDATOR 0x3e771de8 "example.com" found certificate DST Root CA X3
CERTSTORE added certificate iPXE cross-signing CA
X509 chain 0x3e76c4a8 added X509 0x3e775008 "iPXE cross-signing CA"
VALIDATOR 0x3e771de8 "example.com" found certificate iPXE cross-signing CA
CERTSTORE added certificate iPXE root CA
X509 chain 0x3e76c4a8 added X509 0x3e777948 "iPXE root CA"
VALIDATOR 0x3e771de8 "example.com" found certificate iPXE root CA
X509 chain 0x3e76c748 added X509 0x3e76efa8 "DST Root CA X3"
X509 chain 0x3e76c748 added X509 0x3e775008 "iPXE cross-signing CA"
X509 chain 0x3e76c748 added X509 0x3e777948 "iPXE root CA"
X509 0x3e76efa8 "DST Root CA X3" has expired (at time 1646239118)
X509 0x3e777948 "iPXE root CA" is a root certificate
X509 0x3e775008 "iPXE cross-signing CA" successfully validated using issuer 0x3e777948 "iPXE root CA"
X509 0x3e76efa8 "DST Root CA X3" has expired (at time 1646239118)
VALIDATOR 0x3e771de8 "example.com" freed
 Permission denied (http://ipxe.org/0216e48f)
CERTSTORE removed certificate example.com
CERTSTORE removed certificate R3
CERTSTORE removed certificate ISRG Root X1
CERTSTORE removed certificate DST Root CA X3
CERTSTORE removed certificate iPXE cross-signing CA
CERTSTORE removed certificate iPXE root CA
```
</details>

As a side note, would it be possible to publish the scripts used to create the cross-signed `DER` files? [The documentation](https://ipxe.org/crypto) does not mention the fact that ASN.1 sets are used.

I eventually found a way to dump the `DER` files thanks to [an old ML thread](https://lists.ipxe.org/pipermail/ipxe-devel/2015-May/004202.html) but a similar question [had been left unanswered on the forums](https://forum.ipxe.org/showthread.php?tid=8170).

NiKiZe, 2022-03-02:
I assume that it is easy to reproduce this cert chain?
But can it be made available for testing in some way?

In regards to scripts, There is #126 it has been said that the current scripts are not in a state that is suitable to be made public, but hopefully that has changed.

aarnaud, 2022-03-02:
Is there a workarround ?
I tried to add the letsencrypt  ISRG Root X1 in CA and TRUST during compilation but I still have the same issue:

[image]

sbraz, 2022-03-02:
> I assume that it is easy to reproduce this cert chain?

Yes, any Let's Encrypt certificate deployed with `certbot` should use it.

> But can it be made available for testing in some way?

I'll see what I can do tomorrow.

> Is there a workarround ?

If you have control over the server, I'd suggest switching to the shorter trust chain (for `certbot`, it can be done with `--preferred-chain "ISRG Root X1"`).
Otherwise, I doubt this is a great idea but you could probably increase the timestamp error margin:
https://github.com/ipxe/ipxe/blob/1192edf394aaebde666c6aafda1c8b782b08a845/src/config/crypto.h#L54

Or you could host your own copy of the cross-signed certificates but that requires significantly more work.

aarnaud, 2022-03-02:
```
export CERT=ca.pem,isrgrootx1.pem,lets-encrypt-r3.pem
export TRUST=ca.pem,isrgrootx1.pem,lets-encrypt-r3.pem
curl -s http://ca.ipxe.org/ca.crt > src/ca.pem
curl -s https://letsencrypt.org/certs/isrgrootx1.pem > src/isrgrootx1.pem
curl -s https://letsencrypt.org/certs/lets-encrypt-r3.pem > src/lets-encrypt-r3.pem
```
With the Intermediate Certificates R3 it works, for a workaround.

The fix will be:
> I would expect iPXE to check that the R3 certificate was signed by a trusted authority (the X1 CA is present in the official list of cross-signed certificate as c92eb62f.der) and therefore consider that the chain is valid.

aarnaud, 2022-03-02:
I think in crosscert we have to sign the `isrg-root-x1` that was signed by `DST Root CA X3`
- https://letsencrypt.org/certs/isrg-root-x1-cross-signed.pem
and not
- https://letsencrypt.org/certs/isrgrootx1.pem that is selfsign

if we sign the `isrgrootx1.pem` that is selfsign iPXE try the other [Certification Paths](https://www.ssllabs.com/ssltest/analyze.html?d=boot.ipxe.org&s=54.246.183.96&latest) that ending on `DST Root CA X3` that is expired

sbraz, 2022-03-03:
> But can it be made available for testing in some way?

Please try the following script:
```
#!ipxe

chain https://sbraz.xyz/refind/refind_x64.efi
```

stappersg, 2022-03-03:
Because I was curious about the `refind_x64.efi`, I did a test:
```text
$ cd /tmp
stappers@paddy:/tmp
$ wget --quiet https://sbraz.xyz/refind/refind_x64.efi
stappers@paddy:/tmp
$ file refind_x64.efi 
refind_x64.efi: MS-DOS executable PE32+ executable (EFI application) x86-64, for MS Windows
stappers@paddy:/tmp
$ ls -lh refind_x64.efi 
-rw-r--r-- 1 stappers stappers 249K 14 mrt  2021 refind_x64.efi
stappers@paddy:/tmp
$ 
```
And want to share this [refind link](http://www.rodsbooks.com/refind/) with you.

sbraz, 2022-03-03:
@stappersg I'm not sure what your test does.
I realise my link could be scary, just FYI I downloaded the EFI application from the site you linked:
```shell
$ curl -sS https://downloads.sourceforge.net/project/refind/0.13.2/refind-bin-0.13.2.zip -L | bsdtar Oxf - refind-bin-0.13.2/refind/refind_x64.efi | sha512sum 
79d9bca3fff15b219eabe66b82d0f46e4b0d980913ad94e0d09d0fffbe1bf5fe1eb4e64e15cf9f84fb87302cf244534337ec29b9e920a468ee03c1843e2f9a96  -
$ curl -sS https://sbraz.xyz/refind/refind_x64.efi | sha512sum
79d9bca3fff15b219eabe66b82d0f46e4b0d980913ad94e0d09d0fffbe1bf5fe1eb4e64e15cf9f84fb87302cf244534337ec29b9e920a468ee03c1843e2f9a96  -
```

And I found another way of dumping the `DER` certificates without Python dependencies:
```shell
#!/bin/bash

while getopts "w" opt; do
    case "$opt" in
        w)
            write=1
            ;;
        ?)
            exit 1
            ;;
    esac
done
shift $((OPTIND -1))

for certificate in "${@}"; do
    # Find offset of all elements at depth 1
    openssl asn1parse -inform der -in "$certificate" | sed -En 's/^\s+([0-9]+):d=1\s.*/\1/p' | nl | while read index offset; do
        echo "Set element $index from $certificate:"
        command=(openssl asn1parse -inform der -in "$certificate" -strparse "$offset" -noout -out)
        "${command[@]}" - | openssl x509 -inform der -noout -text
        if [[ "$write" ]]; then
            "${command[@]}" "${certificate%.der}.$index.der"
        fi
    done
done
```

edmundoferreira, 2022-03-18:
I can also confirm this issue.

This is a picture of the failed ipxe boot with error https://ipxe.org/0216e43c (I rebuilt my ipxe payloads with DEBUG=x509)

[image]

For reference, my certificate chain is:
```
Certificate chain
 0 s:CN = hrilab.dev
   i:C = US, O = Let's Encrypt, CN = R3
 1 s:C = US, O = Let's Encrypt, CN = R3
   i:C = US, O = Internet Security Research Group, CN = ISRG Root X1
 2 s:C = US, O = Internet Security Research Group, CN = ISRG Root X1
   i:O = Digital Signature Trust Co., CN = DST Root CA X3
```
Which matches @sbraz.

My workaround for the moment is quite extreme. I removed the expiration check in `src/crypto/x509.c`. Use at your own risk.
```
diff --git a/src/crypto/x509.c b/src/crypto/x509.c
index 1f017eb0..1a9fa933 100644
--- a/src/crypto/x509.c
+++ b/src/crypto/x509.c
@@ -1301,7 +1301,7 @@ int x509_check_time ( struct x509_certificate *cert, time_t time ) {
        if ( validity->not_after.time < ( time - TIMESTAMP_ERROR_MARGIN ) ) {
                DBGC ( cert, "X509 %p \"%s\" has expired (at time %lld)\n",
                       cert, x509_name ( cert ), time );
-               return -EACCES_EXPIRED;
+               // return -EACCES_EXPIRED;
        }
 
        DBGC2 ( cert, "X509 %p \"%s\" is valid (at time %lld)\n",
```

sbrudenell, 2022-04-27:
Fastly uses the `DST Root CA X3` cross-signed certificate as of writing. So no one can load any of the published Alpine Linux netboot images from https://dl-cdn.alpinelinux.org, or anything else from fastly, today. So this seems like a big issue

For me, it was easier to rebuild ipxe than to stand up a custom host with the "right" front-end certificate. I used the `TRUST=` solution from https://github.com/ipxe/ipxe/issues/606#issuecomment-1057367152

I was surprised to learn the fingerprint is **different** between the self-signed cert (https://letsencrypt.org/certs/isrgrootx1.pem) and the cross-signed cert (https://letsencrypt.org/certs/lets-encrypt-r3.pem). I would've though the fingerprint would depend only on modulus, and thus be the same for both certificates, but that's not the case. `TRUST=` depends on the fingerprint, so both of these certs need to be included in `TRUST=`

[LetsEncrypt's announcement](https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/) about this says that older Android devices will continue to reference the `DST Root CA X3` cert despite its expiration. I speculate that fastly, and other large hosts, won't want to move away from their `DST Root CA X3` cross-signed certs yet.

nshalman, 2023-04-21:
> Fastly uses the `DST Root CA X3` cross-signed certificate as of writing. So no one can load any of the published Alpine Linux netboot images from https://dl-cdn.alpinelinux.org, or anything else from fastly, today. So this seems like a big issue
> 
> For me, it was easier to rebuild ipxe than to stand up a custom host with the "right" front-end certificate. I used the `TRUST=` solution from [#606 (comment)](https://github.com/ipxe/ipxe/issues/606#issuecomment-1057367152)

This is definitely still an issue for LetsEncrypt certificates. The linked workaround still works, but it would be good to get this properly fixed.

```
iPXE> imgfetch https://www.shalman.org/files/workaround.ipxe
TLS 0x36944 created session www.shalman.org
https://www.shalman.org/files/workaround.ipxe...TLS 0x36944 using protocol versi
on 3.3
TLS 0x36944 selected dhe-rsa-aes_gcm-128-sha256
TLS 0x36944 new session ID:
00000000 : d2 48 c4 da 57 0f 82 70-a8 ac f6 6d fa e6 21 d5 : .H..W..p...m..!.
00000010 : ec 29 57 ba 03 41 a2 14-bc b7 cd f4 c8 7f 95 b4 : .)W..A..........
CERTSTORE added certificate shalman.org
X509 chain 0x364a4 added X509 0x39604 "shalman.org"
TLS 0x36944 found certificate shalman.org
CERTSTORE added certificate R3
X509 chain 0x364a4 added X509 0x39c34 "R3"
TLS 0x36944 found certificate R3
CERTSTORE added certificate ISRG Root X1
X509 chain 0x364a4 added X509 0x3a214 "ISRG Root X1"
TLS 0x36944 found certificate ISRG Root X1
X509 chain 0x364a4 found no usable certificates
CERTSTORE added certificate DST Root CA X3
X509 chain 0x368e4 added X509 0x37fe4 "DST Root CA X3"
CERTSTORE added certificate iPXE cross-signing CA
X509 chain 0x368e4 added X509 0x383d4 "iPXE cross-signing CA"
CERTSTORE added certificate iPXE root CA
X509 chain 0x368e4 added X509 0x387f4 "iPXE root CA"
X509 chain 0x364a4 added X509 0x37fe4 "DST Root CA X3"
X509 chain 0x364a4 added X509 0x383d4 "iPXE cross-signing CA"
X509 chain 0x364a4 added X509 0x387f4 "iPXE root CA"
X509 0x37fe4 "DST Root CA X3" has expired (at time 1682103495)
X509 chain 0x364a4 found no usable certificates
TLS 0x36944 certificate validation failed: Permission denied (https://ipxe.org/0
216eb3c)
 Permission denied (https://ipxe.org/0216eb3c)
```

<details>
 <summary>For completeness, output from a build with the workaround applied</summary>

```
PXE->EB: !PXE at 9C6C:0710, entry point at 9C6C:0160
         UNDI code segment 9C6C:0802, data segment 9CF0:2CE0 (625-639kB)
         UNDI device is PCI 00:03.0, type DIX+802.3
         625kB free base memory after PXE unload
CERTSTORE added certificate iPXE root CA
CERTSTORE permanent certificate 0 is iPXE root CA
CERTSTORE added certificate ISRG Root X1
CERTSTORE permanent certificate 1 is ISRG Root X1
CERTSTORE added certificate R3
CERTSTORE permanent certificate 2 is R3
iPXE initialising devices...ok

iPXE 1.21.1+ (g1d1cf7) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP HTTPS NFS TFTP VLAN ELF MBOOT PXE bzImage Menu PXEXT

iPXE> dhcp
Configuring (net0 52:54:00:12:34:56)...... ok
iPXE> imgfetch https://www.shalman.org/files/workaround.ipxe
TLS 0x379c4 created session www.shalman.org
https://www.shalman.org/files/workaround.ipxe...TLS 0x379c4 using protocol versi
on 3.3
TLS 0x379c4 selected dhe-rsa-aes_gcm-128-sha256
TLS 0x379c4 new session ID:
00000000 : d0 a3 70 bc ee 9a 2e 3b-36 ac 34 b2 15 80 68 25 : ..p....;6.4...h%
00000010 : 58 e2 d4 84 e0 79 04 59-1d 86 67 db 54 94 3b f0 : X....y.Y..g.T.;.
CERTSTORE added certificate shalman.org
X509 chain 0x37524 added X509 0x3a604 "shalman.org"
TLS 0x379c4 found certificate shalman.org
X509 chain 0x37524 added X509 0xb7110 "R3"
TLS 0x379c4 found certificate R3
CERTSTORE added certificate ISRG Root X1
X509 chain 0x37524 added X509 0x3ac34 "ISRG Root X1"
TLS 0x379c4 found certificate ISRG Root X1
X509 0xb7110 "R3" is a root certificate
X509 0x3a604 "shalman.org" successfully validated using issuer 0xb7110 "R3"
TLS 0x379c4 certificate validation succeeded
```

</details>

NiKiZe, 2023-04-21:
> This is definitely still an issue for LetsEncrypt certificates. The linked workaround still works, but it would be good to get this properly fixed.
> 

Just to have all data..., which version of iPXE are you using here?

nshalman, 2023-04-21:
> Just to have all data..., which version of iPXE are you using here?

Manually built from 1d1cf74a5e58811822bee4b3da3cff7282fcdfca

NiKiZe, 2023-04-21:
> > Just to have all data..., which version of iPXE are you using here?
> 
> Manually built from [1d1cf74](https://github.com/ipxe/ipxe/commit/1d1cf74a5e58811822bee4b3da3cff7282fcdfca)

That is current git master, and matching your output from working test? If so it is already properly fixed?

macpijan, 2023-12-21:
Confirmed to be a problem still on the latest iPXE revision 3fc1b407d204e950a41c567e84f4c4f079cc052e

[image]

miczyg1, 2023-12-21:
Yes, but at least the chain is properly retrieved...

macpijan, 2023-12-21:
> Yes, but at least the chain is properly retrieved...

This issue is not about losing something from the chain.

Paper-Dragon, 2023-12-27:
Have you solved it?

mcb30, 2024-02-15:
For reference (quoting from commit https://github.com/ipxe/ipxe/commit/929f06a76):

Certificates issued by Let's Encrypt have two options for their chain of trust: the chain can either terminate in the self-signed ISRG Root X1 root certificate, or in an intermediate ISRG Root X1 certificate that is signed in turn by the self-signed DST Root CA X3 root certificate. This is a historical artifact: when Let's Encrypt first launched as a project, the chain ending in DST Root CA X3 was used since existing clients would not have recognised the ISRG Root X1 certificate as a trusted root certificate.

The DST Root CA X3 certificate expired in September 2021, and so is no longer trusted by clients (such as iPXE) that validate the expiry times of all certificates in the certificate chain.

In order to maintain usability of certificates on older Android devices, the default certificate chain provided by Let's Encrypt still terminates in DST Root CA X3, even though that certificate has now expired. On newer devices which include ISRG Root X1 as a trusted root certificate, the intermediate version of ISRG Root X1 in the certificate chain is ignored and validation is performed as though the chain had terminated in the self-signed ISRG Root X1 root certificate. On older Android devices which do not include ISRG Root X1 as a trusted root certificate, the validation succeeds since Android chooses to ignore expiry times for root certificates and so continues to trust the DST Root CA X3 root certificate.

This backwards compatibility hack unfortunately breaks the cross- signing mechanism used by iPXE, which assumes that the certificate chain will always terminate in a non-expired root certificate.

Now fixed via #1152

closed 2024-02-15 by mcb30