ipxe/ipxe #607: PC freeze after 1 hour

arinci, 2022-03-02:
Hello, I try to describe a strange problem that I saw on two server PCs based on Advantech motherboard, model ASMB-813 (this one: https://www.advantech.com/products/serverboard/asmb-813/mod_3310bc5e-4c0f-4422-a13a-6633e0241e3b) when booted with UEFI iPXE served from a TFTP server: after having network booted I execute Passmark Memtest86 v9.4, to verify that memory is OK.
The Memtest86 application starts fine, but after 1 hour it freeze. No way to use CTRL+ALT+DEL or any other keys combination, the only way to exit from this hang is to reset (or power cycle) the PC.
This behaviour has been seen two times on two server PC, so a total of 4 times.
Please note that this strange issue has been observed with latest revision [2acdc92](https://github.com/ipxe/ipxe/commit/2acdc92994e7aca397b0d24b112e4973e82e0f91) of iPXE source code.

After that, where I thought that the problem could be related to a bad power-save setting in the BIOS, I reverted to the previous version of ipxe.efi on my TFTP server before start playing with BIOS parameters: tested again as above, and the problem is disappeared. I don't know of the revision of this working ipxe.efi file, I just know that it was created using the rom-o-matic website service and that the file date is July 1st 2020. Tested many times, and with this old iPXE revision everything is OK!
At the end I can survive with this old release, it's working very well, I just like to have latest revision of the product, this time built using the docker container available here: https://github.com/lps-rocks/rom-o-matic

Any ideas?

NiKiZe, 2022-03-02:
Please try to figure out exactly which commit is causing the issues, see https://ipxe.org/howto/bisect if iPXE at all.
More probably bad hardware or firmware issues. However consider checking for any watchdog settings.

NiKiZe, 2022-03-04:
Any more info here?

arinci, 2022-03-06:
I'm using a docker image to build old version of iPXE binaries, but now it stopped working...I don't know if I will be able to restore this build machine.
This is the relevant project on Git: https://github.com/lps-rocks/rom-o-matic

On next days I will try the same iPXE revision I'm having troubles on a different PC, this to understand if the problem is related to a specific motherboard or if it is more general.

arinci, 2022-03-09:
Little update: testing the same version I'm having trouble on a different PC, the problem does not occur.
It means that the culprit is probably the UEFI firmware on the server motherboard.
I will try to build and test some old iPXE releases, to understand from which commit the problem arose...this process need time, as I have to wait 1 hour to see if the problem is still present or no.

arinci, 2022-03-12:
Just completed the test campaign aimed to identify the commit that is causing the timed freeze on Advantech server motherboard.
Latest working release is this one: dced22d6dee346ac4d98502c9008fd3d5c6197d2, dated Dec 29, 2020.
From following releases, starting from this 988d2c13cdf0f0b4140685af35ced70ac5b3283c dated Dec 31, 2020, I have the problem on my config. The commit description is relevant on EFI implementation:
`**[efi] Use segment and bus number to identify PCI root bridge I/O protocol**`.

Just to let you know, probably a bug in UEFI firmware cause this issue, nothing worthwhile further investigations. 

Final note: isn't a bit strange that the problem arise after 1 hour of a memtest session?

NiKiZe, 2022-07-04:
Did you try finding a firmware upgrade, and have you reported this to manufacturer?

NiKiZe, 2023-07-09:
Closing due to lack of feedback.

This feels timer related, unless there is a way that we know anyone can reproduce this, the actual hardware is probably needed.
Might be interesting to try and see if this can be reproduced in EFI shell.

One other possibility is that memtests overwrites some memory after 1h of running, and that memory is still used by something, which then crashes.

closed 2023-10-08 by zube[bot]