ipxe/ipxe #608: "/ |" key not working (EFI) keymaping not working (BIOS)

jensgw, 2022-03-07:
Hi there, some issues with some special keys:

ipxe.efi
When using ipxe.efi in a QEMU/libvirt VM the "/ |" key is not working (the one left to z). Switching keymaps de -> us -> de works but the key does not come alive. In neither keymap.

undionly.kpxe
When using undionly.kpxe in VM-Ware VM the "/ |" is alive with a german layout as < > but not with alt gr, so pipe is impossible to type. Switching keymaps has no effect alt all. Y and Z are handeled according to us -> de switch. Any other key is not handeled correctly, e.g. 
";" should be "ö"
"[" should be "ü"
You would assume that then "]" would stay, but no, this is correctly mapped to "+".

Somewhere is some weirdnes going on ...

ipxe was built with this commit id:
2acdc92994e7aca397b0d24b112e4973e82e0f91

NiKiZe, 2022-03-07:
In bios there is no way to detect AltGr so you need to press both shift keys (if I remember correctly) several chars does not exist or isn't standard and depends on the actuall font which is unknown, as such don't expect umlauts to work.

You might want to read thru the keymap commits before the commit you just mentioned.

jensgw, 2022-03-07:
Thx will read.
Some observations:

EFI
AltGr is working for @ (AltGr q) and [ ] (AlrGr 8 9) but not | (AltGr <)
set keymap has no effect

BIOS
shift-R + shift-L < produces the correct |, so thanks!
set keymap has no effect

Umlauts are not important, just a thing i noticed.

mcb30, 2022-03-08:
> ipxe.efi When using ipxe.efi in a QEMU/libvirt VM the "/ |" key is not working (the one left to z). Switching keymaps de -> us -> de works but the key does not come alive. In neither keymap.

This is down to a bug in the EDK2 keyboard driver.  It simply does not generate any keypress data for that physical key: there is no way for iPXE to know that the key has been pressed.

You can work around this problem by building iPXE with a native USB keyboard drivers (e.g. `bin-x86_64-efi/ipxe--usbkbd.efi` should work, or any iPXE build that includes a USB NIC driver such as `bin-x86_64-efi/ipxe--ecm.efi`).  With the native USB keyboard driver, iPXE is able to see the original scancode and that key should work.

> When using undionly.kpxe in VM-Ware VM the "/ |" is alive with a german layout as < > but not with alt gr, so pipe is impossible to type. Switching keymaps has no effect alt all. Y and Z are [...]

Non-ASCII characters are not yet supported, but this is work in progress.  For now, any keypress that should result in a non-ASCII character is likely to produce the key that would exist at that position on a US keyboard layout (i.e. iPXE simply does not remap the key).

jensgw, 2022-03-09:
Thx for the hints. I tried ipxe--usbkbd.efi and ipxe--ecm.efi. Neither did not work with QEMU nor VM-WARE.

NiKiZe, 2022-07-04:
> Thx for the hints. I tried ipxe--usbkbd.efi and ipxe--ecm.efi. Neither did not work with QEMU nor VM-WARE.

You need to have the keyboard over USB, so use separate keyboard with USB passthru?