ipxe/ipxe #609: Lenovo T14 iPXE boot failed

smallfish01, 2022-03-09:
Hello,
I have a new  laptop and like to install Windows 10 from iPXE server, when I boot laptop from iPXE it can be get IP address from DHCP server, but the next step get error: "net1 link down" and cannot install OS from iPXE server, do you know why?

My laptop NIC is Realtek RTL8111H-CG.

Please check the screenshot.
[image]

Thank you so much.

NiKiZe, 2022-03-09:
Exactly which model? Having several T14 (Intel) that works fine.
You seem to have blured out some details, which makes me believe that you are not using official iPXE, and also are not using latest version.

Please try latest ipxe.efi from boot.ipxe.org

smallfish01, 2022-03-10:
Hello,
ThinkPad T14 AMD Gen 2,  CPU：AMD Ryzen 5 PRO 5650U 2.3Ghz. 
I pasted two new screenshots as below, please have a look, also I will update the latest version later.
[image]
[image]

NiKiZe, 2022-03-10:
Will not even try to debug this, run latest pure iPXE and reproduce.

smallfish01, 2022-03-10:
Hello, 
Just download the latest iPXE and verify, it works.

So the reason is iPXE does not have laptop NIC driver?

Thank you so much.

NiKiZe, 2022-03-10:
You downloaded the latest iPXE which works?, then the issue have eighter been fixed in a later version. Or your other builds are broken.

So not a current issue with iPXE, correct?

smallfish01, 2022-03-11:
Sure. I downloaded the latest iPXE(V1.21.1) then my new laptop can boot from iPXE and install Windows OS,  so the latest iPXE works for T14 laptop.

Thank you so much.

NiKiZe, 2022-03-11:
Awesome!

closed 2022-03-11 by NiKiZe