ipxe/ipxe #613: Intel Ethernet Controller I225-V

x-stride, 2022-03-13:
Hello all,

Can we expect iPXE to work on these devices?

The below might be a bad idea... Or not??? :)
```
git diff 2acdc92994e7aca397b0d24b112e4973e82e0f91
diff --git a/src/drivers/net/intel.c b/src/drivers/net/intel.c
index ea3ebf68..4e705d39 100644
--- a/src/drivers/net/intel.c
+++ b/src/drivers/net/intel.c
@@ -1167,6 +1167,7 @@ static struct pci_device_id intel_nics[] = {
        PCI_ROM ( 0x8086, 0x15e1, "i219lm-9", "I219-LM (9)", INTEL_I219 ),
        PCI_ROM ( 0x8086, 0x15e2, "i219v-9", "I219-V (9)", INTEL_I219 ),
        PCI_ROM ( 0x8086, 0x15e3, "i219lm-5", "I219-LM (5)", INTEL_I219 ),
+       PCI_ROM ( 0x8086, 0x15f3, "i225v-3", "I225-V (3)", INTEL_I219 ),
        PCI_ROM ( 0x8086, 0x15f4, "i219lm-15", "I219-LM (15)", INTEL_I219 ),
        PCI_ROM ( 0x8086, 0x15f5, "i219v-15", "I219-V (15)", INTEL_I219 ),
        PCI_ROM ( 0x8086, 0x15f9, "i219lm-14", "I219-LM (14)", INTEL_I219 ),
```

/Stride

NiKiZe, 2022-03-13:
From what I can see that is a completely different type of nic, based on the Linux driver being igc? You also have intelx.

You should always expect iPXE to work on top of undi or SNP drivers if chained withou native drivers.

x-stride, 2022-03-14:
> From what I can see that is a completely different type of nic, based on the Linux driver being igc? You also have intelx.
> 
> You should always expect iPXE to work on top of undi or SNP drivers if chained withou native drivers.

Thanks, I'll give it a go :)
The goal is a 100% stateless deployment, DHCP only - no PXE/TFTP etc. No NIC iPXE ROM excludes that as an option, correct?

NiKiZe, 2022-03-14:
in efi mode SNP/NII interface should be available.
But it does require that the original NIC ROM is available, so if you replace it then it won't work. (But you should test from USB first)
Also BIOS mode will probably not work without PXE chaining.

stappersg, 2022-05-02:
For which reason is this issue open?
(This update is a link to #641)

MischaBaars, 2022-11-02:
Ok, so I have a similar (brand new) network interface that does not boot either using iPXE (version 1.0.0+? - PXE 2.1 build 083)

The network interface:

06:00.0 Ethernet controller: Intel Corporation Ethernet Controller I225-V (rev 01)
	Subsystem: Gigabyte Technology Co., Ltd Device e000
	Flags: bus master, fast devsel, latency 0, IRQ 38, IOMMU group 24
	Memory at fc400000 (32-bit, non-prefetchable) [size=1M]
	Memory at fc500000 (32-bit, non-prefetchable) [size=16K]
	Capabilities: <access denied>
	Kernel driver in use: igc
	Kernel modules: igc

The error:

https://ipxe.org/040ee1 Error code 040ee1

The ipxe.iso image is the same for the two nodes involved. The second node boots just fine, kernel 6.0.6.

Help would be appreciated.

MischaBaars, 2022-11-02:
06:00.0 0200: 8086:15f3 (rev 01)
	Subsystem: 1458:e000

NiKiZe, 2022-11-02:
As reference of where to look for adding PCIID the igc linux driver: https://github.com/torvalds/linux/blob/master/drivers/net/ethernet/intel/igc/igc_hw.h#L19-L34
iPXE does not seem to have any of these (but that is just a quick look) so someone need to write (read fund) such a driver.

In the mean time, the mention above of SNP/NII in EFI still stands.

MischaBaars, 2022-11-02:
> As reference of where to look for adding PCIID the igc linux driver:
> https://github.com/torvalds/linux/blob/master/drivers/net/ethernet/intel/igc/igc_hw.h#L19-L34
> iPXE does not seem to have any of these (but that is just a quick look) so
> someone need to write (read fund) such a driver.

I've already asked Intel. We're waiting for a response.

In the mean time, the mention above of SNP/NII in EFI still stands.

I do not understand what it says. Is there another way to get the DHCP
working?

stappersg, 2022-11-02:
On Wed, Nov 02, 2022 at 08:14:29AM -0700, Mischa Baars wrote:
> On Wed, 2 Nov 2022, 15:50 Christian I. Nilsson wrote:
> > In the mean time, the mention above of SNP/NII in EFI still stands.
> 
> I do not understand what it says.
> Is there another way to get the DHCP working?

Yes. It is what NiKiZe advice is about.

MischaBaars, 2022-11-03:
Ok, the email address that Intel provides in the Linux kernel drivers does not exist. I have contacted support, but this is going to take a while I suppose.

So, what this about SNP/NII? It doesn't really ring a bell. If I boot using the network interface, the first thing that probably happens is that iPXE detects the network interface and negotiates an IP address. Are you proposing to intercept the detection of the network interface and use alternative drivers?

NiKiZe, 2022-11-03:
> Ok, the email address that Intel provides in the Linux kernel drivers does not exist. I have contacted support, but this is going to take a while I suppose.
> 

Intel is unlikely to support iPXE directly, however for their new devices they seem to use paid services from @mcb30 

> So, what this about SNP/NII? It doesn't really ring a bell. If I boot using the network interface, the first thing that probably happens is that iPXE detects the network interface and negotiates an [...]

Are you using efi? The firmware driver for the NIC is usable under iPXE by default.

danieltwagner, 2023-04-24:
I have one of these boxes: https://www.servethehome.com/fanless-4x-2-5gbe-intel-n5105-i225-v-firewall-tested
and the supplied UEFI BIOS doesn't seem to have a network stack that can be enabled. I was hoping iPXE might be able to help me out but then I discovered this issue. Am I reading it right that it's unlikely we'll see i225 support soon?

Given that the BIOS doesn't seem to know about the NICs, I imagine this isn't relevant in my case?
> The firmware driver for the NIC is usable under iPXE by default.

MischaBaars, 2023-04-24:
Hi Daniel,

Why do you want to boot this machine over the network? Unless you are
planning to boot a lot more of them, installing and configuring Linux from
USB image is the easier way to get things working.

I can't really remember where iPXE came in, must have been functioning as a
bridge between the BIOS and Syslinux. I now use Grub instead. Although
booting is a lot slower than with Syslinux for some reason, at least Grub
can be compiled. None of the Syslinux versions available can.

If there's no network boot option in your BIOS, your machine probably can't
be used as a network node. I've used four Beelink (AZW) MII-V dual core
nodes to get my cluster up and running. They are somewhat outdated, but the
newer Beelink nodes will most probably boot over the network just as easy.

Something like this should get you on your way. You can buy more expensive
nodes when you're done playing. Remember to enable BIOS write protect
before plugging them into the network. I lost one of four five minutes
after plugging it in.

https://m.alibaba.com/product/1600823329030/Intel-Celeron-N5095-Mini-PC-Quad.html?__sceneInfo=%7B%22cacheTime%22%3A%221800000%22%2C%22type%22%3A%22appDetailShare%22%7D&from=Android

Mischa.

qdrop17, 2023-09-18:
Yeah we have the same issue: We have an ASUS PN61 mini PC with a i226-v interface. We fail to get our networkboot working: It is stuck at "iPXE initialising devices". We're already targeting snponly.

As far as I'm aware, there's no workaround available currently, right? How can we support to get these NICs working? They are only gaining in popularity onwards.

MischaBaars, 2023-09-19:
https://linuxguideandhints.com/el/pxeboot.html explains how to use grub to
make your nodes boot from the network. Grub does not have an issue with
these network adapters.

qdrop17, 2023-09-28:
> As reference of where to look for adding PCIID the igc linux driver: https://github.com/torvalds/linux/blob/master/drivers/net/ethernet/intel/igc/igc_hw.h#L19-L34 iPXE does not seem to have any of [...]

Hi there. We're building our iPXE bootloader with snponly but we still fail to bring our NIC online. Am I assuming it correctly, that the NIC ROM is not available on this machine? Would this ever work without the drivers for these NICs built into iPXE?

Any help is appreciated.

NiKiZe, 2023-09-28:
If you can chain into snponly over the network then EFI has a "working" driver for the NIC, however iPXE might expose bugs in that driver and that might need a fix, in this case by Intel, and then update of machine firmware.

NiKiZe, 2026-04-07:
I assume things has changed since #1565 d73981aece2544ee05685dccaf5a71b533fabb44
If anyone could report back that would be great, if still not working we will reopen, but closing as assumed fixed.

closed 2026-04-07 by NiKiZe