ipxe/ipxe #620: unable to compile on fedora 36

damnms, 2022-03-20:
i checked out the source and tried to compile the usb boot file with make bin/ipxe.usb, but unfortunately it failed 

>   [BUILD] bin/gdbstub.o
>   [BUILD] bin/gdbudp.o
> In Datei, eingebunden von include/string.h:23,
>                  von core/gdbudp.c:27:
> In Funktion »__constant_memcpy«,
>     eingefügt von »memcpy« bei arch/x86/include/bits/string.h:168:10,
>     eingefügt von »gdbudp_send« bei core/gdbudp.c:206:2,
>     eingefügt von »gdbudp_send« bei core/gdbudp.c:163:13:
> arch/x86/include/bits/string.h:94:44: Fehler: Arrayindex »const union <anonym>[0]« ist teilweise außerhalb der Arraygrenzen von »uint8_t[6]« {alias »unsigned char[6]«} [-Werror=array-bounds]
>    94 |                 dest_u->u32[0] = src_u->u32[0];
>       |                                  ~~~~~~~~~~^~~
> core/gdbudp.c: In Funktion »gdbudp_send«:
> core/gdbudp.c:53:16: Anmerkung: Objekt »dest_eth« der Größe 6
>    53 | static uint8_t dest_eth[ETH_ALEN];
>       |                ^~~~~~~~
> In Funktion »__constant_memcpy«,
>     eingefügt von »memcpy« bei arch/x86/include/bits/string.h:168:10,
>     eingefügt von »gdbudp_send« bei core/gdbudp.c:206:2,
>     eingefügt von »gdbudp_send« bei core/gdbudp.c:163:13:
> arch/x86/include/bits/string.h:95:44: Fehler: Arrayindex »const union <anonym>[0]« ist teilweise außerhalb der Arraygrenzen von »uint8_t[6]« {alias »unsigned char[6]«} [-Werror=array-bounds]
>    95 |                 dest_u->u16[2] = src_u->u16[2];
>       |                                  ~~~~~~~~~~^~~
> core/gdbudp.c: In Funktion »gdbudp_send«:
> core/gdbudp.c:53:16: Anmerkung: Objekt »dest_eth« der Größe 6
>    53 | static uint8_t dest_eth[ETH_ALEN];
>       |                ^~~~~~~~
> In Funktion »__constant_memcpy«,
>     eingefügt von »memcpy« bei arch/x86/include/bits/string.h:168:10,
>     eingefügt von »gdbudp_recv« bei core/gdbudp.c:137:4:
> arch/x86/include/bits/string.h:94:32: Fehler: Arrayindex »union <anonym>[0]« ist teilweise außerhalb der Arraygrenzen von »uint8_t[6]« {alias »unsigned char[6]«} [-Werror=array-bounds]
>    94 |                 dest_u->u32[0] = src_u->u32[0];
>       |                 ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
> core/gdbudp.c: In Funktion »gdbudp_recv«:
> core/gdbudp.c:53:16: Anmerkung: Objekt »dest_eth« der Größe 6
>    53 | static uint8_t dest_eth[ETH_ALEN];
>       |                ^~~~~~~~
> In Funktion »__constant_memcpy«,
>     eingefügt von »memcpy« bei arch/x86/include/bits/string.h:168:10,
>     eingefügt von »gdbudp_recv« bei core/gdbudp.c:137:4:
> arch/x86/include/bits/string.h:95:32: Fehler: Arrayindex »union <anonym>[0]« ist teilweise außerhalb der Arraygrenzen von »uint8_t[6]« {alias »unsigned char[6]«} [-Werror=array-bounds]
>    95 |                 dest_u->u16[2] = src_u->u16[2];
>       |                 ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
> core/gdbudp.c: In Funktion »gdbudp_recv«:
> core/gdbudp.c:53:16: Anmerkung: Objekt »dest_eth« der Größe 6
>    53 | static uint8_t dest_eth[ETH_ALEN];
>       |                ^~~~~~~~
> cc1: Alle Warnungen werden als Fehler behandelt
> make: *** [Makefile.housekeeping:952: bin/gdbudp.o] Fehler 1

stappersg, 2022-03-20:
And which version has your compiler?
Which compiler are you using anyway?

What is the source that is checked out?  (Yeah, probably "git".
State git branch and latest git commit hash.)

Oh, by the way, prefixing `make bin/ipxe.usb` with `LANG=C`
will get us more international messages. So please do
```
LANG=C make bin/ipxe.usb
```
when your audience might not speak German.

damnms, 2022-03-20:
gcc (GCC) 12.0.1 20220308 (Red Hat 12.0.1-0)
source is master, latest commit is dd3547543811dd2e996b910bdadb609414904352 

```
  [PARSEROM]
  [BUILD] bin/gdbudp.o
In file included from include/string.h:23,
                 from core/gdbudp.c:27:
In function '__constant_memcpy',
    inlined from 'memcpy' at arch/x86/include/bits/string.h:168:10,
    inlined from 'gdbudp_send' at core/gdbudp.c:206:2,
    inlined from 'gdbudp_send' at core/gdbudp.c:163:13:
arch/x86/include/bits/string.h:94:44: error: array subscript 'const union <anonymous>[0]' is partly outside array bounds of 'uint8_t[6]' {aka 'unsigned char[6]'} [-Werror=array-bounds]
   94 |                 dest_u->u32[0] = src_u->u32[0];
      |                                  ~~~~~~~~~~^~~
core/gdbudp.c: In function 'gdbudp_send':
core/gdbudp.c:53:16: note: object 'dest_eth' of size 6
   53 | static uint8_t dest_eth[ETH_ALEN];
      |                ^~~~~~~~
In function '__constant_memcpy',
    inlined from 'memcpy' at arch/x86/include/bits/string.h:168:10,
    inlined from 'gdbudp_send' at core/gdbudp.c:206:2,
    inlined from 'gdbudp_send' at core/gdbudp.c:163:13:
arch/x86/include/bits/string.h:95:44: error: array subscript 'const union <anonymous>[0]' is partly outside array bounds of 'uint8_t[6]' {aka 'unsigned char[6]'} [-Werror=array-bounds]
   95 |                 dest_u->u16[2] = src_u->u16[2];
      |                                  ~~~~~~~~~~^~~
core/gdbudp.c: In function 'gdbudp_send':
core/gdbudp.c:53:16: note: object 'dest_eth' of size 6
   53 | static uint8_t dest_eth[ETH_ALEN];
      |                ^~~~~~~~
In function '__constant_memcpy',
    inlined from 'memcpy' at arch/x86/include/bits/string.h:168:10,
    inlined from 'gdbudp_recv' at core/gdbudp.c:137:4:
arch/x86/include/bits/string.h:94:32: error: array subscript 'union <anonymous>[0]' is partly outside array bounds of 'uint8_t[6]' {aka 'unsigned char[6]'} [-Werror=array-bounds]
   94 |                 dest_u->u32[0] = src_u->u32[0];
      |                 ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
core/gdbudp.c: In function 'gdbudp_recv':
core/gdbudp.c:53:16: note: object 'dest_eth' of size 6
   53 | static uint8_t dest_eth[ETH_ALEN];
      |                ^~~~~~~~
In function '__constant_memcpy',
    inlined from 'memcpy' at arch/x86/include/bits/string.h:168:10,
    inlined from 'gdbudp_recv' at core/gdbudp.c:137:4:
arch/x86/include/bits/string.h:95:32: error: array subscript 'union <anonymous>[0]' is partly outside array bounds of 'uint8_t[6]' {aka 'unsigned char[6]'} [-Werror=array-bounds]
   95 |                 dest_u->u16[2] = src_u->u16[2];
      |                 ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
core/gdbudp.c: In function 'gdbudp_recv':
core/gdbudp.c:53:16: note: object 'dest_eth' of size 6
   53 | static uint8_t dest_eth[ETH_ALEN];
      |                ^~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:952: bin/gdbudp.o] Error 1

```

andmn, 2022-04-02:
as libvirt cant boot with X540 ixgbevf I tried to made some efi rom and end up with same issue, make example

make bin-x86_64-efi/80861515.efirom
  [DEPS] core/fnrec.c
  [BUILD] bin-x86_64-efi/fnrec.o
In function ‘fnrec_invalidate’,
    inlined from ‘fnrec_init’ at core/fnrec.c:171:3:
core/fnrec.c:95:29: error: array subscript 0 is outside array bounds of ‘void[0]’ [-Werror=array-bounds]
   95 |         fnrec_buffer->magic = 0;
      |         ~~~~~~~~~~~~~~~~~~~~^~~
cc1: all warnings being treated as errors

stappersg, 2022-04-04:
On Sat, Apr 02, 2022 at 08:45:12AM -0700, andmn wrote:
> ... and end up with same issue ...

And that is with which compiler  version?

stappersg, 2022-04-05:
I can reproduce the error with GCC version 12 on a _Debian system_.

The _"failing"_ version:
```text
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/12/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 12-20220319-1'
   --with-bugurl=file:///usr/share/doc/gcc-12/README.Bugs
   --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr
   --with-gcc-major-version-only --program-suffix=-12 --program-prefix=x86_64-linux-gnu-
   --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext
   --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu
   --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new
   --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie
   --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto
   --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686
   --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic
   --enable-offload-targets=nvptx-none=/build/gcc-12-Sgcwxm/gcc-12-12-20220319/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-12-Sgcwxm/gcc-12-12-20220319/debian/tmp-gcn/usr
   --enable-offload-defaulted --without-cuda-driver --enable-checking=release
   --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.0.1 20220319 (experimental) [master r12-7719-g8ca61ad148f] (Debian 12-20220319-1) 
```
With  `gcc` _symlinked_ to `/usr/bin/gcc-11` I can compile iPXE.

@damnms  please retitle this issue to _"unable to compile with gcc 12"_

ctxnop, 2022-05-20:
The issue seems to come from gcc 12 as discussed here: https://bugzilla.suse.com/show_bug.cgi?id=1197018#c1

stappersg, 2022-05-20:
> The issue seems to come from gcc 12 as discussed here: https://bugzilla.suse.com/show_bug.cgi?id=1197018#c1

Quoting https://bugzilla.suse.com/show_bug.cgi?id=1197018#c13
> Fixed in devel project. Thanks for working on that.

@damnms please retitle this issue to _"unable to compile with gcc 12"_

stappersg, 2022-05-28:
[comment 2 from that bugzilla issue:](https://bugzilla.suse.com/show_bug.cgi?id=1197018#c2)

So the patch is:
```text
diff --git a/roms/ipxe/src/Makefile.housekeeping b/roms/ipxe/src/Makefile.housekeeping
index 134becd50..4b0aa9e96 100644
--- a/roms/ipxe/src/Makefile.housekeeping
+++ b/roms/ipxe/src/Makefile.housekeeping
@@ -183,6 +183,20 @@ WNAPM_TEST = $(CC) -Wno-address-of-packed-member -x c -c /dev/null \
 WNAPM_FLAGS := $(shell $(WNAPM_TEST) && \
 		 $(ECHO) '-Wno-address-of-packed-member')
 WORKAROUND_CFLAGS += $(WNAPM_FLAGS)
+
+# gcc 12.1 generates false positive warnings.  Inhibit the warnings.
+WNAB_TEST = $(CC) -Wno-array-bounds -x c -c /dev/null \
+		   -o /dev/null >/dev/null 2>&1
+WNAB_FLAGS := $(shell $(WNAB_TEST) && \
+		 $(ECHO) '-Wno-array-bounds')
+WORKAROUND_CFLAGS += $(WNAB_FLAGS)
+
+WNDP_TEST = $(CC) -Wno-dangling-pointer-x c -c /dev/null \
+		   -o /dev/null >/dev/null 2>&1
+WNDP_FLAGS := $(shell $(WNAB_TEST) && \
+		 $(ECHO) '-Wno-dangling-pointer')
+WORKAROUND_CFLAGS += $(WNDP_FLAGS)
+
 endif
 
 # Some versions of gas choke on division operators, treating them as
```
But I hit another problem:
```
[ 1051s] Sdk/C/LzmaEnc.c: In function 'LzmaEnc_CodeOneMemBlock':
[ 1051s] Sdk/C/LzmaEnc.c:2641:19: error: storing the address of local variable 'outStream' in '*p.rc.outStream' [-Werror=dangling-pointer=]
[ 1051s]  2641 |   p->rc.outStream = &outStream.vt;
[ 1051s]       |   ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
[ 1051s] Sdk/C/LzmaEnc.c:2624:28: note: 'outStream' declared here
[ 1051s]  2624 |   CLzmaEnc_SeqOutStreamBuf outStream;
[ 1051s]       |                            ^~~~~~~~~
[ 1051s] Sdk/C/LzmaEnc.c:2624:28: note: 'pp' declared here
```

Please disable the warning there as well.

NiKiZe, 2022-07-04:
Marking as bug, but to be clear, this is an issue with gcc12, you should be able to install a working gcc in a slotted fashion.

stappersg, 2022-07-20:
Solution is reported to the mailing list of _iPXE_, archived as https://lists.ipxe.org/pipermail/ipxe-devel/2022-July/thread.html

stappersg, 2022-07-20:
> Solution is reported to the mailing list of _iPXE_, archived as https://lists.ipxe.org/pipermail/ipxe-devel/2022-July/thread.html

Deeper link is https://lists.ipxe.org/pipermail/ipxe-devel/2022-July/007571.html

kallisti5, 2022-10-17:
Same issue seen here.  GCC 12.2.0

MischaBaars, 2022-11-02:
This deeper link work around qemu-ipxe fails to compile on my end just as the ipxe master branch did, only now the error is:

  [BUILD] bin/ansicoldef.o
In file included from include/string.h:23,
                 from include/ipxe/uaccess.h:25,
                 from core/acpi.c:28:
In function ‘__constant_memcpy’,
    inlined from ‘memcpy’ at arch/x86/include/bits/string.h:168:10,
    inlined from ‘trivial_memcpy_user’ at include/ipxe/uaccess.h:109:2,
    inlined from ‘memcpy_user’ at arch/x86/include/librm.h:155:2,
    inlined from ‘copy_from_user’ at include/ipxe/uaccess.h:338:2,
    inlined from ‘acpi_sx_zsdt’ at core/acpi.c:217:3:
arch/x86/include/bits/string.h:79:32: error: array subscript ‘union <anonymous>[0]’ is partly outside array bounds of ‘union <anonymous>[1]’ [-Werror=array-bounds]
   79 |                 dest_u->u32[0] = src_u->u32[0];
      |                 ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
core/acpi.c: In function ‘acpi_sx_zsdt’:
core/acpi.c:191:11: note: object ‘buf’ of size 4
  191 |         } buf;
      |           ^~~
  [BUILD] bin/ansiesc.o

MischaBaars, 2022-11-02:
> A workaround at this point is to run with `NO_WERROR`:
> 
> ```shell
> make NO_WERROR=1
> ```

NO_WERROR=1 make

stappersg, 2022-12-14:
Commit 54c4c1d is an improvement.

The build now fails with:
```text
crypto/deflate.c: In function ‘deflate_inflate’:
crypto/deflate.c:514:41: error: storing the address of local variable ‘zlib_header’ in ‘*deflate.resume’ [-Werror=dangling-pointer=]
  514 |                         deflate->resume = &&zlib_header;
      |                         ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
crypto/deflate.c:507:2: note: ‘zlib_header’ declared here
  507 |  zlib_header: {
      |  ^~~~~~~~~~~
crypto/deflate.c:507:2: note: ‘deflate’ declared here
crypto/deflate.c:543:41: error: storing the address of local variable ‘block_header’ in ‘*deflate.resume’ [-Werror=dangling-pointer=]
  543 |                         deflate->resume = &&block_header;
      |                         ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
crypto/deflate.c:535:2: note: ‘block_header’ declared here
  535 |  block_header: {
      |  ^~~~~~~~~~~~
crypto/deflate.c:535:2: note: ‘deflate’ declared here
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:954: bin/deflate.o] Error 1
```
With
```diff
--- a/src/Makefile.housekeeping
+++ b/src/Makefile.housekeeping
@@ -462,7 +462,9 @@ ifeq ($(CCTYPE),gcc)
 CFLAGS         += -ffreestanding
 CFLAGS         += -fcommon
 CFLAGS         += -Wall -W -Wformat-nonliteral -Wno-array-bounds
+CFLAGS         += -Wno-dangling-pointer
 HOST_CFLAGS    += -Wall -W -Wformat-nonliteral -Wno-array-bounds
+HOST_CFLAGS    += -Wno-dangling-pointer
 endif
 CFLAGS         += $(WORKAROUND_CFLAGS) $(EXTRA_CFLAGS)
 ASFLAGS                += $(WORKAROUND_ASFLAGS) $(EXTRA_ASFLAGS)
```
I do get a _succesfull compile_ on
```text
$ gcc --version
gcc (Debian 12.2.0-3) 12.2.0
```
When I test the _compile result_ is yet unknown.

stappersg, 2022-12-19:
While building with #820 I learnt there is commit [b0ded89e917](https://github.com/ipxe/ipxe/commit/b0ded89e917b48b73097d3b8b88dfa3afb264ed0) that _disables dangling pointer checking for GCC_.  So I get clean compiles with the main branch. When I test the _compile result_ is still unknown.

mcb30, 2023-01-07:
All GCC12 issues should by now have been fixed: please open a new issue if there are still problems.

closed 2023-01-07 by mcb30

purpleidea, 2024-08-30:
I get the following error:

```
  [BUILD] bin-x86_64-efi/acpi.o
In file included from include/string.h:23,
                 from include/ipxe/uaccess.h:25,
                 from core/acpi.c:28:
In function ‘__constant_memcpy’,
    inlined from ‘memcpy’ at arch/x86/include/bits/string.h:168:10,
    inlined from ‘trivial_memcpy_user’ at include/ipxe/uaccess.h:109:2,
    inlined from ‘memcpy_user’ at include/ipxe/efi/efi_uaccess.h:69:2,
    inlined from ‘copy_from_user’ at include/ipxe/uaccess.h:338:2,
    inlined from ‘acpi_sx_zsdt’ at core/acpi.c:217:3:
arch/x86/include/bits/string.h:79:32: error: array subscript ‘union <anonymous>[0]’ is partly outside array bounds of ‘union <anonymous>[1]’ [-Werror=array-bounds=]
   79 |                 dest_u->u32[0] = src_u->u32[0];
      |                 ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
core/acpi.c: In function ‘acpi_sx_zsdt’:
core/acpi.c:191:11: note: object ‘buf’ of size 4
  191 |         } buf;
      |           ^~~
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:943: bin-x86_64-efi/acpi.o] Error 1

```

I tried this with version tag: `v1.21.1` but it has the same issue with git master.

I'm using gcc:

```
gcc --version
gcc (GCC) 14.2.1 20240801 (Red Hat 14.2.1-1)
...
```

On Fedora 40.

I can make a new issue for this if you prefer. You can this error running `make` or make with some bin/... target.

HTH

mcb30, 2024-09-02:
> I get the following error:

See https://github.com/ipxe/ipxe/issues/1293#issuecomment-2324653291