ipxe/ipxe #624: Stuck on iPXE initialising devices VMware Workstation 16.2.1

marcohald, 2022-03-25:
The boot on the VM Stuck at iPXE initialising devices, so I compiled the latest master with this configuration and enabled the VMware Console output:
```
make bin-x86_64-efi/ipxe.efi  EMBED=../boot.ipxe DEBUG="device,runtime,pci,init,nii,snp,snpnet,snponly,efi_watchdog,efi_image,image_cmd,image,efi_autoboot,efi_timer,efi_driver:3,efi_debug,efi_init,efi_pci,efi_wrap,intel" LOG_LEVEL=LOG_ALL
```
The Build with the configuration 
```make bin-x86_64-efi/ipxe.efi  EMBED=../boot.ipxe ```
works without a problem on a Lenovo e15 G3 so it could be a VM NIC specific issue.
The IPXE.efi is chainloaded through a Zenworks PXE evnironment which uses Grub as bootloader.
This is the config of the chainload:
```
insmod chain
chainloader wimboot/ipxe.efi
boot
```
Here is the Output from the vmware.log
```
2022-03-25T07:46:24.818Z In(05) vcpu-0 Guest: Status upon boot failure: No Media
2022-03-25T07:46:24.820Z In(05) vcpu-0 Guest: Status upon boot failure: No Media
2022-03-25T07:46:30.381Z In(05) vcpu-0 Guest: About to do EFI boot: EFI Network
2022-03-25T07:46:31.573Z In(05) vcpu-0 DDB: "longContentID" = "bf52d5b9c4e3065efb8df1c8f45cc875" (was "d3031ce8b8368e3c7b68e6e1e37e5723")
2022-03-25T07:46:43.487Z In(05) vcpu-0 Guest: EFI PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) found autoboot link-layer address:
2022-03-25T07:46:43.492Z In(05) vcpu-0 Guest: 00000000 : 00 0c 29 31 cd ba                               : ..)1..          
2022-03-25T07:46:43.497Z In(05) vcpu-0 Guest: IMAGE boot.ipxe at [da6e000,da6e03a) registered
2022-03-25T07:46:43.502Z In(05) vcpu-0 Guest: EFIIMAGE 0xdab5620 could not load: Error 0x7f048183 (https://ipxe.org/7f048183)
2022-03-25T07:46:43.507Z In(05) vcpu-0 Guest: IMAGE boot.ipxe is not EFI: Error 0x7f048183 (https://ipxe.org/7f048183)
2022-03-25T07:46:43.511Z In(05) vcpu-0 Guest: IMAGE boot.ipxe is script
2022-03-25T07:46:43.516Z In(05) vcpu-0 Guest: iPXE initialising devices...INIT startup efi_tick...
2022-03-25T07:46:43.521Z In(05) vcpu-0 Guest: EFI timer started at 32 ticks per second
2022-03-25T07:46:43.525Z In(05) vcpu-0 Guest: INIT startup devices...
2022-03-25T07:46:43.529Z In(05) vcpu-0 Guest: Adding 3c509 root bus
2022-03-25T07:46:43.544Z In(05) vcpu-0 Guest: Adding EFI root bus
2022-03-25T07:46:43.548Z In(05) vcpu-0 Guest: EFIDRV connecting our drivers
2022-03-25T07:46:43.553Z In(05) vcpu-0 Guest: EFIPCI 0000:00:00.0 (8086:7190 class 060000) has no driver
2022-03-25T07:46:43.558Z In(05) vcpu-0 Guest: EFIPCI 0000:00:01.0 (8086:7191 class 060400) has no driver
2022-03-25T07:46:43.563Z In(05) vcpu-0 Guest: EFIPCI 0000:00:07.0 (8086:7110 class 060100) has no driver
2022-03-25T07:46:43.567Z In(05) vcpu-0 Guest: EFIPCI 0000:00:07.1 (8086:7111 class 01018a) has no driver
2022-03-25T07:46:43.572Z In(05) vcpu-0 Guest: EFIPCI 0000:00:07.3 (8086:7113 class 068000) has no driver
2022-03-25T07:46:43.577Z In(05) vcpu-0 Guest: EFIPCI 0000:00:07.7 (15ad:0740 class 088000) has no driver
2022-03-25T07:46:43.582Z In(05) vcpu-0 Guest: EFIPCI 0000:00:0f.0 (15ad:0405 class 030000) has no driver
2022-03-25T07:46:43.587Z In(05) vcpu-0 Guest: EFIPCI 0000:00:11.0 (15ad:0790 class 060401) has no driver
2022-03-25T07:46:43.592Z In(05) vcpu-0 Guest: EFIPCI 0000:02:00.0 (15ad:0774 class 0c0300) has no driver
2022-03-25T07:46:43.597Z In(05) vcpu-0 Guest: EFIPCI 0000:02:01.0 (15ad:1977 class 040300) has no driver
2022-03-25T07:46:43.602Z In(05) vcpu-0 Guest: EFIPCI 0000:02:02.0 (15ad:0770 class 0c0320) has no driver
2022-03-25T07:46:43.607Z In(05) vcpu-0 Guest: EFIPCI 0000:02:03.0 (15ad:07e0 class 010601) has no driver
2022-03-25T07:46:43.612Z In(05) vcpu-0 Guest: EFIPCI 0000:00:15.0 (15ad:07a0 class 060400) has no driver
2022-03-25T07:46:43.617Z In(05) vcpu-0 Guest: EFIPCI 0000:03:00.0 (8086:10d3 class 020000) has driver "82574l"
2022-03-25T07:46:43.622Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) has driver "PCI"
2022-03-25T07:46:43.627Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) before disconnecting:
2022-03-25T07:46:43.636Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) e3161450-ad0f-11d9-9669-0800200c9a66 supported
2022-03-25T07:46:43.641Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo supported
2022-03-25T07:46:43.650Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo opened 261x (H) by DxeCore for <NULL>
2022-03-25T07:46:43.659Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo opened 1x (D) by Intel(R) PRO/1000 4.0.15 PCI-E 
2022-03-25T07:46:43.665Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo opened 1x (C) by Intel(R) PRO/1000 4.0.15 PCI-E  for PciRoot(0x0)/Pci(0x15
2022-03-25T07:46:43.674Z In(05) vcpu-0 Guest: ,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0)
2022-03-25T07:46:43.679Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo opened 1x (G) by Simple Network Protocol Driver for PciRoot(0x0)/Pci(0x15,
2022-03-25T07:46:43.688Z In(05) vcpu-0 Guest: 0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0)
2022-03-25T07:46:43.693Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath supported
2022-03-25T07:46:43.702Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath opened 75x (H) by DxeCore for <NULL>
2022-03-25T07:46:43.711Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath opened 1x (D) by Intel(R) PRO/1000 4.0.15 PCI-E 
2022-03-25T07:46:43.721Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath opened 1x (T) by Platform Console Management Driver
2022-03-25T07:46:43.730Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath opened 1x (T) by Platform Console Management Driver
2022-03-25T07:46:43.740Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath opened 1x (T) by Simple Network Protocol Driver
2022-03-25T07:46:43.745Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) disconnecting existing drivers
2022-03-25T07:46:43.764Z In(05) vcpu-0 Guest: NII PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) is an NII device
2022-03-25T07:46:43.774Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) has driver "NII"
2022-03-25T07:46:43.783Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) DRIVER_START
2022-03-25T07:46:43.792Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) refusing to start during disconnection
```

marcohald, 2022-03-25:
Booting from a ISO which is compiled with this Command
```
make bin-x86_64-efi/ipxe.iso  EMBED=../boot.ipxe DEBUG="device,runtime,pci,init,nii,snp,snpnet,snponly,efi_watchdog,efi_image,image_cmd,image,efi_autoboot,efi_timer,efi_driver:3,efi_debug,efi_init,efi_pci,efi_wrap,intel" LOG_LEVEL=LOG_ALL
```
s working, so maybe the NIC Driver is not the Issue but something with the chainloading ?
Log:
```
2022-03-25T07:58:43.657Z In(05) vcpu-0 CDROM: switching CDROM state from 5 to 1 using error sense key.
2022-03-25T07:58:43.686Z In(05) vcpu-0 Guest: About to do EFI boot: EFI VMware Virtual SATA CDROM Drive (1.0)
2022-03-25T07:58:43.696Z In(05) vcpu-0 Guest: EFI PciRoot(0x0)/Pci(0x11,0x0)/Pci(0x3,0x0)/Sata(0x1,0x0,0x0)/CDROM(0x1)/HD(1,MBR,0x00000000,0x0,0xB40) has no SNP instance: Er
2022-03-25T07:58:43.697Z In(05) vcpu-0 Guest: ror 0x7f53e083 (https://ipxe.org/7f53e083)
2022-03-25T07:58:43.698Z In(05) vcpu-0 Guest: IMAGE boot.ipxe at [e8f0000,e8f003a) registered
2022-03-25T07:58:43.699Z In(05) vcpu-0 Guest: EFIIMAGE 0xe937620 could not load: Error 0x7f048183 (https://ipxe.org/7f048183)
2022-03-25T07:58:43.700Z In(05) vcpu-0 Guest: IMAGE boot.ipxe is not EFI: Error 0x7f048183 (https://ipxe.org/7f048183)
2022-03-25T07:58:43.700Z In(05) vcpu-0 Guest: IMAGE boot.ipxe is script
2022-03-25T07:58:43.701Z In(05) vcpu-0 Guest: iPXE initialising devices...INIT startup efi_tick...
2022-03-25T07:58:43.702Z In(05) vcpu-0 Guest: EFI timer started at 32 ticks per second
2022-03-25T07:58:43.702Z In(05) vcpu-0 Guest: INIT startup devices...
2022-03-25T07:58:43.706Z In(05) vcpu-0 Guest: Adding 3c509 root bus
2022-03-25T07:58:43.721Z In(05) vcpu-0 Guest: Adding EFI root bus
2022-03-25T07:58:43.725Z In(05) vcpu-0 Guest: EFIDRV connecting our drivers
2022-03-25T07:58:43.730Z In(05) vcpu-0 Guest: EFIPCI 0000:00:00.0 (8086:7190 class 060000) has no driver
2022-03-25T07:58:43.735Z In(05) vcpu-0 Guest: EFIPCI 0000:00:01.0 (8086:7191 class 060400) has no driver
2022-03-25T07:58:43.740Z In(05) vcpu-0 Guest: EFIPCI 0000:00:07.0 (8086:7110 class 060100) has no driver
2022-03-25T07:58:43.745Z In(05) vcpu-0 Guest: EFIPCI 0000:00:07.1 (8086:7111 class 01018a) has no driver
2022-03-25T07:58:43.751Z In(05) vcpu-0 Guest: EFIPCI 0000:00:07.3 (8086:7113 class 068000) has no driver
2022-03-25T07:58:43.755Z In(05) vcpu-0 Guest: EFIPCI 0000:00:07.7 (15ad:0740 class 088000) has no driver
2022-03-25T07:58:43.760Z In(05) vcpu-0 Guest: EFIPCI 0000:00:0f.0 (15ad:0405 class 030000) has no driver
2022-03-25T07:58:43.765Z In(05) vcpu-0 Guest: EFIPCI 0000:00:11.0 (15ad:0790 class 060401) has no driver
2022-03-25T07:58:43.770Z In(05) vcpu-0 Guest: EFIPCI 0000:02:00.0 (15ad:0774 class 0c0300) has no driver
2022-03-25T07:58:43.775Z In(05) vcpu-0 Guest: EFIPCI 0000:02:01.0 (15ad:1977 class 040300) has no driver
2022-03-25T07:58:43.780Z In(05) vcpu-0 Guest: EFIPCI 0000:02:02.0 (15ad:0770 class 0c0320) has no driver
2022-03-25T07:58:43.785Z In(05) vcpu-0 Guest: EFIPCI 0000:02:03.0 (15ad:07e0 class 010601) has no driver
2022-03-25T07:58:43.790Z In(05) vcpu-0 Guest: EFIPCI 0000:00:15.0 (15ad:07a0 class 060400) has no driver
2022-03-25T07:58:43.795Z In(05) vcpu-0 Guest: EFIPCI 0000:03:00.0 (8086:10d3 class 020000) has driver "82574l"
2022-03-25T07:58:43.800Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) has driver "PCI"
2022-03-25T07:58:43.804Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) before disconnecting:
2022-03-25T07:58:43.809Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo supported
2022-03-25T07:58:43.819Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo opened 5x (H) by DxeCore for <NULL>
2022-03-25T07:58:43.824Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath supported
2022-03-25T07:58:43.833Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath opened 65x (H) by DxeCore for <NULL>
2022-03-25T07:58:43.838Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) disconnecting existing drivers
2022-03-25T07:58:43.843Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) after disconnecting:
2022-03-25T07:58:43.848Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo supported
2022-03-25T07:58:43.857Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo opened 5x (H) by DxeCore for <NULL>
2022-03-25T07:58:43.862Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath supported
2022-03-25T07:58:43.872Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath opened 65x (H) by DxeCore for <NULL>
2022-03-25T07:58:43.877Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) connecting new drivers
2022-03-25T07:58:43.882Z In(05) vcpu-0 Guest: EFIPCI 0000:03:00.0 (8086:10d3 class 020000) has driver "82574l"
2022-03-25T07:58:43.886Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) has driver "PCI"
2022-03-25T07:58:43.891Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DRIVER_START
2022-03-25T07:58:43.896Z In(05) vcpu-0 Guest: EFIPCI 0000:03:00.0 (8086:10d3 class 020000) has driver "82574l"
2022-03-25T07:58:43.901Z In(05) vcpu-0 Guest: 0000:03:00.0 (8086:10d3) has driver "82574l"
2022-03-25T07:58:43.906Z In(05) vcpu-0 Guest: 0000:03:00.0 has mem fea20000 io 3000 irq 255
[... cut ...]
2022-03-25T07:58:52.784Z In(05) vcpu-0 Guest: IMAGE pxe.ipxe at [ea7d000,ea7d088) registered
2022-03-25T07:58:52.789Z In(05) vcpu-0 Guest: EFIIMAGE 0xe958168 could not load: Error 0x7f048183 (https://ipxe.org/7f048183)
2022-03-25T07:58:52.794Z In(05) vcpu-0 Guest: IMAGE pxe.ipxe is not EFI: Error 0x7f048183 (https://ipxe.org/7f048183)
2022-03-25T07:58:52.798Z In(05) vcpu-0 Guest: IMAGE pxe.ipxe is script
2022-03-25T07:58:52.798Z In(05) vcpu-0 Guest: Executing "pxe.ipxe"
2022-03-25T07:58:52.803Z In(05) vcpu-0 Guest: IMAGE pxe.ipxe unregistered
```

marcohald, 2022-03-25:
Booting from a ISO which is compiled with this Command
```
make bin-x86_64-efi/ipxe.iso  EMBED=../boot.ipxe DEBUG="device,runtime,pci,init,nii,snp,snpnet,snponly,efi_watchdog,efi_image,image_cmd,image,efi_autoboot,efi_timer,efi_driver:3,efi_debug,efi_init,efi_pci,efi_wrap,intel" LOG_LEVEL=LOG_ALL
```
s working, so maybe the NIC Driver is not the Issue but something with the chainloading ?
Log:
```
2022-03-25T07:58:43.657Z In(05) vcpu-0 CDROM: switching CDROM state from 5 to 1 using error sense key.
2022-03-25T07:58:43.686Z In(05) vcpu-0 Guest: About to do EFI boot: EFI VMware Virtual SATA CDROM Drive (1.0)
2022-03-25T07:58:43.696Z In(05) vcpu-0 Guest: EFI PciRoot(0x0)/Pci(0x11,0x0)/Pci(0x3,0x0)/Sata(0x1,0x0,0x0)/CDROM(0x1)/HD(1,MBR,0x00000000,0x0,0xB40) has no SNP instance: Er
2022-03-25T07:58:43.697Z In(05) vcpu-0 Guest: ror 0x7f53e083 (https://ipxe.org/7f53e083)
2022-03-25T07:58:43.698Z In(05) vcpu-0 Guest: IMAGE boot.ipxe at [e8f0000,e8f003a) registered
2022-03-25T07:58:43.699Z In(05) vcpu-0 Guest: EFIIMAGE 0xe937620 could not load: Error 0x7f048183 (https://ipxe.org/7f048183)
2022-03-25T07:58:43.700Z In(05) vcpu-0 Guest: IMAGE boot.ipxe is not EFI: Error 0x7f048183 (https://ipxe.org/7f048183)
2022-03-25T07:58:43.700Z In(05) vcpu-0 Guest: IMAGE boot.ipxe is script
2022-03-25T07:58:43.701Z In(05) vcpu-0 Guest: iPXE initialising devices...INIT startup efi_tick...
2022-03-25T07:58:43.702Z In(05) vcpu-0 Guest: EFI timer started at 32 ticks per second
2022-03-25T07:58:43.702Z In(05) vcpu-0 Guest: INIT startup devices...
2022-03-25T07:58:43.706Z In(05) vcpu-0 Guest: Adding 3c509 root bus
2022-03-25T07:58:43.721Z In(05) vcpu-0 Guest: Adding EFI root bus
2022-03-25T07:58:43.725Z In(05) vcpu-0 Guest: EFIDRV connecting our drivers
2022-03-25T07:58:43.730Z In(05) vcpu-0 Guest: EFIPCI 0000:00:00.0 (8086:7190 class 060000) has no driver
2022-03-25T07:58:43.735Z In(05) vcpu-0 Guest: EFIPCI 0000:00:01.0 (8086:7191 class 060400) has no driver
2022-03-25T07:58:43.740Z In(05) vcpu-0 Guest: EFIPCI 0000:00:07.0 (8086:7110 class 060100) has no driver
2022-03-25T07:58:43.745Z In(05) vcpu-0 Guest: EFIPCI 0000:00:07.1 (8086:7111 class 01018a) has no driver
2022-03-25T07:58:43.751Z In(05) vcpu-0 Guest: EFIPCI 0000:00:07.3 (8086:7113 class 068000) has no driver
2022-03-25T07:58:43.755Z In(05) vcpu-0 Guest: EFIPCI 0000:00:07.7 (15ad:0740 class 088000) has no driver
2022-03-25T07:58:43.760Z In(05) vcpu-0 Guest: EFIPCI 0000:00:0f.0 (15ad:0405 class 030000) has no driver
2022-03-25T07:58:43.765Z In(05) vcpu-0 Guest: EFIPCI 0000:00:11.0 (15ad:0790 class 060401) has no driver
2022-03-25T07:58:43.770Z In(05) vcpu-0 Guest: EFIPCI 0000:02:00.0 (15ad:0774 class 0c0300) has no driver
2022-03-25T07:58:43.775Z In(05) vcpu-0 Guest: EFIPCI 0000:02:01.0 (15ad:1977 class 040300) has no driver
2022-03-25T07:58:43.780Z In(05) vcpu-0 Guest: EFIPCI 0000:02:02.0 (15ad:0770 class 0c0320) has no driver
2022-03-25T07:58:43.785Z In(05) vcpu-0 Guest: EFIPCI 0000:02:03.0 (15ad:07e0 class 010601) has no driver
2022-03-25T07:58:43.790Z In(05) vcpu-0 Guest: EFIPCI 0000:00:15.0 (15ad:07a0 class 060400) has no driver
2022-03-25T07:58:43.795Z In(05) vcpu-0 Guest: EFIPCI 0000:03:00.0 (8086:10d3 class 020000) has driver "82574l"
2022-03-25T07:58:43.800Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) has driver "PCI"
2022-03-25T07:58:43.804Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) before disconnecting:
2022-03-25T07:58:43.809Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo supported
2022-03-25T07:58:43.819Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo opened 5x (H) by DxeCore for <NULL>
2022-03-25T07:58:43.824Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath supported
2022-03-25T07:58:43.833Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath opened 65x (H) by DxeCore for <NULL>
2022-03-25T07:58:43.838Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) disconnecting existing drivers
2022-03-25T07:58:43.843Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) after disconnecting:
2022-03-25T07:58:43.848Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo supported
2022-03-25T07:58:43.857Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) PciIo opened 5x (H) by DxeCore for <NULL>
2022-03-25T07:58:43.862Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath supported
2022-03-25T07:58:43.872Z In(05) vcpu-0 Guest: HANDLE PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DevicePath opened 65x (H) by DxeCore for <NULL>
2022-03-25T07:58:43.877Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) connecting new drivers
2022-03-25T07:58:43.882Z In(05) vcpu-0 Guest: EFIPCI 0000:03:00.0 (8086:10d3 class 020000) has driver "82574l"
2022-03-25T07:58:43.886Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) has driver "PCI"
2022-03-25T07:58:43.891Z In(05) vcpu-0 Guest: EFIDRV PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0) DRIVER_START
2022-03-25T07:58:43.896Z In(05) vcpu-0 Guest: EFIPCI 0000:03:00.0 (8086:10d3 class 020000) has driver "82574l"
2022-03-25T07:58:43.901Z In(05) vcpu-0 Guest: 0000:03:00.0 (8086:10d3) has driver "82574l"
2022-03-25T07:58:43.906Z In(05) vcpu-0 Guest: 0000:03:00.0 has mem fea20000 io 3000 irq 255
[... cut ...]
2022-03-25T07:58:52.784Z In(05) vcpu-0 Guest: IMAGE pxe.ipxe at [ea7d000,ea7d088) registered
2022-03-25T07:58:52.789Z In(05) vcpu-0 Guest: EFIIMAGE 0xe958168 could not load: Error 0x7f048183 (https://ipxe.org/7f048183)
2022-03-25T07:58:52.794Z In(05) vcpu-0 Guest: IMAGE pxe.ipxe is not EFI: Error 0x7f048183 (https://ipxe.org/7f048183)
2022-03-25T07:58:52.798Z In(05) vcpu-0 Guest: IMAGE pxe.ipxe is script
2022-03-25T07:58:52.798Z In(05) vcpu-0 Guest: Executing "pxe.ipxe"
2022-03-25T07:58:52.803Z In(05) vcpu-0 Guest: IMAGE pxe.ipxe unregistered
```

stappersg, 2022-03-25:
On Fri, Mar 25, 2022 at 01:11:59AM -0700, marcohald wrote:
> ```
> make bin-x86_64-efi/ipxe.iso  EMBED=../boot.ipxe DEBUG="device,runtime,pci,init,nii,snp,snpnet,snponly,efi_watchdog,efi_image,image_cmd,image,efi_autoboot,efi_timer,efi_driver:3,efi_debug,efi_init,efi_pci,efi_wrap,intel" LOG_LEVEL=LOG_ALL
> ```

Acknowledge on "I'm lost"      :-/

For what it is worth "Less is more"
Less debug output is more helpfull on debugging.
Prevent yourself from being flooded.

    ...
> 2022-03-25T07:58:43.956Z In(05) vcpu-0 Guest: INTEL 0xe955ab0 MAC reset (00100260/00080383 was 00100248/00080383)
> 2022-03-25T07:58:43.960Z In(05) vcpu-0 Guest: INTEL 0xe955ab0 has autoloaded MAC [...]

That last URL redirects to https://ipxe.org/err/140860
Partial quote from it:

  * Try building iPXE with the debug option DEBUG=netdevice

My advice:

  make bin-x86_64-efi/ipxe.iso  EMBED=../boot.ipxe DEBUG="netdevice,snp"

marcohald, 2022-03-30:
>  My advice: make bin-x86_64-efi/ipxe.iso EMBED=../boot.ipxe DEBUG="netdevice,snp"
This make a lot less Debug Output:
```
2022-03-30T06:28:25.647Z In(05) vcpu-0 Guest: iPXE initialising devices...NII PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) is an NII device
2022-03-30T06:28:25.648Z In(05) vcpu-0 Guest: NII PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) is an NII device
2022-03-30T06:28:25.649Z In(05) vcpu-0 Guest: NII PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) is an NII device
2022-03-30T06:28:25.650Z In(05) vcpu-0 Guest: NII PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) is an NII device
2022-03-30T06:28:25.691Z In(05) vcpu-0 Guest: NETDEV net0 registered (phys 0000:03:00.0 hwaddr 00:0c:29:31:cd:ba)
2022-03-30T06:28:25.692Z In(05) vcpu-0 Guest: NETDEV net0 link is up
2022-03-30T06:28:25.706Z In(05) vcpu-0 Guest: SNP PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x1)/IPv4(0.0.0.0) is an SNP device
2022-03-30T06:28:25.711Z In(05) vcpu-0 Guest: SNP PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x1)/IPv4(0.0.0.0) is an SNP device
2022-03-30T06:28:25.716Z In(05) vcpu-0 Guest: NETDEV rejecting duplicate (phys SNP-0xf41f698) of net0 (phys 0000:03:00.0)
2022-03-30T06:28:25.721Z In(05) vcpu-0 Guest: SNP PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x1)/IPv6(0000:0000:0000:0000:0000:0000:0000:0000) is an SNP devic
2022-03-30T06:28:25.725Z In(05) vcpu-0 Guest: e
2022-03-30T06:28:25.731Z In(05) vcpu-0 Guest: SNP PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x1)/IPv6(0000:0000:0000:0000:0000:0000:0000:0000) is an SNP devic
2022-03-30T06:28:25.735Z In(05) vcpu-0 Guest: e
2022-03-30T06:28:25.740Z In(05) vcpu-0 Guest: NETDEV rejecting duplicate (phys SNP-0xf404698) of net0 (phys 0000:03:00.0)
2022-03-30T06:28:25.746Z In(05) vcpu-0 Guest: SNP PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x1)/IPv6(0000:0000:0000:0000:0000:0000:0000:0000) is an SNP devic
2022-03-30T06:28:25.750Z In(05) vcpu-0 Guest: e
2022-03-30T06:28:25.756Z In(05) vcpu-0 Guest: SNP PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x1)/IPv6(0000:0000:0000:0000:0000:0000:0000:0000) is an SNP devic
2022-03-30T06:28:25.760Z In(05) vcpu-0 Guest: e
2022-03-30T06:28:25.765Z In(05) vcpu-0 Guest: NETDEV rejecting duplicate (phys SNP-0xf41f718) of net0 (phys 0000:03:00.0)
```
Is the rejected message from the Duplicated IPv4 and IPv6 Stack ?

stappersg, 2022-03-30:
On Tue, Mar 29, 2022 at 11:46:02PM -0700, marcohald wrote:
> This make a lot less Debug Output

Seeing an error is much better as being flooded by errors ...

```
iPXE initialising devices...
NII PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) is an NII device
NII PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) is an NII device
NII PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) is an NII device
NII PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x0) is an NII device
NETDEV net0 registered (phys 0000:03:00.0 hwaddr 00:0c:29:31:cd:ba)
NETDEV net0 link is up
SNP PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x1)/IPv4(0.0.0.0) is an SNP device
SNP PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x1)/IPv4(0.0.0.0) is an SNP device
NETDEV rejecting duplicate (phys SNP-0xf41f698) of net0 (phys 0000:03:00.0)
SNP PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x1)/IPv6(0000:0000:0000:0000:0000:0000:0000:0000) is an SNP device
SNP PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x1)/IPv6(0000:0000:0000:0000:0000:0000:0000:0000) is an SNP device
NETDEV rejecting duplicate (phys SNP-0xf404698) of net0 (phys 0000:03:00.0)
SNP PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x1)/IPv6(0000:0000:0000:0000:0000:0000:0000:0000) is an SNP device
SNP PciRoot(0x0)/Pci(0x15,0x0)/Pci(0x0,0x0)/MAC(000C2931CDBA,0x1)/IPv6(0000:0000:0000:0000:0000:0000:0000:0000) is an SNP device
NETDEV rejecting duplicate (phys SNP-0xf41f718) of net0 (phys 0000:03:00.0)
```

> Is the rejected message from the Duplicated IPv4 and IPv6 Stack ?

I don't know.

Thing I would do if I was stakeholder of this, is to determine if this is an iPXE issue or a VMware Workstation 16.2.1 issue.

stappersg, 2022-05-19:
Acknowledge on _See same problem_. Consider to use this ugly _github_ as a place where smart developers meet. Convert _"Add some prints in src/interface/efi/efi_pci.c"_ into _"this patch will help reproducing this output"_.  Be bold enough to provide output twice, once as picture (for the color information) and once as text (for making further analyzing possible).  Show the (github) world that _"Problem on all Intel drivers with EFI."_ actual means _"Problem on all Intel drivers with EFI on Virtual Box version X.Y.Z.  Succes with the foobar drivers on that Virtual Box version.  Host operating system being used is  Be OS, version 23."_

cvtsi2sd, 2023-05-17:
I see pretty much the same problem on VirtualBox 7.0.8; same Intel 82540em emulated board, and it does work when booting in legacy BIOS mode. I didn't perform the efi_pci.c modifications, but enabled each and every possible debug option compiling with

```
make bin-x86_64-efi/ipxe.efi DEBUG="readline:3,linux_args:3,pxemenu:3,lotest:3,aes:3,md5:3,hash_df:3,asn1:3,x509:3,chap:3,sha256:3,sha1:3,rbg:3,ocsp:3,rootcert:3,deflate:3,entropy:3,certstore:3,hmac_drbg:3,md4:3,privkey:3,gcm:3,drbg:3,dhe:3,cms:3,ntlm:3,rsa:3,sha512:3,md4_test:3,utf8_test:3,test:3,gcm_test:3,memcpy_test:3,iobuf_test:3,sha1_test:3,settings_test:3,tcpip_test:3,md5_test:3,bofm_test:3,bigint_test:3,dns_test:3,sha512_test:3,aes_test:3,ipv4_test:3,ntlm_test:3,linebuf_test:3,ipv6_test:3,profile_test:3,sha256_test:3,vsprintf_test:3,hmac_test:3,setjmp_test:3,ocsp_test:3,librm_mgmt:3,hvm:3,vmware:3,vmconsole:3,guestrpc:3,guestinfo:3,com32_call:3,comboot_call:3,pxe_file:3,pxe_tftp:3,pxe_udp:3,pxe_call:3,rtc_time:3,pcicloud:3,bios_reboot:3,bios_cachedhcp:3,acpipwr:3,int13con:3,vesafb:3,memtop_umalloc:3,memmap:3,bios_console:3,pnpbios:3,acpi_timer:3,biosint:3,apm:3,rsdp:3,int13:3,runtime:3,pic8259:3,relocate:3,debugcon:3,cpuid:3,rdtsc_timer:3,gdbmach:3,rdrand:3,cpuid_settings:3,sdi:3,bootsector:3,initrd:3,elfboot:3,nbi:3,comboot:3,multiboot:3,pxe_image:3,bzimage:3,com32:3,pccrc:3,stp:3,eth_slow:3,neighbour:3,netdev_settings:3,ping:3,fc:3,nfs_open:3,pccrd:3,icmp:3,infiniband:3,ethernet:3,tls:3,fcns:3,ndp:3,rndis:3,httpcore:3,httpconn:3,httpblock:3,ftp:3,httpbasic:3,iscsi:3,httpauth:3,httpdigest:3,syslogs:3,httpntlm:3,vlan:3,peermux:3,fakedhcp:3,eapol:3,aoe:3,tcp:3,lldp:3,peerdisc:3,eap:3,tcpip:3,fcels:3,fcp:3,ipv4:3,fragment:3,ntp:3,syslog:3,dhcpv6:3,tftp:3,slam:3,dhcp:3,dns:3,ib_pathrec:3,ib_cm:3,ib_mcast:3,ib_sma:3,ib_mi:3,ib_smc:3,ib_cmrc:3,ib_packet:3,ib_srp:3,xsigo:3,validator:3,udp:3,dhcpopts:3,icmpv4:3,ipv6:3,icmpv6:3,peerblk:3,netdevice:3,arp:3,retry:3,wpa_tkip:3,wpa_psk:3,sec80211:3,rc80211:3,wpa_ccmp:3,wpa:3,net80211:3,wep:3,peerdist:3,fcoe:3,i2c_bit:3,spi_bit:3,xhci:3,uhci:3,usbhid:3,usbio:3,usbnet:3,usbkbd:3,usbhub:3,usbblk:3,ehci:3,spi:3,nvsvpd:3,nvs:3,threewire:3,pciea:3,pcivpd:3,usb:3,pcibackup:3,isapnp:3,pci:3,isa:3,eisa:3,virtio-pci:3,ecam:3,pciextra:3,mca:3,pci_settings:3,pcimsix:3,pcibridge:3,srp:3,ata:3,ibft:3,scsi:3,golan:3,qib7322:3,hermon:3,linda:3,arbel:3,flexboot_nodnic:3,tlan:3,intel:3,jme:3,forcedeth:3,amd8111e:3,ecm:3,rhine:3,sfc_hunt:3,efx_common:3,efx_hunt:3,smsc75xx:3,netvsc:3,phantom:3,skge:3,sis190:3,velocity:3,realtek:3,eoib:3,ipoib:3,3c90x:3,pcnet32:3,epic100:3,snponly:3,snpnet:3,nii:3,snp:3,sundance:3,ncm:3,w89c840:3,dmfe:3,igbvf_main:3,mii:3,vmxnet3:3,intelx:3,legacy:3,smsc95xx:3,ns8390:3,netfront:3,intelxl:3,eepro:3,myson:3,ice:3,cs89x0:3,dm96xx:3,depca:3,iphone:3,etherfabric:3,lan78xx:3,ena:3,sky2:3,axge:3,myri10ge:3,rdc:3,intelvf:3,icplus:3,pnic:3,virtio-net:3,skeleton:3,3c5x9:3,ne2k_isa:3,3c515:3,smscusb:3,eepro100:3,3c595:3,sis900:3,intelxlvf:3,atl1e:3,thunderx:3,davicom:3,exanic:3,bnxt:3,3c509:3,acm:3,natsemi:3,smc9000:3,rtl818x:3,pxe_loader:3,ath5k_attach:3,ath5k_desc:3,ath5k_reset:3,ath5k_rfkill:3,ath5k_phy:3,ath5k_initvals:3,rtc_entropy:3,pxe_preboot:3,hidemem:3,pxe_undi:3,ath9k:3,ath_key:3,ath9k_eeprom_def:3,ath9k_ar9002_calib:3,ath9k_ar9002_mac:3,ath9k_xmit:3,ath9k_mac:3,ath9k_eeprom_4k:3,ath9k_calib:3,ath9k_main:3,ath9k_ar9003_mac:3,ath9k_ar9003_calib:3,ath9k_ar9002_hw:3,ath9k_init:3,ath9k_ar5008_phy:3,ath9k_eeprom:3,ath9k_eeprom_9287:3,ath9k_ani:3,ath9k_ar9003_phy:3,ath9k_recv:3,ath9k_ar9003_eeprom:3,ath_regd:3,b44:3,ath9k_hw:3,tg3:3,intelxvf:3,xengrant:3,bofm:3,xenbus:3,xenstore:3,tg3_phy:3,tg3_hw:3,tulip:3,efi_time:3,efi_bofm:3,efi_console:3,efi_local:3,efi_umalloc:3,efi_smbios:3,efi_entropy:3,efi_watchdog:3,efi_debug:3,efi_init:3,efi_file:3,efi_timer:3,efi_snp_hii:3,efi_fdt:3,efi_snp:3,efi_autoexec:3,efi_veto:3,efi_cachedhcp:3,efi_pci:3,efi_fbcon:3,efi_cmdline:3,efi_rng:3,efi_pxe:3,efi_utils:3,efi_autoboot:3,efi_download:3,efi_block:3,efi_driver:3,efi_hii:3,smbios:3,smbios_settings:3,pixbuf:3,efi_wrap:3,efi_usb:3,fbcon:3,fdt:3,vmbus:3,acpi_settings:3,dynkeymap:3,pinger:3,ansicoldef:3,timer:3,utf8:3,acpimac:3,cpio:3,keymap:3,debug:3,bitmap:3,job:3,parseopt:3,params:3,time:3,profile:3,menu:3,device:3,process:3,malloc:3,interface:3,ansiesc:3,posix_io:3,open:3,refcnt:3,xferbuf:3,pending:3,blocktrans:3,serial:3,acpi:3,dummy_sanboot:3,init:3,nvo:3,uri:3,xfer:3,memmap_settings:3,base64:3,image:3,archive:3,ansicol:3,exec:3,cachedhcp:3,resolv:3,sanboot:3,embedded:3,elf:3,zlib:3,der:3,settings:3,gzip:3,efi_image:3,script:3,segment:3,pem:3,pnm:3,png" LOG_LEVEL=LOG_ALL
```

and the log gets stuck at the exact same point.

```
<0-Y>EFIDRV PciRoot(0x0)/Pci(0x3,0x0) connecting new drivers
EFIPCI 0000:00:03.0 (8086:100e class 020000) has driver "82540em"
<0-Y>EFIDRV PciRoot(0x0)/Pci(0x3,0x0) has driver "PCI"
<0-Y>EFIDRV PciRoot(0x0)/Pci(0x3,0x0) DRIVER_START
Allocating 0xbe (aligned 0x1+0)
[0x1d5efa40,0x1d66fa28) -> [0x1d5efa40,0x1d5efa40) + [0x1d5efb00,0x1d66fa28)
Allocated [0x1d5efa40,0x1d5efafe)
EFIPCI 0000:00:03.0 (8086:100e class 020000) has driver "82540em"
Allocating 0xe0 (aligned 0x1+0)
[0x1d5efb00,0x1d66fa28) -> [0x1d5efb00,0x1d5efb00) + [0x1d5efbe0,0x1d66fa28)
Allocated [0x1d5efb00,0x1d5efbe0)
```

[log.log](https://github.com/ipxe/ipxe/files/11497474/log.log)

NiKiZe, 2023-05-17:
> I see pretty much the same problem on VirtualBox 7.0.8; same Intel 82540em emulated board, and it does work when booting in legacy BIOS mode. I didn't perform the efi_pci.c modifications, but [...]

Just don't do this, it is not helpful to enable debug on everything, and this might actually create issues. Some debug flags actually intentionally hangs iPXE, otherwise they wouldn't help.

cvtsi2sd, 2023-05-17:
Ok, but this was done essentially out of desperation; until then, I tried only with "reasonable" ones (`make bin-x86_64-efi/ipxe.efi DEBUG=init,efi_utils,efi_pci,efi_driver:3,pci:3,intel:3`), and it hang at `EFIPCI 0000:00:03.0 (8086:100e class 020000) has driver "82540em"`. Given that it seemed strange that I couldn't get anything more by the Intel driver, I gradually enabled everything to see if I was missing something obvious.

stappersg, 2023-05-19:
Knowing that a there are several "`intel*.c`" files.
```text
stappers@paddy:~/src/githubforks/ipxe
$ ls src/drivers/net/intel*c
src/drivers/net/intel.c    src/drivers/net/intelx.c   src/drivers/net/intelxlvf.c
src/drivers/net/intelvf.c  src/drivers/net/intelxl.c  src/drivers/net/intelxvf.c
stappers@paddy:~/src/githubforks/ipxe
$ 
```

I started with `git grep -l 8086` to find out if the `DEBUG=intel:3` correct is.
Getting only one file back, I knew it was the correct one and zoomed in
with `grep 8086 src/drivers/net/intel.c`. Then digging deeper:

```text
stappers@paddy:~/src/githubforks/ipxe
$ grep 8086 src/drivers/net/intel.c  | grep 0x100
	PCI_ROM ( 0x8086, 0x1000, "82542-f", "82542 (Fiber)", 0 ),
	PCI_ROM ( 0x8086, 0x1001, "82543gc-f", "82543GC (Fiber)", 0 ),
	PCI_ROM ( 0x8086, 0x1004, "82543gc", "82543GC (Copper)", 0 ),
	PCI_ROM ( 0x8086, 0x1008, "82544ei", "82544EI (Copper)", 0 ),
	PCI_ROM ( 0x8086, 0x1009, "82544ei-f", "82544EI (Fiber)", 0 ),
	PCI_ROM ( 0x8086, 0x100c, "82544gc", "82544GC (Copper)", 0 ),
	PCI_ROM ( 0x8086, 0x100d, "82544gc-l", "82544GC (LOM)", 0 ),
	PCI_ROM ( 0x8086, 0x100e, "82540em", "82540EM", 0 ),
	PCI_ROM ( 0x8086, 0x100f, "82545em", "82545EM (Copper)", INTEL_VMWARE ),
stappers@paddy:~/src/githubforks/ipxe
$
```
Switched focus to those `0 ),` and that `INTEL_VMWARE ),`:
```text
stappers@paddy:~/src/githubforks/ipxe
$ grep 8086 src/drivers/net/intel.c  | grep 0x100 | awk -F, '{ print $5 }' | sort -u
 0 )
 INTEL_VMWARE )
stappers@paddy:~/src/githubforks/ipxe
$ grep PCI_ROM src/drivers/net/intel.c  | awk -F, '{ print $5 }' | sort -u
 0 )
 INTEL_I219 )
 INTEL_NO_ASDE )
 INTEL_NO_PHY_RST )
 INTEL_PBS_ERRATA )
 INTEL_PBSIZE_RST )
 INTEL_VMWARE )
stappers@paddy:~/src/githubforks/ipxe
$ sed  --silent -e '318,335p' src/drivers/net/intel.h
/** Driver flags */
enum intel_flags {
	/** PBS/PBA errata workaround required */
	INTEL_PBS_ERRATA = 0x0001,
	/** VMware missing interrupt workaround required */
	INTEL_VMWARE = 0x0002,
	/** PHY reset is broken */
	INTEL_NO_PHY_RST = 0x0004,
	/** ASDE is broken */
	INTEL_NO_ASDE = 0x0008,
	/** Reset may cause a complete device hang */
	INTEL_RST_HANG = 0x0010,
	/** PBSIZE registers must be explicitly reset */
	INTEL_PBSIZE_RST = 0x0020,
};

/** The i219 has a seriously broken reset mechanism */
#define INTEL_I219 ( INTEL_NO_PHY_RST | INTEL_RST_HANG )
stappers@paddy:~/src/githubforks/ipxe
$
```
What I suggest:
```text
stappers@paddy:~/src/githubforks/ipxe
$ $EDITOR src/drivers/net/intel.c
stappers@paddy:~/src/githubforks/ipxe
$ git diff
diff --git a/src/drivers/net/intel.c b/src/drivers/net/intel.c
index 46527bdb..18f360cc 100644
--- a/src/drivers/net/intel.c
+++ b/src/drivers/net/intel.c
@@ -1050,7 +1050,7 @@ static struct pci_device_id intel_nics[] = {
        PCI_ROM ( 0x8086, 0x1009, "82544ei-f", "82544EI (Fiber)", 0 ),
        PCI_ROM ( 0x8086, 0x100c, "82544gc", "82544GC (Copper)", 0 ),
        PCI_ROM ( 0x8086, 0x100d, "82544gc-l", "82544GC (LOM)", 0 ),
-       PCI_ROM ( 0x8086, 0x100e, "82540em", "82540EM", 0 ),
+       PCI_ROM ( 0x8086, 0x100e, "82540em", "82540EM", INTEL_VMWARE ),
        PCI_ROM ( 0x8086, 0x100f, "82545em", "82545EM (Copper)", INTEL_VMWARE ),
        PCI_ROM ( 0x8086, 0x1010, "82546eb", "82546EB (Copper)", 0 ),
        PCI_ROM ( 0x8086, 0x1011, "82545em-f", "82545EM (Fiber)", 0 ),
stappers@paddy:~/src/githubforks/ipxe
$ 
```

cvtsi2sd, 2023-05-22:
Thank you for your reply; I had that suspect before, but I excluded it because the problem was present even selecting "Intel PRO/1000 MT Server (82545EM)" as virtual network card, which was already marked with the `INTEL_VMWARE` workaround. Just to make sure, I tried to change this anyway and it hangs just the same. Again, it's working fine with snp.efi (which is reasonable – AFAICT in that case all is handled internally by the EFI firmware), but, intriguingly, it works fine in BIOS mode; maybe the VirtualBox EFI implementation is leaving the virtual NIC in some unexpected state?

stappersg, 2023-05-22:
For what it is worth:
 * Acknowledge on _I tried to change this anyway and it hangs just the same_
and on the _ maybe the VirtualBox EFI implementation is leaving the virtual NIC in some unexpected state?_
 * Now has this issue, with in the title **VMware**, two reports from **Virtualbox** users.
 * All reports are with an EFI environment.

Reonaydo, 2023-06-14:
Same problem with Virtualbox 7.0.8 and all Intel 1000 cards in EFI mode

mcb30, 2023-06-22:
> Same problem with Virtualbox 7.0.8 and all Intel 1000 cards in EFI mode

Split off VirtualBox as #977

NiKiZe, 2026-02-07:
Old issue, can this still be reproduced with current master? If yes do you know if it has worked with any previous version, and if so could you use git bisect to maybe find if there was a specific commit that introduced changed behavior?