ipxe/ipxe #628: [intel] OOM as efirom

9001, 2022-03-28:
When running as efirom, the Intel driver immediately fills the heap on a busy network.

Hardware: i350-t4v2 in a Dell Optiplex 3090 (bios 2.2.0)

I haven't studied it too closely, but:
* UEFI triggers `efi_snp_initialize`, starting up the NIC
* `intel_poll` is called repeatedly
  * calls `intel_poll_rx` which passes half of the ringbuffer to `netdev_rx`
  * calls `intel_refill_rx` which allocates another 8 packets
  * the packets that were passed to `netdev_rx` never get freed
* out of memory after 0x100 packets have gone across the network
  * the heap is filled with 0x800 byte objects from `alloc_rx_iob`

I haven't looked at what `netdev_rx` does yet, or checked if anything calls `intel_empty_rx`.

The following workaround works well for my usecase, since I don't need iPXE to provide SNP devices:

```patch
diff --git a/src/interface/efi/efi_snp.c b/src/interface/efi/efi_snp.c
index 6649eb1b..5b7bce65 100644
--- a/src/interface/efi/efi_snp.c
+++ b/src/interface/efi/efi_snp.c
@@ -238,4 +238,6 @@ static EFI_STATUS EFIAPI
 efi_snp_initialize ( EFI_SIMPLE_NETWORK_PROTOCOL *snp,
 		     UINTN extra_rx_bufsize, UINTN extra_tx_bufsize ) {
+	return 0;
+
 	struct efi_snp_device *snpdev =
 		container_of ( snp, struct efi_snp_device, snp );
```

9001, 2022-03-29:
Looks like this isn't related to intel after all -- it's SNP:
* `efi_snp_get_status` and `efi_snp_receive` is called repeatedly
  * those call `efi_snp_poll` which grabs 8 more packets from `netdev_poll`
* nothing calls `net_poll` or `intel_empty_rx`

So every time a packet is fetched with `efi_snp_receive` it grabs 8 more, quickly running out of memory. And when UEFI eventually tries to boot the iPXE efirom, it's already out of memory, so even stuff like `set dbg 1` will fail.

Maybe it would be an idea to stop retrieving more packets in `efi_snp_poll` when `freemem` gets too low, or when `snpdev->rx` gets too big?

mcb30, 2022-03-30:
I think the problem is that the UEFI firmware is not calling `.Receive()` at a sufficiently high rate to consume the received packets, which is leading to unbounded growth of the queue in the `snpdev->rx` list.

The correct fix is almost certainly to skip the call to either `efi_snp_poll()` or `netdev_poll()` if the receive queue is already non-empty.  I need to spend some time sanity checking this, since `netdev_poll()` is also required in order to report transmit completions, and we also need to ensure that we still assert `EFI_SIMPLE_NETWORK_RECEIVE_INTERRUPT` as needed.

mcb30, 2022-03-30:
(For the record, `intel_empty_rx()` is used only as part of the shutdown path, and `net_poll()` is not supposed to be called via the SNP interface, so neither of those are problems here.)

stappersg, 2022-04-22:
For what it is worth: reporter of this issue also reported #634.
That [possible memory corruption in intel and virtio drivers under bios](https://github.com/ipxe/ipxe/issues/634) issue has detailed information.