ipxe/ipxe #634: possible memory corruption in intel and virtio drivers under bios

9001, 2022-04-21:
Accidentally rebooted an UEFI machine with an i350-t4v2 into BIOS/CSM and saw iPXE mention memory corruption as soon as it tried to DHCP.

Turns out to reproduce easily in VirtualBox with `make DEBUG=dhcp,malloc,iobuf bin/ipxe.iso` on a clean master @ 70995397e5bdfd3431e12971aa40630c7014785f (latest);

* UEFI appears to be unaffected (only happens under BIOS)
* `82540em` (intel) and `virtio-net` are affected -- looks like `Am79C973` (pcnet32) is OK
* No DHCP replies required; the ICMPv6 chatter is enough to trigger it
* Happens with both GCC 4.8.5 and GCC 11.2.0, so don't think it's compiler-related

Relevant snippets below, full logs attached:

```
# vbox-intel
net0: 08:00:27:59:fb:1d using 82540em on 0000:00:03.0 (Ethernet) [open]
Configuring (net0 08:00:27:59:fb:1d)...DHCP 0xe64f4 DHCPDISCOVER
DHCP 0xe64f4 DHCPOFFER from 192.168.56.100:67 for 192.168.56.127
assert(! dma_mapped ( &iobuf->map )) failed at core/iobuf.c line 157
Possible memory corruption detected from 0x33083

# vbox-virtio
Press Ctrl-B for the iPXE command line...
assert(( align == 0 ) || ( ( align & ( align - 1 ) ) == 0 )) failed at core/malloc.c line 292
net0: 08:00:27:59:fb:1d using virtio-net on 0000:00:03.0 (Ethernet) [open]
Possible memory corruption detected from 0x33033
DHCP 0xe61a4 DHCPOFFER from 192.168.56.100:67 for 192.168.56.127
assert(! dma_mapped ( &iobuf->map )) failed at core/iobuf.c line 157
```

Don't think I'll be debugging this since it seems to be fine in UEFI, but figured it'd be good to report :>

[ipxe-intel.log](https://github.com/ipxe/ipxe/files/8535320/ipxe-intel.log) | [ipxe-virtio.log](https://github.com/ipxe/ipxe/files/8535322/ipxe-virtio.log) | [ipxe-pcnet.log](https://github.com/ipxe/ipxe/files/8535321/ipxe-pcnet.log)

NiKiZe, 2022-07-04:
Does this still happen with the current images from boot.ipxe.org?

9001, 2022-08-14:
Sorry to say this still occurs on master right now (cad1cc6b449b63415ffdad8e12f13df4256106fb). I believe the boot.ipxe.org images are not built with `DEBUG=dhcp,malloc,iobuf` so the error would not appear visibly on those.

The problem reproduced right away in a VirtualBox VM after cloning the repo and `make -j8 DEBUG=dhcp,malloc,iobuf bin/ipxe.iso` -- nic can be either virtio-net or 82540em/intel to reproduce.

PS: probably a good idea to set the vm nic to bridged mode, so it can pick up some traffic to trigger the bug with.

NiKiZe, 2022-08-14:
Does the same happen with qemu as well?

9001, 2022-08-14:
Yep :+1: 
```
qemu-system-x86_64 -cdrom bin/ipxe.iso 
```

mcb30, 2022-08-28:
Reproduced in qemu: thank you for the detailed reproduction instructions

mcb30, 2022-08-28:
I'm pretty certain this is a false positive.  Specifically:

Under BIOS the IOMMU is never enabled and so all of the DMA mapping operations are no-ops.  Since we care about code size, the DMA API is arranged so that `dma_map()` etc get optimised away completely by the compiler (including the error handling paths, since DMA mapping cannot fail when the mapping is a no-op).

The [various `static inline` functions in `dma.h`](https://github.com/ipxe/ipxe/blob/a80124456ee9ade7a63b69f167c3b1348c8d93c0/src/include/ipxe/dma.h#L169-L306) that implement this also include debugging code that tracks the number of DMA mappings for the underlying DMA device (e.g. a PCI network card) along with various assertions.  This helps driver developers to ensure that all mappings are eventually unmapped, nothing is mapped twice or unmapped twice, etc.

As with all debug code in iPXE, the debug code is optimised away completely when debugging is disabled.  Debugging is enabled on a per-object basis at build time.

When the DMA API was first implemented, the device driver was solely responsible for all DMA mapping and unmapping operations.  All of the code from `dma.h` that touched a single DMA device was therefore contained within a single object, and the `dma.h` debugging for that device was therefore in a well-defined state of being either enabled or disabled.

Commit https://github.com/ipxe/ipxe/commit/8d337ecda allowed drivers to delegate responsibility for mapping and unmapping I/O buffers to the common shared code in `netdevice.c`.  This commit drastically simplified the changes required to each driver in order to support the new DMA API (which is required in order to function under UEFI if the IOMMU is enabled).

A side effect of this commit is that DMA mapping and unmapping operations are no longer handled solely by a single object.  It is now possible to have a situation in which the code performing the DMA mapping is in a different source code file (and hence a different object) from the code performing the corresponding DMA unmapping.  This in turn means that it is possible for debugging to be enabled for mapping but not for unmapping, or vice versa.

In your example with the `intel` or `virtio` drivers and with `DEBUG=iobuf` enabled:

* The driver uses `alloc_rx_iob()` to allocate the receive I/O buffer.  The DMA mapping is performed in `iobuf.c` with debugging **enabled** and so `map->dma` is set to point to the DMA device.
* The driver passes the received packet to the network device core using `netdev_rx()`.  The DMA unmapping is performed in `netdevice.c` with debugging **disabled** and so `map->dma` is **not** cleared to `NULL`.
* iPXE has a zero-copy data path, and so the same I/O buffer gets passed all the way up the stack to `dhcp.c` and is eventually freed when `dhcp_deliver()` calls `free_iob()`.  The assertion in `free_iob()` is triggered because nothing has cleared `map->dma`.

Adding `netdevice` to the list of debug objects (i.e. `DEBUG=dhcp,malloc,iobuf,netdevice`) causes the debug code to be consistently enabled for both mapping and unmapping operations, and the (false positive) error message then disappears.

mcb30, 2022-08-28:
I'm not sure on the best way to prevent people from tripping over this false positive in future.  It can trivially be solved by e.g. modifying the `flat` API versions of `dma_map()` and `dma_unmap()` to always update `map->dma`.  However, that would unconditionally increase the code size.

A similar fix could be to remove the inlined functions in `iobuf.h` that call `dma_map()` or `dma_unmap()`, replacing them with non-inline functions in `iobuf.c`.  This would ensure that any DMA operations using the I/O buffer API would have a consistent debug state controlled only by `DEBUG=iobuf`.  However, this would also cause even more unnecessary bloat in the non-debug builds.

NiKiZe, 2022-08-28:
I don't think anything needs to be done, but maybe consider if we can have warnings of compiling with debug malloc but without netdevice or similar?