ipxe/ipxe #637: AR8131 Gigabit Ethernet support

Fhiss, 2022-04-29:
Could you add Qualcomm Atheros AR8131 Gigabit Ethernet support, please?
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/ethernet/atheros/atl1c/

NiKiZe, 2022-04-29:
Adding NICs is not done by using Linux kernel source code. What is needed is documentation for the NIC so that new code can be written in a manner that is suitable for iPXE which runs in a quite limited environment.

stappersg, 2022-04-30:
Link https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/ethernet/atheros/atl1c/ was visited.
PCI-IDs were found at https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/ethernet/atheros/atl1c/atl1c_main.c#n22
( then I stumbled upon a broken `util/get-pci-ids` )

@Fhiss  your yes-no-question
> Could you add Qualcomm Atheros AR8131 Gigabit Ethernet support?

is in https://github.com/genodelabs/genode/issues/4487#issuecomment-1111024524 partial answered by
> If **you chose to invest your time** into further exploration/experimentation/porting, you may consider trying to port the Linux-based network driver, following the footsteps of the drivers found at repos/pc/. 

Notice that I have highlighted the _you chose to invest your time_.  Link that I add: https://ipxe.org/dev/driver  (note the `dev` that indicates _intendent audience are **dev**elopers_).

And in https://github.com/genodelabs/genode/issues/4487#issuecomment-1111024524 is
> If you don't, I'd appreciate you closing this issue because I see none of the core Genode developers picking it up.

That is also valid here, _iPXE_.   Do use the insights of [_how to ask smart questions_](http://www.catb.org/~esr/faqs/smart-questions.html) how to continue.

stappersg, 2022-08-26:
For @Fhiss and all others who come along
* Get true technical information about the NIC
* Understand the iPXE code base
* Enable others to help you
* Keep reminding yourself on the goal