ipxe/ipxe #640: ipxe with QLogic Nic on HPE BL460c?

Rampftl, 2022-05-02:
Ist there a chance to make ipxe work with a HPE BL460c server, which has a QLogic NIC (QLogic BCM57840) onboard?

NiKiZe, 2022-05-02:
First of what is the pciid ?

Use lspci -nn in Linux or any other method you are comfortable with.

Rampftl, 2022-05-02:
I looked into the PCI Information Section of the Bios. I post you the screenshots of the output.

[image]
[image]

The pxe boot is done from Port 1. So I guess the pciid for that port is:
14E4-16A2