ipxe/ipxe #641: [Generic] Add my NIC

stappersg, 2022-05-02:
This issue is for documenting how to deal with yet unsupported Network Interface Cards.

Time, the most valuable resource, will tell if an iPXE unsupported NIC remains an iPXE unsupported NIC.

There are several ways to speed up the process.  The fastest way is providing a _merge request_. First a list of slowing down support for _your NIC_
* Assuming your mom works here
* Complaining that you, yourself, lack the skills to do it
* Not sharing what you known about the network interface card
* False flattering like _for gods like you is this easy_

And worst of all: Keeping it **your NIC**. The trick is creating an incentive for others to join you in facing the challenge of iPXE supporting another Network Interface Card.

Advice for you and your companions
* Get true technical information about the NIC
* Understand the iPXE code base
* Enable others to help you
* Keep reminding yourself on the goal

stappersg, 2022-05-02:
This issue was created after seeing #640 and a few days after #637.

NiKiZe, 2022-05-02:
Closing as it is not an issue with iPXE

closed 2022-05-02 by NiKiZe