ipxe/ipxe #642: Your next action build will fail

lookfirst, 2022-05-06:
I cloned your repo and put it into my own private repo and the actions are failing because it can't run `git describe` due to the new "security" mechanism in git. I'm still working on figuring out my own workaround.

see https://github.com/actions/checkout/issues/760

Since you haven't had an action build in a bit, this hasn't come up.

Good times!

lookfirst, 2022-05-06:
Never mind. I found the issue, I cloned the repo, but not all the tags... so I don't have your tags, which causes the build to not generate a good version number... which causes it to fail with a weird error.

closed 2022-05-06 by lookfirst

NiKiZe, 2022-05-06:
If you clone a repo, do a proper clone ;)
But even so, a clone has actions disabled by default. If you want to have actions in a clone, that is up to you. But if you want help, then you should post a link to the failing action, or at least the full output.